Lora-net/LoRaMac-node #518: Build fail for sx1276

x893, 2018-07-25:
Hello,
i use next commands after git clone (develop branch)

cd build
cmake
 -DCMAKE_TOOLCHAIN_FILE="../cmake/toolchain-arm-none-eabi.cmake"
 -DAPPLICATION="LoRaMac"
 -DCLASS="classA"
 -DACTIVE_REGION="LORAMAC_REGION_EU868"
 -DREGION_EU868="ON"
 -DBOARD="NucleoL152"
 -DRADIO="sx1276"
 ..

and see some error after make

[ 12%] Built target peripherals
[ 53%] Built target NucleoL152
[ 54%] Built target radio
[ 68%] Built target system
[ 96%] Built target mac
[ 97%] Linking C executable LoRaMac-classA
CMakeFiles/LoRaMac-classA.dir/classA/NucleoL152/main.c.obj: In function `main':
LoRaMac-node/src/apps/LoRaMac/classA/NucleoL152/main.c:1084: undefined reference to `Radio'
../../mac/CMakeFiles/mac.dir/LoRaMac.c.obj: In function `RxWindowSetup':
LoRaMac-node/src/mac/LoRaMac.c:1676: undefined reference to `Radio'
../../mac/CMakeFiles/mac.dir/LoRaMac.c.obj: In function `ProcessRadioTxDone':
LoRaMac-node/src/mac/LoRaMac.c:846: undefined reference to `Radio'
../../mac/CMakeFiles/mac.dir/LoRaMac.c.obj: In function `ProcessRadioTxTimeout':
LoRaMac-node/src/mac/LoRaMac.c:1272: undefined reference to `Radio'
../../mac/CMakeFiles/mac.dir/LoRaMac.c.obj: In function `HandleRadioRxErrorTimeout':
LoRaMac-node/src/mac/LoRaMac.c:1349: undefined reference to `Radio'
../../mac/CMakeFiles/mac.dir/LoRaMac.c.obj:LoRaMac-node/src/mac/LoRaMac.c:1606: more undefined references to `Radio' follow
../../boards/NucleoL152/CMakeFiles/NucleoL152.dir/board.c.obj: In function `BoardDeInitMcu':
LoRaMac-node/src/boards/NucleoL152/board.c:212: undefined reference to `SX126x'
../../boards/NucleoL152/CMakeFiles/NucleoL152.dir/board.c.obj: In function `BoardInitMcu':
LoRaMac-node/src/boards/NucleoL152/board.c:189: undefined reference to `SX126x'
../../boards/NucleoL152/CMakeFiles/NucleoL152.dir/sx1261dvk1bas-board.c.obj: In function `SX126xIoInit':
LoRaMac-node/src/boards/NucleoL152/sx1261dvk1bas-board.c:56: undefined reference to `SX126x'
../../boards/NucleoL152/CMakeFiles/NucleoL152.dir/sx1261dvk1bas-board.c.obj: In function `SX126xIoDeInit':
LoRaMac-node/src/boards/NucleoL152/sx1261dvk1bas-board.c:68: undefined reference to `SX126x'
collect2: error: ld returned 1 exit status
src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/build.make:268: recipe for target 'src/apps/LoRaMac/LoRaMac-classA' failed
make[2]: *** [src/apps/LoRaMac/LoRaMac-classA] Error 1
CMakeFiles/Makefile2:397: recipe for target 'src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/all' failed
make[1]: *** [src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

How to build classA for SX1276 ?

Thanks

x893, 2018-07-25:
And what ?

mluis1, 2018-07-25:
As said by @nestorayuso the RADIO definition is automatically set by the scripts.

Your cmake command is missing some parameters. 
The Nucleo platforms support different shields. Thus, the MBED_RADIO_SHIELD symbol must be defined.

To build ClassA example for NucleoL152 platform using SX1276MB1MAS shield you must issue the following cmake command:

```cmake
cmake -DCMAKE_BUILD_TYPE=Debug \
           -DCMAKE_TOOLCHAIN_FILE="../cmake/toolchain-arm-none-eabi.cmake" \
           -DAPPLICATION="LoRaMac" \
           -DCLASS="classA" \
           -DCLASSB_ENABLED="OFF" \
           -DACTIVE_REGION="LORAMAC_REGION_EU868" \
           -DREGION_EU868="ON" \
           -DREGION_US915="OFF" \
           -DREGION_CN779="OFF" \
           -DREGION_EU433="OFF" \
           -DREGION_AU915="OFF" \
           -DREGION_AS923="OFF" \
           -DREGION_CN470="OFF" \
           -DREGION_KR920="OFF" \
           -DREGION_IN865="OFF" \
           -DREGION_RU864="OFF" \
           -DBOARD=" NucleoL152" \
           -DMBED_RADIO_SHIELD=" SX1276MB1MAS " .. && make -j4
```

x893, 2018-07-25:
Perfect !!! Now all good.

closed 2018-07-25 by x893