Lora-net/LoRaMac-node #520: Duty cyle management in a different way

sda3422, 2018-08-01:
Hi, I am not sure if this subject has already been discussed here.
Is there a plan to add a different way to manage the duty cycle in the stack ? It could be great to be able to manage it like it is today : idle time computed on each TX to have a "constant" duty cycle management, but also by "time credit" : idle time is computed when the maximum time available on the band is reached. As a consequence (a simple example), in this mode we could be able to transmit during "36 seconds" on 1% duty cycle band before being advised by stack to stay in idle during "1 hour". I think in a lot of case a device would like to transmit few consecutive frames with a very short delay in between but only 1 or 2 times per day. In this use case the current duty cycle management is very restrictive.
Of course I know that applications can disable duty cycle management by the stack and do it by itself !

jamesl-dm, 2018-08-01:
I also wish it worked that way - particularly with NbReps. Would be nicer if all the repeats transmitted without interruption, and the duty cycle backoff happened afterwards.

djaeckle, 2018-08-01:
Hi sda3422,

thanks for your proposal. I think this would be a nice feature. We will set it on the roadmap for discussion.

andershedberg, 2018-11-15:
This is a MUST for async alarm sensors like open/closing of doors/windows which is a major LoRaWAN market for building networks. These sensors might not transmit for a day, but when they transmit they must be able to send rather quickly several messages that are triggered by external events with a small time period in between like "Door open sensor activated" followed by "Tamper sensor activated".

closed 2020-04-21 by mluis1