Lora-net/LoRaMac-node #521: openocd  Cannot identify target as a STM32 family

donnick138, 2018-08-07:
Hi
I'm using OS:Ubuntu 16.04 and MCU STM32L073 + STLinkV2, then follow the development-environment.md to setup debug environment.
When i using GDB (target extended-remote localhost:3333) to connect openocd
it shows "Cannot identify target as a STM32 family" how can i fix it? thanks

Nick

djaeckle, 2018-08-07:
Hi donnick138,

could you provide the complete log? Which OpenOCD version are you using?

donnick138, 2018-08-08:
Hi djaeckle 
The OpenOCD version is Open On-Chip Debugger 0.9.0 (2018-01-24-01:05) Licensed under GNU GPL v2
that i got it by using "sudo apt-get install openocd" from Ubuntu16.04.
Log:
**OpenOCD side**
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v25 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.233202
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : accepting 'gdb' connection on tcp/3333
Warn : Cannot identify target as a STM32L family.
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected

**GDB side**
GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 8.1.0.20180315-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) file src/apps/LoRaMac/LoRaMac-classA
Reading symbols from src/apps/LoRaMac/LoRaMac-classA...done.
(gdb) target extended-remote localhost:3333
Remote debugging using localhost:3333
Remote connection closed
(gdb)

djaeckle, 2018-08-08:
It seems like the OpenOCD does not support this device in your version. Could you update OpenOCD to version 0.10. and double check if it resolves the issue?

donnick138, 2018-08-09:
OK, I download and compiler the OpenOCD version 0.10.0+dev-00077-ge4d98b8 (2018-08-07-11:18), it slove the "Cannot identify target as a STM32L family." issue, but meet another issue is when I using GDB to load firmware it shows "Error: failed erasing sectors 0 to 18" "Error: flash_erase returned -4", is there anything I need to config? Sorry, I am not familar with Linux, almost using windows Keil or sw4stm32 to develop mcu project.

**Openocd Log:**
Open On-Chip Debugger 0.10.0+dev-00077-ge4d98b8 (2018-08-07-11:18)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
WARNING: interface/stlink-v2.cfg is deprecated, please switch to interface/stlink.cfg
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
adapter speed: 300 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : clock speed 240 kHz
Info : STLINK v2 JTAG v25 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.267984
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x21000003 pc: 0x080000ce msp: 0x20004fc0
Info : Device: STM32L0xx (Cat.5)
Info : STM32L flash has dual banks. Bank (0) size is 96kb, base address is 0x8000000
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x08000110 msp: 0x20005000
Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x08000110 msp: 0x20005000
STM32L0: Enabling HSI16
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
Info : Unable to match requested speed 2500 kHz, using 1800 kHz
adapter speed: 1800 kHz
Error: failed erasing sectors 0 to 18
Error: flash_erase returned -4

**GDB Log:**
GNU gdb (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 8.1.0.20180315-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) file src/apps/LoRaMac/LoRaMac-classA
Reading symbols from src/apps/LoRaMac/LoRaMac-classA...done.
(gdb) target extended-remote localhost:3333
Remote debugging using localhost:3333
0x080000ce in __do_global_dtors_aux ()
(gdb) monitor reset halt
Unable to match requested speed 300 kHz, using 240 kHz
Unable to match requested speed 300 kHz, using 240 kHz
adapter speed: 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x08000110 msp: 0x20005000
(gdb) load
Error erasing flash with vFlashErase packet

donnick138, 2018-08-16:
Hi djaeckle
I'm try to setup debug environment with Windows OS, the build and debug function are works fine.
But the Ubuntu environment still stuck in the loading FW to target board.

Nick

djaeckle, 2018-08-17:
Hi donnick138,

thanks for the update. From my point of view, your development environment is configured properly on both, Linux and Windows. As you already mentioned, the debugger is not able to perform a flash erase.

Have you tried to erase or program your device with STM32 ST-Link Utility? Could it be possible that your device is in low power mode and the "Enable Debug in Low power mode" is not activated?

donnick138, 2018-08-17:
Hi djaeckle
I have tried to erase or program my device with STM32 ST-Link Utility, its works fine.
I also check low power mode and connect st-link with RST pin, and check no low power mode connect st-link without RST pin, there are all fail in Linux environment.

djaeckle, 2018-08-17:
Hi donnick138,

do you have a JTAG or a SWD connection?

donnick138, 2018-08-17:
No, I don’t

donnick138, 2018-08-20:
Hi djaeckle
I try to change to stlink-v2.cfg and stm32l0.cfg, it still shows 
1: (3325) <-1007-interpreter-exec console "load"
1: (10847) ->&"Error erasing flash with vFlashErase packet\n"
1: (10847) ->1007^error,msg="Error erasing flash with vFlashErase packet"

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1

StefanFrings, 2019-03-27:
Can we re.open that issue? Because I have the same problem with a new Nucleo-F073RZ board under Debian Linux. I had to add "set CPUTAPID 0x0bc11477" to my debug configuration but then get the following error messages:

srst_only separate srst_nogate srst_open_drain connect_assert_srst
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 240 kHz
adapter_nsrst_delay: 100
Info : clock speed 240 kHz
Info : STLINK v2.1 JTAG v28 API v2 M17 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.255900
Info : Stlink adapter speed set to 240 kHz
Info : STM32F303RETx.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : accepting 'gdb' connection on tcp/3333
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x080024f4 msp: 0x20000570
Error: Cannot identify target as a stm32x
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected

The ST-Link GUI has no problem to access the microcontroller.

StefanFrings, 2019-03-27:
I solved it. For whatever reason, my (generated) debug configuration file contained a wrong include for the STM32F3 instead of STM32L0.
source [find target/stm32f3x.cfg]
I fixed that, now it works fine.

Septeract7, 2019-08-30:
Hi, can we please reopen the issue? I have a similar problem: 

Device: stm32f722
Board: custom
Connection: SWD
Programmer: ST-Link v2

OpenOCD output:
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.240079
Info : stm32f7x.cpu: hardware has 8 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection on tcp/3333         // <----------------------- try to connect from gdb
Info : device id = 0x10006452                                 // <----------------------- wrong id ?
Warn : Cannot identify target as a STM32 family.
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
```

arm-none-eabi-gdb output
```
GNU gdb (GDB) 8.3
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from diyFCSW.elf...
(gdb) target extended-remote :3333
Remote debugging using :3333
Remote connection closed
(gdb) quit
```

The Device ID for stm32f722 devices is specified in RM0431 from ST, section 34.6. 
I think the problem here is that OpenOCD reads the whole register (see section 34.6.1). But in that register only the first 12 Bits represent the Device ID (0x452). The other Bits are the Revision ID (0x1000) and some reserved Bits (apparently 0x6, as seen in OpenOCD output). 

I am fairly new to embedded programming so please excuse if the fault lies somewhere else and I am just making an obvious mistake.

djaeckle, 2019-08-30:
Please open up a new issue.

firmware-hala, 2020-02-14:
similar issue here trying to develop for STM32L412KB in VSCode with Platform.io

PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = stlink
PlatformIO: Initializing remote target...
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : tcl server disabled
Info : telnet server disabled
Info : clock speed 500 kHz
Info : STLINK V2J31M21 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.250593
Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection from pipe
**Info : device id = 0x10006464
Warn : Cannot identify target as an STM32L4 family device.**
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
Error: error during select: Unknown error
.pioinit:13: Error in sourced command file:
Building in debug mode
Checking size .pio\build\nucleo_l412kb\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   2.2% (used 896 bytes from 40960 bytes)
Flash: [=         ]   8.9% (used 11672 bytes from 131072 bytes)
========================= [SUCCESS] Took 3.78 seconds =========================
Reading symbols from c:\Users\firmware\Documents\PlatformIO\Projects\noah_test_analog_02\.pio\build\nucleo_l412kb\firmware.elf...
done.
Remote communication error.  Target disconnected.: Succes

mluis1, 2020-02-14:
Theses issues are related to the openocd tool usage. Users seeking help on this kind of issues will get far better support if they post their questions on forums discussing similar subjects.

Usually, when I observe such problem most of the time it happens because more than one development platform is connected to the same PC. 

By default openocd connects to the first development platform it finds. In the provided log it looks like that you want to connect to a STM32L4 MCU but openocd has found another MCU to connect to.

To verify this assumption I would recommend that you disconnect every platform from your PC and only keep the one you want to debug.

kokkonisd, 2020-12-29:
I have the same issue on macOS with an STM32F103RB and an STM32L452RE. Building from the latest commit of the `develop` branch worked for me.