Lora-net/LoRaMac-node #523: Mulitcast Keys

yoniwt, 2018-08-08:
Hi, 
Thinking that the develop branch work against the LoRaWAN 1.0.3 specification, I have a doubt on some keys that are provided on the Mib_t structure. 
1. As we only need one key for LoRaWAN 1.0,3 multicast specification -- that is NwkSkey -- which one should I use among the MIB_MC_APP_S_KEY and MIB_MC_NWK_S_KEY according to your implementation when initializing

2. What are this two keys: 
    a) multicast encryption key (MIB_MC_KE_KEY)
    b) root key (MIB_MC_KEY)
The LoRaWAN specification doesn't require this keys and requires the kind of keys that are required for the unicast operation -- the only difference being that the devices must share the key. What am I missing?

Thank you very much for the help ...

djaeckle, 2018-08-17:
Hi yoniwt,

thanks for your report. 

1. You need to set both. E.g.:

```
   /*!
    * AES encryption/decryption cipher network session key
    */
   #define MULTICAST_CHANNEL_0_NWKSKEY                 { 0x2B, 0x7E, 0x15, 0x16, 0x28, 0xAE, 0xD2, 0xA6, 
   0xAB, 0xF7, 0x15, 0x88, 0x09, 0xCF, 0x4F, 0x3C }
   /*!
    * AES encryption/decryption cipher application session key
    */
   #define MULTICAST_CHANNEL_0_APPSKEY                 { 0x2B, 0x7E, 0x15, 0x16, 0x28, 0xAE, 0xD2, 0xA6, 
   0xAB, 0xF7, 0x15, 0x88, 0x09, 0xCF, 0x4F, 0x3C }

   static const uint8_t Multicast0NwkSKey[] = MULTICAST_CHANNEL_0_NWKSKEY;
   static const uint8_t Multicast0AppSKey[] = MULTICAST_CHANNEL_0_APPSKEY;

    // Manually setup Multicast Keys.
    MibRequestConfirm_t mibReq;

    // MC channel 0 App session key setup
    mibReq.Type = MIB_MC_APP_S_KEY_0;
    mibReq.Param.McAppSKey0 = ( uint8_t* )Multicast0AppSKey;
    LoRaMacMibSetRequestConfirm( &mibReq );

    // MC channel 0 Nwk session key setup
    mibReq.Type = MIB_MC_NWK_S_KEY_0;
    mibReq.Param.McNwkSKey0 = ( uint8_t* )Multicast0NwkSKey;
    LoRaMacMibSetRequestConfirm( &mibReq );

    MulticastChannel_t channel;
    // Setup Channel 0
    channel.AddrID = MULTICAST_0_ADDR;
    channel.Address = MULTICAST_CHANNEL_0_ADDRESS;
    channel.IsEnabled = true;
    channel.Frequency = MULTICAST_CHANNEL_0_FREQUENCY;
    channel.Datarate = MULTICAST_CHANNEL_0_DATARATE;
    channel.Periodicity = MULTICAST_CHANNEL_0_PERIODICITY;
    LoRaMacMulticastChannelSet( channel );
```

2. I need to double check this with my colleagues.

yoniwt, 2018-08-17:
Thank you for the response
1. Thank you for the detailed answer. Indeed, I will need both session keys because it is treated as ABP device. sorry for the confusion and thank you for the clarification.
2. At the End of the day, the MIB_MC_APP_S_KEY and MIB_MC_NWK_S_KEY should be used in the stack. MIB_MC_KE_KEY is used to  decrypt the McKey_encrypted that comes in the application layer and generate the MIB_MC_KEY. The MIB_MC_KEY is used together with the Multicast Address to generate the MIB_MC_APP_S_KEY and MIB_MC_NWK_S_KEY. However, I didn't get from your stack functions that deal with this translations and therefore I don't really see there use in the current implementation. On top of this using  MIB_MC_APP_S_KEY and MIB_MC_NWK_S_KEY will override the use of other keys that are used in the intermediate step.  Therefore, I think the other keys don't have any use for now unless I extend the stack with an application layer functions that take care of the intermediate function. Can you confirm whether my logic and conclusion is right?

Thank you again,

mluis1, 2019-02-12:
Please take a look at branch `feature/examples-refactoring`. In that branch we have started to implement the new duplicative layer specifications provided by the LoRa-Alliance. 

One of these packages consists on how to setup the multicast channels.

closed 2019-02-12 by mluis1