Lora-net/LoRaMac-node #525: Suggestion: SX1276SetOpMode order of power down

ChrisEAlfred, 2018-08-10:
To get the radio into very low power, we also turn off an external crystal driving the SX1276. The user's radio chip driver implements SX1276SetAntSwLowPower(bool).

The current implementation of sx1276.c :: void SX1276SetOpMode( uint8_t opMode ) is

`void SX1276SetOpMode( uint8_t opMode )
{
   ...

    if( opMode == RF_OPMODE_SLEEP )
    {
        SX1276SetAntSwLowPower( true );
    }
    else
    {
        SX1276SetAntSwLowPower( false );
        SX1276SetAntSw( opMode );
    }
    SX1276Write( REG_OPMODE, ( SX1276Read( REG_OPMODE ) & RF_OPMODE_MASK ) | opMode );
}
`

We suggest changing the order to:

`void SX1276SetOpMode( uint8_t opMode )
{
  ...

    // Power up BEFORE powering/changing SX1276
    if (opMode != RF_OPMODE_SLEEP)
    {
        SX1276SetAntSwLowPower(false);
        SX1276SetAntSw(opMode);
    }
    SX1276Write( REG_OPMODE, ( SX1276Read( REG_OPMODE ) & RF_OPMODE_MASK ) | opMode );

    // Power down AFTER powering down SX1276
    if (opMode == RF_OPMODE_SLEEP)
    {
        SX1276SetAntSwLowPower(true);
    }
}
`

This allows the implementer to power up the crystal in  SX1276SetAntSwLowPower(false), and power down the crystal in  SX1276SetAntSwLowPower(true).

ChrisEAlfred, 2018-08-10:
Might be a little more complicated than this.
We found that the crystal must be running when the SPI commands are issued.

Looking at radio/sx1276/sx1276.c, in there are many places where we end up calling SX1276SetOpMode(RF_OPMODE_SLEEP) and then continue to issue SPI commands i.e. after the Tcxo has been turned off by SX1276SetAntSwLowPower().

mluis1, 2018-11-15:
The B-L072Z-LRWAN1 platform uses a TCXO.
Please take a look at [sx1276-board.c](https://github.com/Lora-net/LoRaMac-node/blob/develop/src/boards/B-L072Z-LRWAN1/sx1276-board.c) to see how TCXO power up/down is handled.

branek, 2019-03-12:
We would like to reopen this, and suggest acceptance of proposed change.

Based on most recent testing on Murata module, current implementation of SX1276SetOpMode() renders SX1276 unresponsive. Reason is that current implementation first turns off TCXO, and than set op mode to RF_OPMODE_SLEEP. That way SX1276 operates without clock for a short period, while in mode diffrent than RF_OPMODE_SLEEP. Based on datasheet reading, our opinion is that should be changed:

When going to mode other than RF_OPMODE_SLEEP:
first turn on TCXO, 
and than set REG_OPMODE

When going to RF_OPMODE_SLEEP:
first set REG_OPMODE
than turn off TCXO.

So, proposed code by @ChrisEAlfred is correct.

> ```
> void SX1276SetOpMode( uint8_t opMode )
> {
>   ...
>
>     // Power up BEFORE powering/changing SX1276
>     if (opMode != RF_OPMODE_SLEEP)
>     {
>         SX1276SetAntSwLowPower(false); [...]

branek, 2019-03-28:
@mluis1 Just a kindly reminde to give some comment on this.

mluis1, 2019-03-28:
As @ChrisEAlfred says it is a little more complex than just change that function.

I guess that the best option would be to change `SX1276SetAntSwLowPower` function. Remove the call to `SX1276SetBoardTcxo( true );` and `SX1276SetBoardTcxo( false );`

The call to `SX1276SetBoardTcxo( true );` should be called every time we set an operating mode different from SLEEP. The issue is that we need to avoid calling the function if the TCXO is already ON due to the `BOARD_TCXO_WAKEUP_TIME` delay.

The call to `SX1276SetBoardTcxo( false );` should be called only when one is sure that no more operations are required.

I'll try to propose a patch in order for you to try and validate that the fix is OK.

mluis1, 2019-03-28:
Could you please apply the below patch to your clone develop branch?

Please let us know if it fixes the issue?
```patch
diff --git a/src/boards/B-L072Z-LRWAN1/sx1276-board.c b/src/boards/B-L072Z-LRWAN1/sx1276-board.c
index c582521c..19981ef1 100644
--- a/src/boards/B-L072Z-LRWAN1/sx1276-board.c
+++ b/src/boards/B-L072Z-LRWAN1/sx1276-board.c
@@ -134,18 +134,16 @@ void SX1276IoDeInit( void )
     GpioInit( &SX1276.DIO5, RADIO_DIO_5, PIN_INPUT, PIN_PUSH_PULL, PIN_NO_PULL, 0 );
 }
 
-/*!
- * \brief Enables/disables the TCXO if available on board design.
- *
- * \param [IN] state TCXO enabled when true and disabled when false.
- */
-static void SX1276SetBoardTcxo( uint8_t state )
+void SX1276SetBoardTcxo( uint8_t state )
 {
     if( state == true )
     {
-        // Power ON the TCXO
-        GpioWrite( &TcxoPower, 1 );
-        DelayMs( BOARD_TCXO_WAKEUP_TIME );
+        if( GpioRead( &TcxoPower ) == 0 )
+        { // TCXO OFF power it up.
+            // Power ON the TCXO
+            GpioWrite( &TcxoPower, 1 );
+            DelayMs( BOARD_TCXO_WAKEUP_TIME );
+        }
     }
     else
     {
diff --git a/src/boards/sx1276-board.h b/src/boards/sx1276-board.h
index f886226b..e9d6a95b 100644
--- a/src/boards/sx1276-board.h
+++ b/src/boards/sx1276-board.h
@@ -121,6 +121,13 @@ void SX1276SetAntSw( uint8_t opMode );
  */
 bool SX1276CheckRfFrequency( uint32_t frequency );
 
+/*!
+ * \brief Enables/disables the TCXO if available on board design.
+ *
+ * \param [IN] state TCXO enabled when true and disabled when false.
+ */
+void SX1276SetBoardTcxo( uint8_t state );
+
 /*!
  * \brief Gets the Defines the time required for the TCXO to wakeup [ms].
  *
diff --git a/src/radio/sx1276/sx1276.c b/src/radio/sx1276/sx1276.c
index 07ef4093..d135b957 100644
--- a/src/radio/sx1276/sx1276.c
+++ b/src/radio/sx1276/sx1276.c
@@ -854,6 +854,10 @@ void SX1276SetSleep( void )
     TimerStop( &TxTimeoutTimer );
 
     SX1276SetOpMode( RF_OPMODE_SLEEP );
+
+    // Disable TCXO radio is in SLEEP mode
+    SX1276SetBoardTcxo( false );
+
     SX1276.Settings.State = RF_IDLE;
 }
 
@@ -1200,6 +1204,8 @@ void SX1276SetOpMode( uint8_t opMode )
     }
     else
     {
+        // Enable TCXO if operating mode different from SLEEP.
+        SX1276SetBoardTcxo( true );
         SX1276SetAntSwLowPower( false );
         SX1276SetAntSw( opMode );
     }
@@ -1227,14 +1233,14 @@ void SX1276SetModem( RadioModems_t modem )
     {
     default:
     case MODEM_FSK:
-        SX1276SetSleep( );
+        SX1276SetOpMode( RF_OPMODE_SLEEP );
         SX1276Write( REG_OPMODE, ( SX1276Read( REG_OPMODE ) & RFLR_OPMODE_LONGRANGEMODE_MASK ) | RFLR_OPMODE_LONGRANGEMODE_OFF );
 
         SX1276Write( REG_DIOMAPPING1, 0x00 );
         SX1276Write( REG_DIOMAPPING2, 0x30 ); // DIO5=ModeReady
         break;
     case MODEM_LORA:
-        SX1276SetSleep( );
+        SX1276SetOpMode( RF_OPMODE_SLEEP );
         SX1276Write( REG_OPMODE, ( SX1276Read( REG_OPMODE ) & RFLR_OPMODE_LONGRANGEMODE_MASK ) | RFLR_OPMODE_LONGRANGEMODE_ON );
 
         SX1276Write( REG_DIOMAPPING1, 0x00 );
```

branek, 2019-03-28:
@mluis1 Issue with enable/disable TCXO is gone now. I just had to remove the call to `SX1276SetBoardTcxo( true );` and `SX1276SetBoardTcxo( false );` from `SX1276SetAntSwLowPower` as that is missing in patch you provided. TCXO is also turned of during MCU sleep.

Changes from SX1276SetSleep( ); to SX1276SetOpMode( RF_OPMODE_SLEEP ); in SX1276SetModem will skip some TimerStop calls and bit settings. Is that intentional?

mluis1, 2019-03-28:
Sorry, I forgot to include that change in the patch.

Yes, the changes were intended. The issue is that `SX1276SetBoardTcxo( false );` is called inside `SX1276SetSleep` function. The modem switch must be done when the radio is in sleep mode thus, we can't turn off the TCXO. I thought about to also add the `TimerStop` function calls in `SX1276SetModem` function but, I guess that it should be OK without those. This is the reason why I asked if you could test the changes. 

Unfortunately currently I have no Murata board up and running so I can't test this right away.

I will apply these changes once I have run some tests on the Murata board.

Please find attached the full patch to be integrated.
[tcxo-fix.patch.txt](https://github.com/Lora-net/LoRaMac-node/files/3018813/tcxo-fix.patch.txt)

branek, 2019-03-29:
Thanks for this, but after one night of testing it does not looks like a full fix. Right now, with this patch applied and with HW based on Murata module, we are experiencing unreliable UL sending. Most of the time more than 7 or 8 Join Request messages are needed to join successfully. JR message was received by the server, but JA was sent randomly.

It sounds that problem is not realted to node, but without a patch (and with change proposed in first post), all is working as expected. We will test more to try to find the reason.

mluis1, 2019-03-29:
Thanks for the feedback.

When I get some time I will setup a B-L072Z-LRWAN1 board connected to a RedwoodComm pre-certificatiopn tool. Like this all the features will be exercised. I'll keep you informed.

mluis1, 2019-03-29:
The pre-certification tests just finished and the have run successfully on B-L072Z-LRWAN1 platform using the provided patch.

Please find attached the pre-certification test report.

[B-L072Z-LRWAN1-EU868_CERTIFICATION_EU_TEST_RESULT_20190329.pdf](https://github.com/Lora-net/LoRaMac-node/files/3023616/B-L072Z-LRWAN1-EU868_CERTIFICATION_EU_TEST_RESULT_20190329.pdf)

branek, 2019-04-01:
We are also able to confirm changes are ok. Problem was on our side, misconfiguration on network server.

Still to confirm is power consumption, but looking at logs, we do not expect problems.

closed 2019-04-02 by mluis1

mitwave, 2020-09-09:
Hi.
I found the bug for TCXO On/Off when OpMode is changed.

```
static void SX1276SetOpMode( uint8_t opMode )
{
#if defined( USE_RADIO_DEBUG )
  :
  :
#endif
    if( opMode == RF_OPMODE_SLEEP )
    {
        SX1276Write( REG_OPMODE, ( SX1276Read( REG_OPMODE ) & RF_OPMODE_MASK ) | opMode ); /* should be there */
        SX1276SetAntSwLowPower( true );
        SX1276SetBoardTcxo( false );    /* should be there */
    }
    else
    {
        // Enable TCXO if operating mode different from SLEEP.
        SX1276SetBoardTcxo( true );
        SX1276SetAntSwLowPower( false );
        SX1276SetAntSw( opMode );
        SX1276Write( REG_OPMODE, ( SX1276Read( REG_OPMODE ) & RF_OPMODE_MASK ) | opMode ); /* should be there */
    }
    /* should not be there SX1276Write( REG_OPMODE, ( SX1276Read( REG_OPMODE ) & RF_OPMODE_MASK ) | opMode ); */
}
```

For changing sleep mode with power saving...
1. SX1276 Reg update
2. SX1276AntSwLowPower(true)
3. SX1276SetBoardTcxo(false)
For changing other mode.
1. SX1276SetBoardTcxo(true)
2. SX1276AntSwLowPower(false)
3. SX1276SetAntSw(opMode);
4. SX1276 Reg update

Would you give me the advice for this change?

Thanks
Ryosuke