Lora-net/LoRaMac-node #527: Bug in CalcNextV11X in v5 LoRaMacAdr.c?

jamesl-dm, 2018-08-13:
There are two 'if( datarate == minTxDatarate )' clauses, and the first one seems to make the second unreachable. So once the datarate hits the minimum, it looks like the ADR flag will be turned off, terminating the sequence prematurely.

djaeckle, 2018-08-17:
Hi jamesl-dm,

the reason for the second occurance of `if( datarate == minTxDatarate )` is, that the MAC decreases the datarate just before. Thus, we need to verify if we are on `minTxDatarate ` to reconfigure the channels mask. However, taking a look at the current implementation for `CalcNextV11X`, it seems like we have not applied the correct sequence. I have to double check that.

djaeckle, 2018-09-19:
Hi jamesl-dm,

i was able to verify the issue and indeed this is a bug. We are working on a fix and we will provide it soon.

closed 2018-09-21 by mluis1