Lora-net/LoRaMac-node #530: AS923 duty cycle / LBT query

jamesl-dm, 2018-08-15:
Can you offer an opinion on the AS923 duty cycle / LBT setup? Looking in the regional params (1.0.2a/b, and 1.1b), the two specified AS923 join channels are listed with a 1% duty cycle, and there is no mention of LBT anywhere except in the KR920 region. But the duty cycle seems to be disabled in AS923 (#define AS923_DUTY_CYCLE_ENABLED 0), and there is an LBT procedure.

I'm guessing from the ARIB standards you've cited before that the LBT is indeed needed for Japan in the AS923 region. So presumably the regional specification just failed to mention this. But I'm not sure what to make of the 1% duty cycle?

mluis1, 2019-02-12:
The 1% duty-cycle for Join procedure is enforced for all regions and cannot be disabled.
The description of this enforcement is described in the LoRaWAN specification chapter 7 "Retransmissions back-off".

As previously said the LBT is performed for all transmissions when using AS923. This is true even on countries listed in AS923 that don't require it. 
The fact that the LBT is always enabled should not harm and makes the end-devices more RF spectrum usage friendly.

closed 2019-02-12 by mluis1

jamesl-dm, 2019-02-13:
>The 1% duty-cycle for Join procedure is enforced for all regions and cannot be disabled.
What's confusing me is what happens after the Join procedure, or in ABP, where there is no join procedure. If the region actually requires a transmit duty cycle (and some of the regions intended for AS923 seem to), surely the band duty cycle needs to be enabled as well as the join duty cycle? Other regions with similar descriptions in the regional parameters documents DO have their band duty cycles enabled, but AS923 doesn't.

iotdevzone-01, 2019-02-13:
James, there's a good chance that regulatory rules in countries like New Zealand where there's no current AS923 ISM dwell-time/duty-cycle enforcement will soon change. There are moves afoot on this already. I'd plan to work with the assumption that it will be enforced at some point in the near future.

jamesl-dm, 2019-02-13:
Thanks - yes I am enabling duty cycle in AS923 on all our devices. Just not sure why it isn't like that upstream...