Lora-net/LoRaMac-node #532: (Bug) Problem in LoRaMacClassBRxBeacon( uint8_t *payload, uint16_t size )

yoniwt, 2018-08-17:
I was doing a test in EU Region on a Network that supports class B. Doing so I faced an issue in locking the beacon. Even if the device was able to lock the beacon sometimes, it failed to lock the beacon in most cases no matter how long I waited. However,  when I bypass CRC by forcing beaconProcessed to 1, the device managed to lock the beacon more easily. I think there is a bug in how the CRC is checked. Are you aware of this problem?

djaeckle, 2018-08-27:
Hi yoniwt,

thanks for the report. Which version are you using?

yoniwt, 2018-08-27:
Thank you for the response,
I am using the develop branch. I cloned it 2 months ago. I also recloned 1 month ago and retired. So I think you can check on the latest version of the develop branch.

yoniwt, 2018-08-28:
Dear djaeckle, 
I cloned the develop branch today and tried again, still required the CRC to be turned off

djaeckle, 2018-08-28:
Hi yoniwt,

actually, we didn't have any issues so far for the beacon crc. Which region are you using?

yoniwt, 2018-08-29:
The region is EU868 and we checked the beacon packet transmitted to the device whether it is on time and also has the right CRC and it indeed was correct.

mluis1, 2018-09-19:
I have just tested the EU868 ClassB example on current develop branch using https://eu.iot.semtech.cloud network server and default [packet forwarder](https://github.com/Lora-net/packet_forwarder).

As you can observe in the end-device below log everything looks to be operating as expected.

```
###### ===== BEACON 1221397120 ==== ######
GW DESC     : 0
GW INFO     : 00 00 00 00 00 00

FREQ        : 869525000
DATA RATE   : DR_3
RX RSSI     : -91
RX SNR      : 7

###### ===== MCPS-Request ==== ######
STATUS      : OK

###### ===== MCPS-Confirm ==== ######
STATUS      : OK

###### ===== UPLINK FRAME 53 ==== ######

CLASS       : B

TX PORT     : 3
TX DATA     : UNCONFIRMED
01 4F 0B B7

DATA RATE   : DR_5
U/L FREQ    : 867100000
TX POWER    : 5
CHANNEL MASK: 00FF

###### ===== MCPS-Indication ==== ######
STATUS      : OK

###### ===== DOWNLINK FRAME 17 ==== ######
RX WINDOW   : Ping-Slot
RX PORT     : 1
RX DATA     :
01

DATA RATE   : DR_3
RX RSSI     : -81
RX SNR      : 7

###### ===== MCPS-Indication ==== ######
STATUS      : OK

###### ===== DOWNLINK FRAME 18 ==== ######
RX WINDOW   : Ping-Slot
RX PORT     : 1
RX DATA     :
00

DATA RATE   : DR_3
RX RSSI     : -81
RX SNR      : 6

###### ===== MCPS-Indication ==== ######
STATUS      : OK

###### ===== DOWNLINK FRAME 19 ==== ######
RX WINDOW   : Ping-Slot
RX PORT     : 1
RX DATA     :
01

DATA RATE   : DR_3
RX RSSI     : -81
RX SNR      : 6
```

yoniwt, 2018-09-19:
Thank you for your response,
we were using NucleoL152 with sx1276mb1mas and sx1272mb2das chipset. Perhaps could the problem be related to such  board and chipset combination?

mluis1, 2018-09-20:
I didn't check this using the same exact platform as you did. I'll run as soon as possible the tests using your platform.

Although, for the moment I don't think that the issue you are observing is related to the used end-device platform. 

In order to have a better understanding of your environment could you please indicate:
1. Which gateway hardware are you using?
2. Is the gateway using the standard packet_forwarder? If yes, could you provide which version of the packet_forwarder is being used?
3. Could you please provide an example of a received beacon buffer data?

Example of received beacon data:
The beacon is defined as follows by the EU868 regional parameters specification:
```c
            // A beacon frame is defined as:
            // Bytes: |  x   |  4   |  2   |     7      |  y   |  2   |
            //        |------|------|------|------------|------|------|
            // Field: | RFU1 | Time | CRC1 | GwSpecific | RFU2 | CRC2 |
            //
            // Field RFU1 and RFU2 have variable sizes. It depends on the region specific implementation
```
For EU868 region the RFU fields are defined as follows:
```
RFU1: 2
RFU2: 0
```

The following received beacon buffer `00 00 80 2A CE 48 A8 60 00 00 00 00 00 00 00 00 00` is decoded as follows:
```
RFU1: 00 00
Time: 80 2A CE 48
CRC1: 60A8
GwSpecific: 00 00 00 00 00 00 00
RFU2: N/A
CRC2: 00 00
```

When printed the beacon data becomes:
```
###### ===== BEACON 1221470848 ==== ######
GW DESC     : 0
GW INFO     : 00 00 00 00 00 00

FREQ        : 869525000
DATA RATE   : DR_3
RX RSSI     : -87
RX SNR      : 6
```

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1

PellegrinoA, 2019-04-18:
> I have just tested the EU868 ClassB example on current develop branch using https://eu.iot.semtech.cloud network server and default [packet forwarder](https://github.com/Lora-net/packet_forwarder). [...]

You can recive all beacon??? or only the first?