Lora-net/LoRaMac-node #535: Can't use a certain Channel Mask

Gecko05, 2018-08-21:
I'm currently using a OTAA LoRaMac-node in a L152RE and having some issues with the node, the first one happens when I manually set the channel mask, in this case 0xFF00, 0x0000, 0x0000, 0x0000, 0x0000. When I use the mentioned mask the MCU appears to get a HardFault produced at the _ComputeNext125kHzJoinChannel_ function, especifically when trying to get **findAvailableChannelsIndex** . Seems like the channel mask I'm trying to use wasn't contemplated to work with the current firmware, what should I do?

ConnyBusy, 2018-08-22:
Hi Gecko05,
from your channel mask you have disabled all channels (both 500kHz & 125KHz). I guess you intended to set the mask to 0x0000, 0x0000, 0x0000, 0x0000, 0x00FF which should enable only the first 8 125kHz channels

ConnyBusy, 2018-08-22:
@Gecko05 Please note that the behavior for the join request have been changed from LoRaWAN version 1.0.3 onwards for the US915/AU915 regions. This might be causing the issue you are seeing. 

[image]
It makes difficult to use hybrid mode as with the previous specifications. A channel is selected from each block of eight channels i.e. Block 0, Block 1 etc..