Lora-net/LoRaMac-node #539: Need help with MacCtx.MacState and LORAMAC_IDLE

Gecko05, 2018-08-23:
Hi, I'm prototyping an End-Node with OTAA, unconfirmed messages, ADR and Region US915. I'm currently facing an issue with MacCtx.MacState, after sending around 20 or 40 uplinks to my TTN gateway the value of this structure stops being reset to _LORAMAC_IDLE_ so my node stops sending frames because it appears to be Busy inside **LoRaMacMcpsRequest** function. So what I did was just add a counter variable to check how many times it appears to be "busy" and manually reset it to _LORAMAC_IDLE_ with **MacCtx.MacState &= ~LORAMAC_TX_RUNNING** when the counter goes over to 2, however I don't want to use this approach because it seems to be far from optimal. 
I need help identifying the source of this issue, is there a timer or an Interrupt in charge of resetting _MacCtx.MacState_ back to _LORAMAC_IDLE_? As an additional detail I've noticed that it occurs when I receive no debug message from _McpsConfirm STATUS_ or when I get a _McpsConfirm STATUS: ERROR_

djaeckle, 2018-08-27:
Hi Gecko05,

which version are you using?

Gecko05, 2018-08-27:
I'm using v1.03 from the develop branch

oyj0594, 2018-08-28:
I have a same problem as @Gecko05 explained. it's randomly blocked after 5~30 frames.

ly1243667342, 2018-09-26:
I have a same problem.
[13:49:12(998)Rx]# ===== MLME-Confirm ==== #####
[13:49:13(031)Rx]#
STATUS      : Error

#####
[13:49:13(078)Rx]# ===== MLME-Request - MLME_JOIN ==== ######

[13:49:13(123)Rx]
STATUS      : Busy

ly1243667342, 2018-09-26:
@Gecko05 @djaeckle 
for handle this issue, i debug many times , find  MacCtx.AckTimeoutTimer.IsRunning  is true , cause MacCtx.MacFlags.Bits.MacDone is not 1 , i remove (if( MacCtx.AckTimeoutTimer.IsRunning == false )), it can work.

becasuse AckTimeoutTimer is not head , MacCtx.AckTimeoutTimer.IsRunning aways true.

    if( MacCtx.AckTimeoutTimer.IsRunning == false )
    {  // Procedure is completed when the AckTimeoutTimer is not running anymore
        MacCtx.MacFlags.Bits.MacDone = 1;
    }

PeterCasavant, 2018-09-27:
I have the same issue. I  do not like my fix, and would like something that is a release.

ly1243667342, 2018-09-28:
@Gecko05 @PeterCasavant @djaeckle 
i think i have solve this issue .
i modified two places. 
first, in the funciton ProcessRadioRxDone() remove below line.
`//if( MacCtx.AckTimeoutTimer.IsRunning == false || macHdr.Bits.MType == FRAME_TYPE_JOIN_ACCEPT)`
second,in the funciton LoRaMacProcess() add two lines.
```
    // MAC proceeded a state and is ready to check
    if( MacCtx.MacFlags.Bits.MacDone == 1 )
    {
        // A error occurs during receiving
        if( ( MacCtx.MacState & LORAMAC_RX_ABORT ) == LORAMAC_RX_ABORT )
        {
            MacCtx.MacState &= ~LORAMAC_RX_ABORT;
            MacCtx.MacState &= ~LORAMAC_TX_RUNNING;
        }

        // An error occurs during transmitting
        if( IsRequestPending( ) > 0 )
        {
               ........
        }
		**else
			MacCtx.MacState &= ~LORAMAC_TX_RUNNING;**

```
i am not sure that will cause other issue. i have test 3 hours , all is going well.
if you can give a warning for me, thanks .

djaeckle, 2018-10-05:
Hi all,

many thanks for all of your analysis. This issue is known and we are working on a solution.

Regimbal, 2018-10-25:
Hey guys,
I'm trying to check if this bug occurs on my end nodes, relying on the last develop branch commit.
My test node sends periodical frames each 2 minutes. The node is configured to use ADR on region EU868.
After more than 2 hours and ~80 uplink frames, the device is okay.

Is there any specific configuration or environment that produces this state?
Also, I can't find line
`//if( MacCtx.AckTimeoutTimer.IsRunning == false || macHdr.Bits.MType == FRAME_TYPE_JOIN_ACCEPT)`
mentioned by @ly1243667342 in his hotfix suggestion...

djaeckle, 2018-10-26:
Hi all,

we have prepared multiple fixes for this issue. Please let us know, if the issue still occurs on your side. Thanks in advance.

Regimbal, 2018-11-02:
Hi all,
We were able to reproduce the bug thanks to @djaeckle description despite the original @Gecko05 described behavior only mentions 20-40 uplink frames. Whatever, the last commits fixed it for me.

Regimbal, 2018-11-02:
Guys,

I don't know if that is due to these specific commits, but I get weird behavior now.

When making join attempts while network is offline (to simulate out of coverage for instance), my debug printf outputs Dutycycle restricted (which is normal) when join attempts are too close to each other, but the join request frame is sent anyway. And if the network is on, then the devices successfully joins the network.

djaeckle, 2018-11-08:
HiI Regimbal,

for now, i wasn't able to reproduce the issue. Which region are you using?

Regimbal, 2018-11-08:
I'm using EU868 band. Will try to reproduce it again tomorrow or next week and will keep you inform

Regimbal, 2018-11-28:
Sorry, I just had time to look at this.
I was actually wrong claiming the joining request happens while it prints "Dutycycle restricted".
Here is what actually happens:
From the time the device unjoins the network in my app (following several consecutive missed ack),
- the device tries to join at the next uplink event.
- the join procedure is correctly done (no dutycycle restriction observed here)
- the join fails and we get a RX2 timeout Status to MLME-Confirm
- In MlmeConfirm function, JoinNetwork() is called again if Status is not ok. -> this call leads to Dutycycle restricted message.

In the end, I'd say the bug concerned by this topic is successfully corrected.

What I am not sure about is the fact that join procedure is now not always subject to DutyCycle restriction. I'll look further into it and eventually open another topic.

djaeckle, 2019-03-04:
Hi Regimbal,

> The thing is the First join shall immediately leads to DutyCycle restriction as long as it is generated right after the previous attempt, no?

This is the correct behaviour, as the EU region specifies a join duty cycle. Can we close this issue now?

gpgib6, 2019-03-19:
Hi djaeckle, I'm using the feature 5.0.0 branch, how can I verify that I have the fix in place? Where can I find the version of the code that I'm using? I checked LoRaMac.c and I found this (Current LoRaWAN Version):

    Version_t lrWanVersion;
    lrWanVersion.Fields.Major    = 1;
    lrWanVersion.Fields.Minor    = 1;
    lrWanVersion.Fields.Revision = 0;
    lrWanVersion.Fields.Rfu      = 0;
    MacCtx.NvmCtx->Version = lrWanVersion;

Does that mean I have the fix in place already? Either way this is still happening where very rarely the device gets stuck in a way where the LoRa stack is thinking it's still in TX mode, when it is not. Then nothing works anymore.

Can you tell me what the proper fix is so I can double-check it?

Thanks!
-Peter

mluis1, 2019-03-22:
@gpgib6 The provided fixes are listed above in this issue. You can check them.

If your code is up to date against the current feature/5.0.0 then the fixes are included in it.

mluis1, 2019-03-29:
Can we close this issue?

closed 2019-04-05 by mluis1