Lora-net/LoRaMac-node #543: Incorrect positive RX window offsets from RegionCommonComputeRxWindowParameters() ?

catsup, 2018-08-30:
Hi,

can someone explain the rationale behind the **receive window offset** calculation in this function ?
```
void RegionCommonComputeRxWindowParameters( double tSymbol, uint8_t minRxSymbols, uint32_t rxError, uint32_t wakeUpTime, uint32_t* windowTimeout, int32_t* windowOffset )
{
    *windowTimeout = MAX( ( uint32_t )ceil( ( ( 2 * minRxSymbols - 8 ) * tSymbol + 2 * rxError ) / tSymbol ), minRxSymbols ); // Computed number of symbols
    *windowOffset = ( int32_t )ceil( ( 4.0 * tSymbol ) - ( ( *windowTimeout * tSymbol ) / 2.0 ) - wakeUpTime );
}
```
The reason I am asking is because I am getting zero or positive windowOffset values, which causes lost downlinks as the receiver is turned on way too late for RX1/RX2 at certain datarates.

Am I missing something obvious here ?

Thanks,
Marc

ps: Below is a dump of various offsets and timeouts calculated by the MAC for EU and US regions. 
```
 Regional RX window parameters for EU868
 DR  BW OFFST TMOUT
  0   0    32     6
  1   0    16     6
  2   0     4     7
  3   0    -3     9
  4   0    -7    14
  5   0    -9    24
  6   1   -10    44
  7   0   -10   129

 Regional RX window parameters for US915
 DR  BW OFFST TMOUT
  0   0     4     7
  1   0    -3     9
  2   0    -7    14
  3   0    -9    24
  4   2   -10    44
  5   0     0     6
  6   0     0     6
  7   0     0     6
  8   2     4     7
  9   2    -3     9
 10   2    -7    14
 11   2    -9    24
 12   2   -10    44
 13   2   -10    83
```

mluis1, 2018-08-31:
Positive offsets are valid. 

The offset is computed as described on the `RegionComputeRxWindowParameters` function documentation.

Please find below the extracted function documentation.

```c
/*
 * Rx window precise timing
 *
 * For more details please consult the following document, chapter 3.1.2.
 * https://www.semtech.com/uploads/documents/SX1272_settings_for_LoRaWAN_v2.0.pdf
 * or
 * https://www.semtech.com/uploads/documents/SX1276_settings_for_LoRaWAN_v2.0.pdf
 *
 *                 Downlink start: T = Tx + 1s (+/- 20 us)
 *                            |
 *             TRxEarly       |        TRxLate
 *                |           |           |
 *                |           |           +---+---+---+---+---+---+---+---+
 *                |           |           |       Latest Rx window        |
 *                |           |           +---+---+---+---+---+---+---+---+
 *                |           |           |
 *                +---+---+---+---+---+---+---+---+
 *                |       Earliest Rx window      |
 *                +---+---+---+---+---+---+---+---+
 *                            |
 *                            +---+---+---+---+---+---+---+---+
 *Downlink preamble 8 symbols |   |   |   |   |   |   |   |   |
 *                            +---+---+---+---+---+---+---+---+
 *
 *                     Worst case Rx window timings
 *
 * TRxLate  = DEFAULT_MIN_RX_SYMBOLS * tSymbol - RADIO_WAKEUP_TIME
 * TRxEarly = 8 - DEFAULT_MIN_RX_SYMBOLS * tSymbol - RxWindowTimeout - RADIO_WAKEUP_TIME
 *
 * TRxLate - TRxEarly = 2 * DEFAULT_SYSTEM_MAX_RX_ERROR
 *
 * RxOffset = ( TRxLate + TRxEarly ) / 2
 *
 * RxWindowTimeout = ( 2 * DEFAULT_MIN_RX_SYMBOLS - 8 ) * tSymbol + 2 * DEFAULT_SYSTEM_MAX_RX_ERROR
 * RxOffset = 4 * tSymbol - RxWindowTimeout / 2 - RADIO_WAKE_UP_TIME
 *
 * Minimal value of RxWindowTimeout must be 5 symbols which implies that the system always tolerates at least an error of 1.5 * tSymbol
 */
/*!
 * Computes the Rx window timeout and offset.
 *
 * \param [IN] region       LoRaWAN region.
 *
 * \param [IN] datarate     Rx window datarate index to be used
 *
 * \param [IN] minRxSymbols Minimum required number of symbols to detect an Rx frame.
 *
 * \param [IN] rxError      System maximum timing error of the receiver. In milliseconds
 *                          The receiver will turn on in a [-rxError : +rxError] ms
 *                          interval around RxOffset
 *
 * \param [OUT]rxConfigParams Returns updated WindowTimeout and WindowOffset fields.
 */
void RegionComputeRxWindowParameters( LoRaMacRegion_t region, int8_t datarate, uint8_t minRxSymbols, uint32_t rxError, RxConfigParams_t *rxConfigParams );

```

jamesl-dm, 2018-09-03:
The equations in docs look a bit off to me.
I think that:

>  * TRxLate  = DEFAULT_MIN_RX_SYMBOLS * tSymbol - RADIO_WAKEUP_TIME
>  * TRxEarly = 8 - DEFAULT_MIN_RX_SYMBOLS * tSymbol - RxWindowTimeout - RADIO_WAKEUP_TIME

Should change to:
TRxLate  = **(8 - DEFAULT_MIN_RX_SYMBOLS)** * tSymbol - RADIO_WAKEUP_TIME
TRxEarly = **DEFAULT_MIN_RX_SYMBOLS** * tSymbol - RxWindowTimeout - RADIO_WAKEUP_TIME

The bottom two lines (solutions for RxWindowTimeout and RxOffset) are consistent with this change, so presumably it was just a typo?

jamesl-dm, 2018-09-03:
On second thought, shouldn't the third equation be removed from the system entirely? The assumption in that system of equations is that DEFAULT_SYSTEM_MAX_RX_ERROR is exactly equal to the permissible timing variation:

> TRxLate - TRxEarly = 2 * DEFAULT_SYSTEM_MAX_RX_ERROR

But this should really be an inequality, since the user gets to set the rx error parameter freely.

Surely what we really want to do here, is align the middle of the receive window with the middle of the 8 symbol preamble? The RxOffset equation does this correctly. But the RxWindowTimeout is giving incorrect answers because we specified that the RX error was exactly dimensioned to the extra preamble symbols. For instance, if you set the DEFAULT_SYSTEM_MAX_RX_ERROR to zero, the RxWindowTimeout will give an answer of 2 symbols, which it too short (and leads to a positive offset). I think what we really want is:

RxWindowTimeout = DEFAULT_MIN_RX_SYMBOLS * tSymbol + 2 * DEFAULT_SYSTEM_MAX_RX_ERROR

catsup, 2018-09-04:
Thanks for the clarification Miguel, I see the point as far as power savings are concerned.

Wouldn't it be worthwhile adjusting the transmitter as well to shave off the unused 3 symbols ? That would save way more power (ie, for DR0, on rx we save 32ms x 10mA, on tx that would be 32ms x 180mA) ?

Marc

BenAtPip, 2018-09-30:
Would anyone be able to comment on the success of implementing jamesl-dm's proposed changes to the receive timing? I am experiencing downlinks not correctly being serviced by an ABP end device in the AS923 region. It results in the device not correctly servicing MAC commands via downlinks.

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1