Lora-net/LoRaMac-node #547: LoRaMacQueryTxPossible return false at first call in class A, ABP and with payload >51bytes

AloyseTech, 2018-09-04:
Hi,

The first call to LoRaMacQueryTxPossible (in SendFrame function)  returns false if the payload size is more than 51 bytes. It seems that the payload size is checked against the PHY_DEF_TX_DR which is DR_0 for region EU868. Since this is SF12, the maximum payload size is 51 bytes. This is the case even if LORAWAN_DEFAULT_DATARATE is set to DR_5 (SF7).

djaeckle, 2018-09-05:
Hi AloyseTech,

thanks for the report. I will double check it.

djaeckle, 2018-10-05:
Hi AloyseTech,

i think this is not an issue of the stack - the MAC does not know `LORAWAN_DEFAULT_DATARATE ` as it is defined in the main application. The symbol `LORAWAN_DEFAULT_DATARATE ` is a helper to define the default datarate for the application. However, if you want to setup the MAC to use a different default DR, then you can set e.g. for the AS region the symbol `AS923_DEFAULT_DATARATE` to DR_5.

closed 2019-02-12 by mluis1