Lora-net/LoRaMac-node #549: Use __NOP() in a system file

yorickbrunet, 2018-09-05:
Hello,

I set up Visual Studio Code using your [development environment doc](https://github.com/Lora-net/LoRaMac-node/blob/develop/Doc/development-environment.md). I can build successfully an application of the develop branch and program my Nucleo-L152RE.

I would like to add a new module in [system](https://github.com/Lora-net/LoRaMac-node/tree/develop/src/system) to add debug possibilities using GPIOs. These GPIOs would be read with a logic analyser or oscilloscope. I need to use __NOP() in order to output longer or shorted pulses. I don't need to have a pulse lasting a specific amount of time, just have longer or shorter pulses. For example:
`GpioWrite( &Debug1, 1 );`
`__NOP( );`
`__NOP( );`
`__NOP( );`
`GpioWrite( &Debug1, 0 );`

However, CMake makes an error when compiling:
>[build] /.../LoRaMac-node/src/system/debug.c:37: undefined reference to `__NOP'

I cannot find which header to import to be able to use __NOP(). Ideally, the header to import should be independent from the board.

Thanks.
Yorick

temporaryaccount, 2018-09-06:
Just include the "stm32l1xx.h" which will in turn include the necessary CMSIS header where `__NOP` is defined.

yorickbrunet, 2018-09-06:
I thought of doing that, but actually the build fails. See error below when adding 
`include "stm32l1xx.h"`
above other includes in the file.

[...]
[build] [ 15%] Building C object src/system/CMakeFiles/system.dir/debug.c.obj
[build] In file included from /.../LoRaMac-node/src/system/debug.c:22:0:
[build] /.../LoRaMac-node/src/boards/NucleoL152/cmsis/stm32l1xx.h:183:3: error: #error "Please select first the target STM32L1xx device used in your application (in stm32l1xx.h file)"
[build]   #error "Please select first the target STM32L1xx device used in your application (in stm32l1xx.h file)"
[build]    ^~~~~
[build] src/system/CMakeFiles/system.dir/build.make:86: recipe for target 'src/system/CMakeFiles/system.dir/debug.c.obj' failed
[build] make[2]: *** [src/system/CMakeFiles/system.dir/debug.c.obj] Error 1
[build] make[1]: *** [src/system/CMakeFiles/system.dir/all] Error 2
[...]