Lora-net/LoRaMac-node #551: US915 & 9 channel Tektelic Pico GW

HankKauffmann, 2018-09-12:
Sorry if this is pretty basic, I'm a new to LoRaWAN and still trying wrap my head around it.

My Tektelic Pico Gateway has eight 125kHz and one 500kHz channel. How can I configure my RegionUS915.h file so my mote's channels are aligned with the GW?

This is what is produced by RegionUS915InitDefaults() when US915_MAX_NB_CHANNELS is set to 9.

0) Freq(125kHz)=902300
1) Freq(500kHz)=903000
2) Freq(500kHz)=904600
3) Freq(500kHz)=906200
4) Freq(500kHz)=907800
5) Freq(500kHz)=909400
6) Freq(500kHz)=911000
7) Freq(500kHz)=912600
8) Freq(500kHz)=914200

- Hank

ConnyBusy, 2018-09-14:
@HankKauffmann 
You may want to set the chmask bits for NvmCtx.ChannelsDefaultMask[i] in RegionUS915InitDefaults() instead not change the US915_MAX_NB_CHANNELS to 9. I am afraid you will not be able to perform OTAA due to the current change in the channel selection mechanism in the LoRaWAN specification.

HankKauffmann, 2018-09-14:
Thanks. You've validated my solution, I found I had to do that as well.  I'll only change ChannelsDefaultMask and call it done. Where I was coming from was that I was hoping for a solution that didn't require code changes to personalize the app.

NvmCtx.ChannelsDefaultMask[0] = 0x00ff; //0xFFFF;
NvmCtx.ChannelsDefaultMask[1] = 0x0000; //0xFFFF;
NvmCtx.ChannelsDefaultMask[2] = 0x0000; //0xFFFF;
NvmCtx.ChannelsDefaultMask[3] = 0x0000; //0xFFFF;
NvmCtx.ChannelsDefaultMask[4] = 0x0001; //0x00FF;
NvmCtx.ChannelsDefaultMask[5] = 0x0000;

closed 2018-09-14 by HankKauffmann

shaunak11, 2018-10-04:
@HankKauffmann Hi, I am trying to update the firmware for Tektelic Pico Gateway but since the file name of firmware is different for the gateway. When I do 
"binary"
"put semtech_1.4.0_pico.bin"

It throws this error:
putting semtech_1.4.0_pico.bin to 10.211.69.44:semtech_1.4.0_pico.bin [octet]
Error code 256: File Not Found

Can you tell me how do you go ahead and update the firmware?