Lora-net/LoRaMac-node #552: DeviceTimeReq/Ans causes Node system time synchronization issues with BeaconTime??

ConnyBusy, 2018-09-13:
Hi,
I am using the current develop branch and experiencing some weird issues after the DeviceTimeReq/DeviceTimeAns MAC commands I can see on the trace terminal that the Node is expecting beacons at times that are not synchronized with the GPS time. Is this a known issue?

ty4tw, 2019-01-26:
Hi, ConnyBusy,

We are facing the same problem. so,I closed my issue.
My fix of #664 can get the beacon, but only once.
Because BeaconCtx's Time related variables are not UTC.
I'm trying to fix this problem. I will share the code when I solve the problem.

ty4tw, 2019-02-02:
Hi,

I could not synchronize the UTC.

Follows are my solution to get the Beacon after execution of DeviceTimeReq Mac command.
```
void LoRaMacClassBDeviceTimeAns( void )
{
#ifdef LORAMAC_CLASSB_ENABLED

    SysTime_t nextBeacon = SysTimeGet( );
    uint32_t currentTimeMs = SysTime2Ms( nextBeacon );

    nextBeacon.Seconds = nextBeacon.Seconds + ( ( CLASSB_BEACON_INTERVAL / 1000 ) - ( nextBeacon.Seconds % ( CLASSB_BEACON_INTERVAL / 1000 ) ) );
/*
    Ctx.NvmCtx->BeaconCtx.NextBeaconRx = SysTime2Ms( nextBeacon );
    if( Ctx.NvmCtx->BeaconCtx.NextBeaconRx > CLASSB_BEACON_INTERVAL )
    {
        Ctx.NvmCtx->BeaconCtx.LastBeaconRx = Ctx.NvmCtx->BeaconCtx.NextBeaconRx - CLASSB_BEACON_INTERVAL;
    }
    else
    {
        Ctx.NvmCtx->BeaconCtx.LastBeaconRx = 0;
    }
*/
    if( LoRaMacConfirmQueueIsCmdActive( MLME_DEVICE_TIME ) == true )
    {
        if( currentTimeMs > Ctx.NvmCtx->BeaconCtx.NextBeaconRx )
        {
            // We missed the beacon already
            Ctx.NvmCtx->BeaconCtx.LastBeaconRx = 0;
            Ctx.NvmCtx->BeaconCtx.NextBeaconRx = 0;
            LoRaMacConfirmQueueSetStatus( LORAMAC_EVENT_INFO_STATUS_BEACON_NOT_FOUND, MLME_DEVICE_TIME );
        }
        else
        {
            Ctx.NvmCtx->BeaconCtx.Ctrl.BeaconDelaySet = 1;
            Ctx.NvmCtx->BeaconCtx.BeaconTimingDelay = Ctx.NvmCtx->BeaconCtx.NextBeaconRx - currentTimeMs;
            Ctx.NvmCtx->BeaconCtx.BeaconTime = nextBeacon.Seconds - UNIX_GPS_EPOCH_OFFSET - ( CLASSB_BEACON_INTERVAL / 1000 );
            LoRaMacConfirmQueueSetStatus( LORAMAC_EVENT_INFO_STATUS_OK, MLME_DEVICE_TIME );
        }
    }

    NvmContextChange( );
#endif // LORAMAC_CLASSB_ENABLED
}
```

mluis1, 2019-02-12:
This is a known issue. We have logged it internally 7 months ago. Unfortunately we didn't had the time to work on it yet.

On our internal system the issue is logged as follows.

>**First beacon after MCU reset and DeviceTimeAns is never acquired**
>
>Currently we base the beacon expected time according to the `BeaconCtx.LastBeaconRx` time. The issue is that after a MCU reset the value will always be 0 which prevents the correct acquisition.
>
>I think that we should rely only on the expected beacon reception time. This is now possible due to the addition of the systime module.

The fix proposed on issue #664 looks promising as a quick fix. Must be verified.

```c
void LoRaMacClassBBeaconTimerEvent( void* context )
{
#ifdef LORAMAC_CLASSB_ENABLED
    Ctx.NvmCtx->BeaconCtx.TimeStamp = TimerGetCurrentTime( );
    TimerStop( &Ctx.BeaconTimer );
    LoRaMacClassBEvents.Events.Beacon = 1;

    if( Ctx.LoRaMacClassBCallbacks.MacProcessNotify != NULL )
    {
        Ctx.LoRaMacClassBCallbacks.MacProcessNotify( );
    }
#endif // LORAMAC_CLASSB_ENABLED
}
```

It should be

```c
    //Ctx.NvmCtx->BeaconCtx.TimeStamp = TimerGetCurrentTime( );
    SysTime_t current = SysTimeGet();
    Ctx.NvmCtx->BeaconCtx.TimeStamp = current.Seconds;
```

tortizactility, 2019-02-12:
@mluis1 

This code workaround seems to enable a permanent listening phase to be sure to catch the beacon.

Best Regards,
Thibault

mluis1, 2019-02-12:
Thanks @tortizactility,

Our first intention was to refactor the ClassB handling in order to only rely on the `SysTimeGet` and `SysTimeSet` functions. As such we think that for instance the `BeaconCtx.LastBeaconRx` variable is useless. We should only rely only on `BeaconTime` value.

Another thing that we need to modify is to update the `SysTime` each time a beacon is received and potentially we could also update the `SysTime` every time something is received on a ping-slot.

ConnyBusy, 2019-02-14:
@mluis1 
Having a closer look at the implementation/processing of the DeviceTimeAns I am beginning to wonder if there is a fundamental misunderstanding on my side on how the specification defines the contents of the DeviceTimeAns or I am not fully grasping what the code is doing.

1.  I would expect the following to happen during processing. Compare contents of DeviceTimeAns with LastTxTime + Transmit Time on Air of the last used data rate. 
Reason: the specification says, the contents of DeviceTimeAns is the time captured at the end of uplink transmission. 
[image]

2. Is there a reason that DeviceTimeReq/DeviceTimeAns is not available in Class A? Unlike PingSlotInfoReq/PingSlotInfoAns and others available in Class B which can only be sent in Class A mode. I would expect DeviceTimeReq/DeviceTimeAns to be available in Class A even though it's not useful in Class A. Moreover, this command is listed in Class A MAC Commands section of the specification.

mluis1, 2019-02-14:
DeviceTimeReq/Ans can be used in all end-device classes operation. There is no requirement to operate in  ClassB.

The `fuota-test-01` example provided on feature branch `feature/examples-refactoring` uses the `DeviceTimeReq` MAC command to synchronize the system time. In this example the end-device starts in ClassA then the network server sets up a temporary ClassC multicast session to be started at a given time in future. At the end of the session the end-device reverts back to ClassA.

Generally, the `DeviceTimeReq/Ans` is useful if one would like to synchronize the end-device current system time as such this MAC command can be issued at any time and when operating in any end-device class.
The `DeviceTimeReq/Ans` can be used to speedup the beacon acquisition and then the end-device can switch from ClassA to ClassB.

The time synchronization is done using the SysTime captured when the TxDone interrupt is handled(end of transmission) and then by adding to the `sysTimeCurrent` the difference between received time and `LastTxSysTime`. Please see below code excerpt.

Please also note that this function has been tested against all major network server providers and there is also one test house (which is currently putting in place the certification for ClassB).

```c
                // Copy received GPS Epoch time into system time
                sysTime = gpsEpochTime;
                // Add Unix to Gps epcoh offset. The system time is based on Unix time.
                sysTime.Seconds += UNIX_GPS_EPOCH_OFFSET;

                // Compensate time difference between Tx Done time and now
                sysTimeCurrent = SysTimeGet( );
                sysTime = SysTimeAdd( sysTimeCurrent, SysTimeSub( sysTime, MacCtx.LastTxSysTime ) );

                // Apply the new system time.
                SysTimeSet( sysTime );
```

ConnyBusy, 2019-02-14:
@mluis1 
Thanks for your clarification on the processing of the DeviceTimeAns. Looks right.

About the class A:
The current source code would not trigger DeviceTimeReq for Class A only example.

mluis1, 2019-02-14:
The ClassA example doesn't trigger the DeviceTimeReq.
But, nothing prevents you to modify the example to do so.

PellegrinoA, 2019-04-10:
Hi @mluis1 .

You said:
"DeviceTimeReq/Ans can be used in all end-device classes operation. There is no requirement to operate in ClassB."

If classB don't use "DeviceTimeReq/Ans", how does it find the beacon???

Thanks.

mluis1, 2019-04-17:
@PellegrinoA 

I guess that you have misunderstood what I meant.

DeviceTimeReq/Ans can be used at any moment.

This MAC command has been added to ease the beacon search on regions like US915 where the beacon hops in frequency.

If you look at ClassB example you will notice that the end-device starts as a ClassA end-device then it issue a DeviceTimeReq MAC command. Once it receives the MAC answer the beacon acquisition is started. Then when the beacon is received the end-device switches to ClassB. Then each time a beacon is received the system time is updated.

This MAC command is also very useful to just synchronize the end-device clock with the network server one. In ClassA or ClassC examples nothing prevents you to issue the DeviceTimeReq MAC command to just synchronize the clock.

closed 2019-04-17 by mluis1