Lora-net/LoRaMac-node #554: Request: Adding RFLR_PACONFIG_PASELECT_PABOOST to "board-config.h"

HankKauffmann, 2018-09-14:
Is it too late in the game to externalize the PASELECT option from the code and into a personalization file like board_config.h?

The rationalization for this is that it's a reminder for the developer during implementation or FW upgrades that this hardware option has to be considered. (Yes I got burned, and I'm not the only one.)

I would extend this request for all variants of circuits such as those with TCXO control of SX126X.

Hank

// changes to "board-config.h"
#define PACONFIG_PASELECT RFLR_PACONFIG_PASELECT_RFO
//#define PACONFIG_PASELECT RFLR_PACONFIG_PASELECT_PABOOST

// changes to "sx1272mb2das-board.c"
static uint8_t SX1272GetPaSelect( uint32_t channel )
{
  return PACONFIG_PASELECT;
  //return RF_PACONFIG_PASELECT_RFO;
}

HankKauffmann, 2018-09-15:
Or for better backwards compatibility ...

// changes to "sx1272mb2das-board.c"
static uint8_t SX1272GetPaSelect( uint32_t channel ) { 
#ifdef OPT_PACONFIG_PASELECT
    return OPT_PACONFIG_PASELECT; 
#else
    return RF_PACONFIG_PASELECT_RFO; 
#endif
}

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1