Lora-net/LoRaMac-node #555: Cmake Error

zirnevis, 2018-09-16:
Hi,
i am using Visual Studio Enterprise 2017 and want to use lora-net to have VS Solution for below parameters:
- LoRaMac
- classA
- REGION_EU868
- SX1276MB1MAS
- NucleoL152
I already changed CmakeLists to desired parameters but have below error while add folder to VS.
i also searched for issues related to my problem an found one that said to add 
**set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_CURRENT_LIST_DIR}/cmake" CACHE STRING "Modules for CMake" FORCE)** to my main CmakeList but still have the same problem.
// ****************************************************************************************
[cmake-server] loading initial cache file c:\users\user\documents\loramac-node-master\build\CMakeTools\InitializeCache.cmake

-- CMake Error at src/CMakeLists.txt:63 (include):
  include could not find load file:

    stm32l1

-- CMake Error at src/apps/LoRaMac/CMakeLists.txt:75 (include):
  include could not find load file:

    gdb-helper

-- CMake Error at src/apps/LoRaMac/CMakeLists.txt:76 (include):
  include could not find load file:

    binutils-arm-none-eabi

-- CMake Error at src/apps/LoRaMac/CMakeLists.txt:79 (generate_run_gdb_stlink):
  Unknown CMake command "generate_run_gdb_stlink".

-- Configuring incomplete, errors occurred!
See also "c:/users/user/documents/loramac-node-master/build/CMakeFiles/CMakeOutput.log".
[vscode] Configure failed: [cmake-server] Configuration failed.
// ****************************************************************************************
Thank you in advance
Regards

ConnyBusy, 2018-09-16:
@zirnevis 
looks like you currently have the BOARD set to LoRaMote not the desired NucleoL152(from the logs that's where it's failing). You may want to check the CMakelist.txt to make sure the BOARD is correctly set.

zirnevis, 2018-09-17:
Dear ConnyBusy
Thanks, no NucleoL152 has been selected correctly.
i double checked it by number of row that error is coming from. it shows that when compiler wants to include files related to NucleoL152 the error comes up. you cans see part of CmakeList below.
// ****************************************************************************************
elseif(BOARD STREQUAL NucleoL152)
    # Configure toolchain for NucleoL152
    set(LINKER_SCRIPT ${CMAKE_CURRENT_SOURCE_DIR}/boards/NucleoL152/cmsis/arm-gcc/stm32l152xe_flash.ld)
    include(stm32l1)
// ****************************************************************************************

ConnyBusy, 2018-09-17:
@zirnevis can you share the CmakeLists.txt file located in c:/users/user/documents/loramac-node-master/src directory. The error message you shared before doesn't point to lines you shared now. Unfortunately, I am using a Unix base OS. On my side I am able to build the binaries for your setup without any errors.

zirnevis, 2018-09-17:
Dear ConnyBusy
Here you can see CmakeLists.txt you requested before.
i think the problem is related to the path of cmake files located in **loramac-node-master/cmake** but i don't know how to fix it.
as i said before i also tried to add **set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} "${CMAKE_CURRENT_LIST_DIR}/cmake" CACHE STRING "Modules for CMake" FORCE)** to mentioned CmakeList but nothing happened.

Thanks for your quick response
Regards
[CMakeLists.txt](https://github.com/Lora-net/LoRaMac-node/files/2387585/CMakeLists.txt)

ConnyBusy, 2018-09-17:
Yes, your CMakeLists.txt is set correctly. 

Sounds like a missing path of the toolchain. You can also clear the CMake cache and see if it persists. I am not sure if you followed these instructions:
"""
For Windows platforms the prefix has to be provided anyway and additionally the CMake Generator for MinGW Makefiles has to be chosen:  
    `cmake -DCMAKE_TOOLCHAIN_FILE="cmake/toolchain-arm-none-eabi.cmake" -DTOOLCHAIN_PREFIX="**_/path/to/the/toolchain_**" -G "MinGW Makefiles" ..
"""
You may also want to make sure the tools path are added to the system path variable.

zirnevis, 2018-09-17:
I moved cmake folder to src folder and now i get this LOG:

// ****************************************************************************************
1> Command line: C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2017\ENTERPRISE\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\CMake\bin\cmake.exe  -G "Ninja" -DCMAKE_INSTALL_PREFIX:PATH="C:\Users\Reza\CMakeBuilds\013eef07-e475-b83f-aa75-7f097687c325\install\x64-Debug (default)"  -DCMAKE_CXX_COMPILER="C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.15.26726/bin/HostX64/x64/cl.exe"  -DCMAKE_C_COMPILER="C:/Program Files (x86)/Microsoft Visual Studio/2017/Enterprise/VC/Tools/MSVC/14.15.26726/bin/HostX64/x64/cl.exe"  -DCMAKE_BUILD_TYPE="Debug" -DCMAKE_MAKE_PROGRAM="C:\PROGRAM FILES (X86)\MICROSOFT VISUAL STUDIO\2017\ENTERPRISE\COMMON7\IDE\COMMONEXTENSIONS\MICROSOFT\CMAKE\Ninja\ninja.exe" "C:\Users\Reza\Desktop\LoRaMac-node-develop"
1> Working directory: C:\Users\Reza\CMakeBuilds\013eef07-e475-b83f-aa75-7f097687c325\build\x64-Debug (default)
1> Linker script: C:/Users/Reza/Desktop/LoRaMac-node-develop/src/boards/NucleoL152/cmsis/arm-gcc/stm32l152xe_flash.ld
1> -- Configuring done
1> -- Generating done
1> -- Build files have been written to: C:/Users/Reza/CMakeBuilds/013eef07-e475-b83f-aa75-7f097687c325/build/x64-Debug (default)
1> Starting CMake target info extraction ...
1> CMake server connection made.
1> Extracted includes paths.
1> Extracted CMake variables.
1> Extracted source files and headers.
1> Extracted global settings.
1> Extracted code model.
1> Extracted CTest info.
1> Collating data ...
1> Target info extraction done.
// ****************************************************************************************
i checked folder: **C:\Users\Reza\CMakeBuilds\013eef07-e475-b83f-aa75-7f097687c325\install\x64-Debug (default)** but no project was there. just found some empty folders like my source folder (you can see pic in attachment)
[image]
[image]
i found in [**http://preshing.com/20170511/how-to-build-a-cmake-based-project/**](url) that in Visual Studio 2017 CMake is not creating project anymore (you can read more in **Built-In CMake Support in Visual Studio 2017** title of previous URL)
what does it mean? is it working fine? where can i find my solution?

i also tried to install CMake itself separately from Visual Studio but now i get previous ERRORs that already solved in VS2017 **(i mean stm32l1, gdb-helper, binutils-arm-none-eabi and ...)**

no idea more and so tired of this...
Regards

ConnyBusy, 2018-09-19:
@zirnevis looks like moving you CMake folder helped. 
To get the binaries you need to look further i.e. x64Debug/src/apps/loramac/ that's where the binaries are located. 
Note: After running cmake you need to run make as well to generate the binaries

zirnevis, 2018-09-23:
Hi ConnyBusy
I really appreciate for your collaboration
Here you can see requested file.
Regards
[CMakeCache.txt](https://github.com/Lora-net/LoRaMac-node/files/2408503/CMakeCache.txt)

ConnyBusy, 2018-09-23:
Hi Zernevis,
I am not quite sure if you have installed the following and added them to your system path variable. I don't seem to see any references to them in your CMakeCache.txt. If not. Please install themand/or add the paths to path variable:

* [GNU Arm Embedded Toolchain](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm)
    * The Make utility is also required, one can use e.g. [MSYS2](http://www.msys2.org) or [MinGW](http://mingw.org/)
        * With MSYS2:
            * Follow the installation instructions provided by them.
            * Install MinGW Make with `pacman -S mingw-w64-x86_64-make`.

     **Note**: Please ensure that both paths are added to the system PATH variable. Add for example: `C:\msys64\mingw64\bin` and `C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q2-update\bin` to `Path` variable.

zirnevis, 2018-10-10:
i didn't install required applications fine.
now it is working

closed 2018-10-10 by zirnevis

CHETAN057, 2018-10-12:
Hi Zirnevis,

I am facing same issue for eclipse configurations, can you provide me solution of this issue?

Which file could be missing ?

Thanks in advance
Chetan

sagar-patel-sls, 2019-03-12:
Hii, @zirnevis @CHETAN057 

Hi,
i am using VSCODE and want to use lora-net to have VS Solution for below parameters:

LoRaMac
classA
REGION_EU433
SX1276MB1MAS
NucleoL152

I am working on LoRaMAC V1.1 and i downloaded **`feature/5.0.0 branch`**
i have configured with my board and generate **CMake** for VSCODE but i got below error while debugging.
```
###### ===== MLME-Confirm ==== ######
STATUS      : Tx timeout
```
[image]

**As per my opinion**
In our case Application Layer send packet on sx1276 but sx1276 not transmit packet Over the Air.

Can you please help me..!
Thank you