Lora-net/LoRaMac-node #556: reusing same DevNonce problem

gpgib6, 2018-09-17:
It appears the DevNonce always starts with 0 and increments
CryptoCtx.NvmCtx->DevNonce++

     * Device nonce is a counter starting at 0 when the device is initially
     * powered up and incremented with every JoinRequest.

On the other hand the network server (TTN) seems to be telling me that this has to be a random number (?) and can't repeat.
Once I add a device, each join takes longer as the previously used nonces get ignored, but if I manually change it to a new value, then the join works out immediately.

The LoRa spec has this to say:
DevNonce is a counter starting at 0 when the device is initially powered up and incremented with every Join-request. A DevNonce value SHALL NEVER be reused for a given JoinEUI value. If the end-device can be power-cycled then DevNonce SHALL be persistent (stored in a non-volatile memory). Resetting DevNonce without changing JoinEUI will cause the Network Server to discard the Join-requests of the device.

I don't know if that means a network server stores every single nonce ever used.

nestorayuso, 2018-09-17:
DevNonce is a random number in LoRaWAN specification 1.0.X
DevNonce is a incremental number starting 0 in LoRaWAN specification 1.1

Current TTN server implements 1.0.X only.

master and develop branches implements 1.0.X
feature/5.0.0 branch https://github.com/Lora-net/LoRaMac-node/tree/feature/5.0.0 implements both 1.0.X and 1.1

if you are using feature/5.0.0 branch you can select here https://github.com/Lora-net/LoRaMac-node/blob/feature/5.0.0/src/apps/LoRaMac/classA/NucleoL152/Commissioning.h#L71 the version of the server you want to connect to.

gpgib6, 2018-09-18:
If I'm not mistaken, the AppEUI has been renamed to JoinEUI, so it doesn't normally change. But the 1.1 spec still says that the DevNonce can never be repeated (except when the device is given a new JoinEUI which usually doesn't happen).
Perhaps this means that all 1.1 spec devices must have non-volatile memory. I guess what we will need is this:
Read DevNonce from eeprom
CryptoCtx.NvmCtx->DevNonce++
Save DevNonce to eeprom

Since a loss of power or reset could happen at any time.
Does that sound right?

nestorayuso, 2018-09-18:
you are right, you need non-volatile memory if you cut the power.

NvmCtx stands for non-volatile memory context and the implementation is ongoing.

gpgib6, 2018-09-18:
Gotcha. For now I will just use the random number because most (if any) network servers aren't at 1.1 yet anyways.
In my case I'm using STM32L433 which doesn't have internal eeprom, other people will find the same issue. At least the joinreq doesn't happen that often so perhaps it's ok to use flash memory.
For those in a similar situation there is a library with wear-leveling to emulate eeprom on flash:
https://www.st.com/en/embedded-software/x-cube-eeprom.html

ZcHoil, 2018-10-10:
hi @gpgib6 ,I seem to have had similar problems.But I've solved the problem.  My nodes use the LoRaWAN specification 1.0.x.  In OnMacStateCheckTimerEvent this function，I changed ScheduleTx to OnTxDelayedTimerEvent.  Now he seems to be working normally, I don't know if it is the same as your problem, I hope it can help you

gpgib6, 2018-10-10:
hi @ZcHoil, what I did was to make it a random number again. So in my case it looks like this (inside function LoRaMacCryptoPrepareJoinRequest):

    // Add device nonce
//..!    CryptoCtx.NvmCtx->DevNonce++;

    SecureElementRandomNumber(&temp);
    CryptoCtx.NvmCtx->DevNonce = (uint16_t) temp;

Either way it's a bandaid for now to continue development.

ZcHoil, 2018-10-11:
hi @gpgib6  ,You are only working as a temporary solution, I don't know if your lorawan version is the latest, because my lorawan version is older, because in my code I found the OnMacStateCheckTimerEvent function in the network. Will send the same devnonce repeatedly instead of randomly generating again, so this problem is caused. Can you show me how your OnMacStateCheckTimerEvent function is implemented?

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1