Lora-net/LoRaMac-node #557: Does LoRaMacSerializer.c distinguish big-endian or small-endian?

wazh012, 2018-09-19:
Hi, There is a question that confuses me in the  file LoRaMACSerialzer.c. Those codes may result in different contents when they are run in different device which MCU is big-endian or small-endian.  
E.g., MIC=0x12345678, it will be sent as "0x12 0x34 0x56 0x78" in pkts by order for big-endian device, while it will be sent as "0x78 0x56 0x34 0x12" by order for small-endian device.
I think pkts should be use network order, what do you think?

mluis1, 2018-09-21:
We think that the `LoRaMacSerializer.c` and `LoRaMacParser.c` functions are implemented in such way that the MCU endianness isn't relevant.

For instance the `LoRaMacParser.c` functions fill the structure fields byte by byte. The compiler takes care of the endianness management.

Example:
```c
LoRaMacParserStatus_t LoRaMacParserData( LoRaMacMessageData_t* macMsg )
{
...
    macMsg->MIC = ( uint32_t ) macMsg->Buffer[( macMsg->BufSize - LORAMAC_MIC_FIELD_SIZE )];
    macMsg->MIC |= ( ( uint32_t ) macMsg->Buffer[( macMsg->BufSize - LORAMAC_MIC_FIELD_SIZE ) + 1] << 8 );
    macMsg->MIC |= ( ( uint32_t ) macMsg->Buffer[( macMsg->BufSize - LORAMAC_MIC_FIELD_SIZE ) + 2] << 16 );
    macMsg->MIC |= ( ( uint32_t ) macMsg->Buffer[( macMsg->BufSize - LORAMAC_MIC_FIELD_SIZE ) + 3] << 24 );
...
}
```

The `LoRaMacSerializer.c` function do the reverse operation of the `LoRaMacParser.c`.

Example:
```c
LoRaMacSerializerStatus_t LoRaMacSerializerData( LoRaMacMessageData_t* macMsg )
{
...
    macMsg->Buffer[bufItr++] = macMsg->MIC & 0xFF;
    macMsg->Buffer[bufItr++] = ( macMsg->MIC >> 8 ) & 0xFF;
    macMsg->Buffer[bufItr++] = ( macMsg->MIC >> 16 ) & 0xFF;
    macMsg->Buffer[bufItr++] = ( macMsg->MIC >> 24 ) & 0xFF;
...
}
```

wazh012, 2018-09-26:
Hi mluis1,

thanks for the report.  

Yes, I think may be you are right.

closed 2018-09-26 by wazh012