Lora-net/LoRaMac-node #558: Repetative mac commands by stack

MaheshKumarPN, 2018-09-20:
I see a repeating mac ans when receiving mac requests
When i debugged into stack mac commands are multiples twice every time  in `ParseMacCommandsToRepeat`
I don't understand why we repeat the same command!!
This is observed when `LoRaMacMcpsRequest()` is called several times,  duty cycle restricted error is received and at some point of time data is sent out .
In my understanding there is something  wrong in handing 
 `MacCommandsBufferIndex = 0;`
and 
`MacCommandsBufferToRepeatIndex`
Which means index should handled bettter
With the understand i have,I updated the code as below
`

            if( ( payload != NULL ) && ( LoRaMacTxPayloadLen > 0 ) )
            {
                LoRaMacBuffer[pktHeaderLen++] = framePort;

                    MacCommandsBufferIndex = 0;
                if( framePort == 0 )
                {
                    // Reset buffer index as the mac commands are being sent on port 0
                    LoRaMacPayloadEncrypt( (uint8_t* ) payload, LoRaMacTxPayloadLen, LoRaMacNwkSKey, LoRaMacDevAddr, UP_LINK, UpLinkCounter, &LoRaMacBuffer[pktHeaderLen] );
                }
                else
                {
                    LoRaMacPayloadEncrypt( (uint8_t* ) payload, LoRaMacTxPayloadLen, LoRaMacAppSKey, LoRaMacDevAddr, UP_LINK, UpLinkCounter, &LoRaMacBuffer[pktHeaderLen] );
                }
            }
`
Looks like working...
please help me out on this..

catsup, 2018-10-23:
Hello Mahesh,

do you have a way to consistently reproduce the problem ?

I'm asking as I have seen similar duplication of sticky mac commands with 4.4.0, after having received a RxParamSetupReq changing the datarate, with the RxParamSetupAns being lost and the network continuing to send downlinks using the old datarate despite receiving the subsequent answers. These downlinks are of course never received by the device, which keeps sending the RxParamSetupAns - and from time to time doubles the MAC payload.

While the fact that the RxParamSetupAns was ignored has been traced back to an issue with the network, the device should not double the sticky commands either and it would probably be good to address that issue in the MAC.

XKottelaar, 2018-10-26:
It's a known issue, the MacCommandsBufferIndex is sometimes not being reset correctly seems to be happening during a RX_ABORT.

When the index isn't reset and a new message is prepared the commands that needs to be repeated are added after the allready repeated Mac commands. The buffer then gets then gets scanned for commands that need to be repeated next time and voila the commands are duplicated.

https://github.com/Lora-net/LoRaMac-node/issues/492

closed 2019-02-12 by mluis1