Lora-net/LoRaMac-node #559: why is there a macCommand number limit

f2ghazi, 2018-09-27:
I have tried to find an explanation in the LoRaWan 1.0.2 but I can't find any. so my question is concerning this line:
uint8_t maxCommands = RegionGetMaxMacCommands(LoRaMacRegion);
why is there a max number of macCommands. And why is it different between the US and the other regions.
Is it just a protection from processing more than 11bytes of sticky macCommandAns and blocking the device?
Thank you for the clarifications.

mluis1, 2018-09-27:
On which branch did you find the `RegionGetMaxMacCommands` function?

I have searched in `master`, `develop` and `feature/5.0.0` and this function doesn't exist or ever existed if I'm not mistaken.

The only functions that start with `RegionGet` are:
* `RegionGetPhyParam` - all branches.
* `RegionGetNvmCtx` - `develop` and  `feature/5.0.0` branches only

f2ghazi, 2018-10-01:
My mistake. After further investigation, it seems that our team is the one who introduced this behaviour in order to avoid blocking the device if it has to send more than 11bytes of sticky mac commands answer.
Sorry for the delay. Issue closed.

closed 2018-10-01 by f2ghazi

mluis1, 2020-03-17:
The commit wrongly reference this issue. It should have referenced #569 issue instead.