Lora-net/LoRaMac-node #560: Reading RTC RtcGetCalendarValue()

Ruben1992, 2018-09-28:
Currenly I am porting this code to a board I am working on. I am not sure if the bug is also on the demo boards like "sx1261dvk1bas", but I cannot test that because I don't have those hardware boards. 

On the RTC-board.c files (for STM boards), we have the following code:
```
static uint64_t RtcGetCalendarValue( RTC_DateTypeDef* date, RTC_TimeTypeDef* time )
{
  ... 
    HAL_RTC_GetTime( &RtcHandle, time, RTC_FORMAT_BIN );

    // Make sure it is correct due to asynchronus nature of RTC
    do
    {
        firstRead = time->SubSeconds;
        HAL_RTC_GetDate( &RtcHandle, date, RTC_FORMAT_BIN );
        HAL_RTC_GetTime( &RtcHandle, time, RTC_FORMAT_BIN );
    }while( firstRead != time->SubSeconds );
```

Reading from the RTC register requires a few tricks (especially if you are directly reading from the clock registers, and not from the shadow registers). 
Here is a quote from ST:
> You must call HAL_RTC_GetDate() after HAL_RTC_GetTime() to unlock the values in the higher-order calendar shadow registers to ensure consistency between the time and date values. Reading RTC current time locks the values in calendar shadow registers until Current date is read.

So what I would expect is the getTime, and then getDate afterwards, because otherwise the rtc would be frozen. 
And also the first getTime() is used for making sure that while reading the registers the time did not change. However, again see quote, after reading the time, you should read the date also.
So what I should look like the following:

```
    // Get Time and Date
    HAL_RTC_GetTime( &RtcHandle, time, RTC_FORMAT_BIN );
    HAL_RTC_GetDate( &RtcHandle, date, RTC_FORMAT_BIN );

    // Make sure it is correct due to asynchronus nature of RTC
    do
    {
        firstRead = time->SubSeconds;
        HAL_RTC_GetTime( &RtcHandle, time, RTC_FORMAT_BIN );
        HAL_RTC_GetDate( &RtcHandle, date, RTC_FORMAT_BIN );
    }while( firstRead != time->SubSeconds );
```
However this piece of code is used quite frequently, So I find it strange that Im the first/only one having this problem.

Regimbal, 2018-10-17:
Your proposition makes sense to me, but can you tell what behavior or issue this resulted in in the first place? Did it solve it?
We had quite a lot of issues with previous versions of timers, where HAL_RTC_Getxxx where implemented the proper way!

Ruben1992, 2018-10-18:
Its quite a while back, so I dont exactly remember the details of the problem, just that it didnt worked and now it does.
but what I think what happens is, is that if we would read the time registers (with the old code). Then the timer values are locked until we read it again. So if you read the time a second time, you get the time of the previous moment that you tried to read it. And you will lock the time values again. 
So while calling RtcGetCalendarValue(), you dont get the time, but you get the timestamp of the previous time that you called RtcGetCalendarValue().
If you have a lot of timers elapsing soon after each other, you wont really notice any difference. However the way i tested my timers was by starting just one timer somewhere in my main and after that letting my main run a while(1); loop. So RtcGetCalendarValue() is only called once. 

"We had quite a lot of issues with previous versions of timers, where HAL_RTC_Getxxx where implemented the proper way!"
Just out of interest, Could you give me an example?

I just patched my lora Firmware so it matches v4.4.2-rc.3, the previous version I believe was 4.3.2 or 4.4.0. But i remember having the same issue with that firmware, (only then i thought it was my mistake, so I didnt post anything about it). So im not sure in what firmware release this bug first occurs, but I think its been there for quite some time.

mluis1, 2018-10-18:
@Ruben1992 You are right I remember seeing on documents provided by ST that we should read the RTC values as you describe.

Please note that the current implementation of the RTC driver code has been done by the ST MCD Application Team. Thus, we relied on their implementation.

On our side we have tested this code on different ST platforms and so far we didn't saw issues with it. But, just to be sure we will ask ST if the current implementation is correct or not.

Looking at the ST "en.i-cube_lrwan 1.2.0" library the RTC driver implementation is done in the same way as on this project.

```c
/*!
 * @brief get current time from calendar in ticks
 * @param pointer to RTC_DateStruct
 * @param pointer to RTC_TimeStruct
 * @retval time in ticks
 */
static uint64_t HW_RTC_GetCalendarValue( RTC_DateTypeDef* RTC_DateStruct, RTC_TimeTypeDef* RTC_TimeStruct )
{
  uint64_t calendarValue = 0;
  uint32_t first_read;
  uint32_t correction;
  uint32_t seconds;
  
  /* Get Time and Date*/
  HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
 
   /* make sure it is correct due to asynchronus nature of RTC*/
  do {
    first_read = RTC_TimeStruct->SubSeconds;
    HAL_RTC_GetDate( &RtcHandle, RTC_DateStruct, RTC_FORMAT_BIN );
    HAL_RTC_GetTime( &RtcHandle, RTC_TimeStruct, RTC_FORMAT_BIN );
  } while (first_read != RTC_TimeStruct->SubSeconds);
 
  /* calculte amount of elapsed days since 01/01/2000 */
  seconds= DIVC( (DAYS_IN_YEAR*3 + DAYS_IN_LEAP_YEAR)* RTC_DateStruct->Year , 4);

  correction = ( (RTC_DateStruct->Year % 4) == 0 ) ? DAYS_IN_MONTH_CORRECTION_LEAP : DAYS_IN_MONTH_CORRECTION_NORM ;
 
  seconds +=( DIVC( (RTC_DateStruct->Month-1)*(30+31) ,2 ) - (((correction>> ((RTC_DateStruct->Month-1)*2) )&0x3)));

  seconds += (RTC_DateStruct->Date -1);
  
  /* convert from days to seconds */
  seconds *= SECONDS_IN_1DAY; 

  seconds += ( ( uint32_t )RTC_TimeStruct->Seconds + 
             ( ( uint32_t )RTC_TimeStruct->Minutes * SECONDS_IN_1MINUTE ) +
             ( ( uint32_t )RTC_TimeStruct->Hours * SECONDS_IN_1HOUR ) ) ;

  
  calendarValue = (((uint64_t) seconds)<<N_PREDIV_S) + ( PREDIV_S - RTC_TimeStruct->SubSeconds);

  return( calendarValue );
}
```

Regimbal, 2018-10-18:
@Ruben1992 our issues were mainly due to the previous implementation of the timers handling without subseconds handling. The thing is, on a batch of ~200 sensors, around one half stopped working after 18 days (precisely) the others after 40 or something (can't remember but all in the same day) and a few continued to work.
We tried to 'reset' the calendar periodically but without success, periodical sending (12h period) worked 2 days then nothing again...

@mluis1 You are right, I did exactly like you when I read @Ruben1992 suggestion. I went to check on ST LoRa Cube extension package 1.2... But could not determine which team actually coded this.

All in all, thanks mluis to ask to ST. Indeed this recommandation still appears on the HAL documentation

mdoreille, 2018-10-18:
On the en.i-cube_lrwan_V1.2 the RTC Shadow register is disabled in the function HW_RTC_SetConfig

/*Enable Direct Read of the calendar registers (not through Shadow) */
 HAL_RTCEx_EnableBypassShadow(&RtcHandle);

mluis1, 2018-10-18:
Thanks @mdoreille on this project we also have this function call on all provided ST platforms.

```c
        // Enable Direct Read of the calendar registers (not through Shadow registers)
        HAL_RTCEx_EnableBypassShadow( &RtcHandle );
```

@Ruben1992 Can we close the issue?

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1