Lora-net/LoRaMac-node #561: LoRaMAC 4.4.1 Ack Time Out behavior - EU868

AlexL871, 2018-10-01:
following the advice of the forum "the things network", I ask my question here.

BenAtPip, 2018-10-01:
Would you be able to expand on how this issues appears to you in your network backend? Does your device incorrectly service some downlink frames and not others?

AlexL871, 2018-10-04:
it seems it only happens in SF12 and with a long downlink. 
The ackTimeOutTimer is triggered too early and in "OnMacStateCheckTimerEvent" when the stack does this test :
```
    if( LoRaMacFlags.Bits.McpsInd == 1 )
    {// Procedure if we received a frame 
      if( ( McpsConfirm.AckReceived == true ) || ( AckTimeoutRetriesCounter > AckTimeoutRetries ) )
      {
        AckTimeoutRetry = false;
        NodeAckRequested = false;
        if( IsUpLinkCounterFixed == false )
        {
          UpLinkCounter++;
        }
        McpsConfirm.NbRetries = AckTimeoutRetriesCounter;
        
        LoRaMacState &= ~LORAMAC_TX_RUNNING;
      }
    }
```
McpsConfirm.AckReceived is false, thus LoRaMACState stays in"LORAMAC_TX_RUNNING" and blocks all other activities. 

Often, This case appears with the first downlink. 

Thanks.

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1

trick2011, 2019-03-11:
We have confirmed the issue when sending a SF12 uplink that the Ack timeout can occur too soon. When fudging the code a bit to go to the absolute minimum of the ACK_TIMEOUT value it occurs consistently. This is with the EU868 region settings.

> Consequently, the LoRaMAC state stay in TxRunning and block the processing.

trick2011, 2019-03-15:
@AlexL871 https://github.com/Lora-net/LoRaMac-node/commit/31df7ec88962c9ce646b1f99899388533f5d2c3c this commit at least allows the detection of a deadlock by polling the loramac state. But this is still a user level solution.