Lora-net/LoRaMac-node #563: ANTANNA_GAIN

GrumpyOldPizza, 2018-10-02:
Ok, this one is confusing. So, EU868, wikipedia:

EIRP(dBw) = Ptx(dBw) + G(dBi)

Hence:

Ptx(dBw) = (MaxEIRP(dBw) - (2 * TxPower)) - G(dBi)

This is what the code in RegionCommonComputeTxPower() does, with a default ANTENNA_GAIN of 2.15.

Now looking at US915, wikipedia:

ERP(dBw) = Ptx(dBw) + G(dBi) - 2.15

Hence:

Ptx(dBw) = (MaxERP(dBw) - (2 * TxPower)) - G(dBi) + 2.15

Now the code for US915 uses this here to call RegionCommonComputeTxPower():

phyTxPower = RegionCommonComputeTxPower( txPowerLimited, US915_DEFAULT_MAX_ERP, 0 );

That seems to be correct for the default antenna gain of 2.15. However if you have a different antenna, shouldn't this be:

phyTxPower = RegionCommonComputeTxPower( txPowerLimited, US915_DEFAULT_MAX_ERP, ( txConfig->AntennaGain - 2.15 ) );

So it kind of boils down to 2 questions:

(1) The ANTENNA_GAIN as implemented is specified in terms of "dBi" ?

(2) The ANTENNA_GAIN as implemented is incorrect for US915 (and other regions using ERP).

catsup, 2018-10-29:
To me the confusion appears to be in the naming, as the value 2.15 is not the _antenna gain_, but a constant value to be added to convert an ERP value (in dB) to an EIRP value. 

Antenna gain would be more the actual path loss/gain between the tx power programmed into the radio chip, and the measured antenna output power, which depends on the rf path and antenna type.

GrumpyOldPizza, 2018-10-29:
I think that was my point. The 2.15 should be used as an offset in addition
to the antenna gain.

mluis1, 2018-10-30:
The `antennaGain` variable is referenced to the isotropic antenna. The value must be in [dBi].
To compute the `antennaGain` value the following formula should be used `antennaGain[dBi] = measuredAntennaGain[dBd] + 2.15`

Some computation examples for the `antennaGain` variable:

Feed power = 14 dBm
Peak antenna gain = 0 dBi
MaxERP = 14 - 2.15 ->  ANTENNA_GAIN = 2.15[dBi]

Feed power = 16 dBm
Peak antenna gain = 3 dBi
MaxERP = 16 - 5.15 -> ANTENNA_GAIN = 2.15 + 3[dBi] = 5.15[dBi]

In order to clarify things we propose to change the `RegionCommonComputeTxPower` function documentation as shown below.

```c
/*!
 * \brief Computes the txPower, based on the max EIRP and the antenna gain.
 * 
 * \remark US915 region uses a conducted power as input value for maxEirp.
 *         Thus, the antennaGain parameter must be set to 0.
 *
 * \param [IN] txPower TX power index.
 *
 * \param [IN] maxEirp Maximum EIRP.
 *
 * \param [IN] antennaGain Antenna gain. Referenced to the isotropic antenna.
 *                         Value is in dBi. ( antennaGain[dBi] = measuredAntennaGain[dBd] + 2.15 )
 *
 * \retval Returns the physical TX power.
 */
int8_t RegionCommonComputeTxPower( int8_t txPowerIndex, float maxEirp, float antennaGain );
```

mluis1, 2018-10-30:
Thanks, yes, you are right. We will update the documentation accordingly.

catsup, 2018-10-30:
There still is an inconsistency with the way the US915 MAC uses the function, as RegionCommonComputeTxPower takes an EIRP value, not an ERP:

`int8_t RegionCommonComputeTxPower( int8_t txPowerIndex, float maxEirp, float antennaGain );`

`phyTxPower = RegionCommonComputeTxPower( txPowerLimited, US915_DEFAULT_MAX_ERP, ( txConfig->AntennaGain - 2.15 ) );
`

mluis1, 2018-10-30:
We wouldn't like to implement a different function just for the US915 region.

As previously said, to clarify things we will update the function documentation in order to say that for US915 region the maxEirp parameter takes the conducted power which is `US915_DEFAULT_MAX_ERP`

```c
/*
...
 * \remark US915 region uses a conducted power as input value for maxEirp.
 *         Thus, the antennaGain parameter must be set to 0.
...
 */
```

The US915 region **doesn't use EIRP or ERP** values it uses a **conducted power** value and no antenna gain correction must be applied.

The things we could eventually change is `US915_DEFAULT_MAX_ERP` definition to `US915_DEFAULT_MAX_CONDUCTED_POWER` and `maxEirp` function parameter to `maxTxPower`if it is more clear for you.

Except these changes, which, are just names change, everything is implemented as specified.

catsup, 2018-10-30:
I'm not suggesting to implement a different function for the US, on the contrary - I'd like things to be consistent and unified across the board, but maybe I didn't express myself well enough.

What I was trying to suggest was to call the RegionCommonComputeTxPower() with the same type of parameters as for other regions, namely, passing an EIRP  value and the proper antenna gain, rather than kludging it by passing an ERP and a 0 antenna gain.

Pushing the idea further, and the conversion being straightforward, why not use EIRP _in the code_ for all regions for consistency sake, even if the specifications are talking about ERP or conducted power for the US ? Ie, replace _#define US915_DEFAULT_MAX_ERP 30.0f_ with _#define US915_DEFAULT_MAX_EIRP 32.0f_ and take it from there ?

Just my .02€...

closed 2018-10-31 by mluis1