Lora-net/LoRaMac-node #565: about class c mode join

ZcHoil, 2018-10-08:
hi，
I have tuned the class A mode, including otaa, and everything looks fine.But when I switch to class c mode, I find that the node can't enter the network normally.Debug found that rxtimeout always appears. I want to know what the conditions for triggering rxtimerout are. Before that, I have confirmed that the server has received the incoming network message and correctly agreed to the network.
thanks!

ZcHoil, 2018-10-09:
hello?
Can someone give me a little help   =.=

ConnyBusy, 2018-10-09:
@ZcHoil. Firstly, what do you mean by "I have tuned class A mode, ...".  Which code branch are you using?
That not withstanding: All devices join any network as Class A and then switch to either Class B or C via appropriate MAC commands exchange initiated by the device. Does the server you use support class C?

ZcHoil, 2018-10-10:
hi @ConnyBusy .First of all, thank you for your answer. 
I can confirm that my server can support the class C mode.Because I can use the latest code to test class C to work properly.This problem occurred in a version a few years ago. Since the latest banch has been committed more than 2,000 times, I have no idea which version this is from.However, I found that compared to the latest code, this older version has a lot less structure, similar to the fact that in the new version rxwindow has a structure called LoRaMacRxSlot_t, and in the old version only RxSlot=1 or RxSlot=0, which means there is no RX_SLOT_WIN_CLASS_C, and so on

ZcHoil, 2018-10-10:
and this is a little bit of code
`
static void OnRadioTxDone( void )
{
    TimerTime_t curTime = TimerGetCurrentTime( );
	
    if( LoRaMacDeviceClass != CLASS_C )
    {
        Radio.Sleep( );
    }
    else
    {
        OnRxWindow2TimerEvent( );
    }
	
    /* Store last Tx channel */
    LastTxChannel = Channel;
	
    /* Update last tx done time for the current channel */
    Bands[Channels[LastTxChannel].Band].LastTxDoneTime = curTime;
	
    /* Update Aggregated last tx done time */
    AggregatedLastTxDoneTime = curTime;

    if( IsRxWindowsEnabled == true )
    {
LOGD("RxWindowTimer1=%d,RxWindow2Delay=%d\r\n" , RxWindowTimer1,RxWindow2Delay);
		if(0 != RxWindow1Delay)
		{
        TimerSetValue( &RxWindowTimer1, RxWindow1Delay );
        TimerStart( &RxWindowTimer1 );
		}
		
		if( LoRaMacDeviceClass != CLASS_C )
        {	
			if(0 != RxWindow2Delay)
			{
                TimerSetValue( &RxWindowTimer2, RxWindow2Delay );
                TimerStart( &RxWindowTimer2 );
			}
        }
        if( ( LoRaMacDeviceClass == CLASS_C ) || ( NodeAckRequested == true ) )
        {
            TimerSetValue( &AckTimeoutTimer, /*RxWindow2Delay*/(RxWindow1Delay + 1000) + ACK_TIMEOUT + randr( -ACK_TIMEOUT_RND, ACK_TIMEOUT_RND ) ); 

            TimerStart( &AckTimeoutTimer );
        }
    }
    else
    {
        McpsConfirm.Status = LORAMAC_EVENT_INFO_STATUS_OK;
        MlmeConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT;

        if( LoRaMacFlags.Value == 0 )
        {
            LoRaMacFlags.Bits.McpsReq = 1;
        }
        LoRaMacFlags.Bits.MacDone = 1;
    }

    if( NodeAckRequested == false )
    {
        McpsConfirm.Status = LORAMAC_EVENT_INFO_STATUS_OK;
        ChannelsNbRepCounter++;
    }
}
`

`
static void OnRadioRxTimeout( void )
{
LOGD("Rx Timeout! RxSlot= %d\r\n",RxSlot);
    if( LoRaMacDeviceClass != CLASS_C )
    {
        Radio.Sleep( );
    }
    else
    {
        OnRxWindow2TimerEvent( );
    }

    if( RxSlot == 1 )
    {
        if( NodeAckRequested == true )
        {
            McpsConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT;
        }
        MlmeConfirm.Status = LORAMAC_EVENT_INFO_STATUS_RX2_TIMEOUT;
        LoRaMacFlags.Bits.MacDone = 1;
    }
}
`

ConnyBusy, 2018-10-10:
Hi ZcHoil,
if I understood you well, when you use the latest code it works. What is the rationale for going back to older code? It's advisable to use the latest code branches, especially, if you intend to certify your product.
There are some recommended practices that have been defined by the technical committee which have been implemented in the latest code branches.

ZcHoil, 2018-10-11:
Hi @ConnyBusy Yes, the latest code is working fine, but the old protocol stack is somewhat different from the latest one, but the old project still needs to be maintained, so I can't discard it for a while, but I probably found the problem at the moment and found that When RX1 was set, it was interrupted by other interrupts.I probably know how to solve it, thank you for your help.

closed 2019-02-12 by mluis1