Lora-net/LoRaMac-node #566: when FCntUp > 65535 server recieve data error

ly1243667342, 2018-10-09:
i tested  a week, when FCntUp > 65535 , server receive data error, sometimes appskey changed. why?

reissjason, 2018-10-09:
Only 2-bytes of the 32-bit uplink counter are sent in the packet.
After 65535 the 2-byte FCNT will rollover to 0 in the uplink packets sent.

If the server does not rollover the 16 MSB bits of the 32-bit counter the MIC will not be calculated properly. Is there a server option for 16 or 32 bit uplink counters?

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1

outely, 2019-04-28:
@ly1243667342  @reissjason 
Hi,I think I'm having this problem too.I tested node for more than ten days and it is normal.But I view lora server log error today.
 _level=error msg="processing uplink frame error" data_base64=QLAcSQGAjQ3Urx2b error="get device-session error: device-session does not exist or invalid fcnt or mic"_
Do you solve it?

fahad80, 2019-05-01:
@outely It is probably a GW or NS issue. Not node. I tested this couple of GWs and NSs. Never had an issue rolling over 65535. So, check those.