Lora-net/LoRaMac-node #567: LoRaMainLoop in LoRaMac-node-4.4.1 (Master Branch)

zirnevis, 2018-10-09:
Hi,
i am using LoRaMac-node-4.4.1 with below settings:
- LoRaMac
- classA
- REGION_EU868
- SX1276MB1MAS
- NucleoL152
i can compile the code using visual studio 2017 (Visual GDB extension) and also i am able to read sensors and send data to cloud. **the problem is when i send first package i am not able to send anymore**.
as i understood MacState is **BUSY** and it doesn't change to **IDLE** because no function is called to do so in main.c file (Functions like main loop of LoRa that checks DIOs and change states of MAC). 
in main.c file we have just a function **Radio.IrqProcess(); (Process Radio IRQ)** that we have this in radio.h header file but for SX1276 radio we do not have any radio.c file (because it is located in sx126x folder and as you can see below it is not related to SX1276 radio at all **(you can see routine of function below)**)
i also checked the **Developer Branch** and find out there is a function called **LoRaMacProcess( )** that handles this issue but we don't have this in master branch of loRa-Net.
				
the questions are:
1. what is the reason of sending just one data 
2. do i have to use radio.c file (it is located in sx126x folder)
3. where is main loop of LoRa that handles my issue

you can download all the files that i used for my project below (i deleted unused files in folders):
[LoRaMac-node-4.4.1.zip](https://github.com/Lora-net/LoRaMac-node/files/2459700/LoRaMac-node-4.4.1.zip)

````
void RadioIrqProcess( void )
{
    if( IrqFired == true )
    {
        BoardDisableIrq( );
        IrqFired = false;
        BoardEnableIrq( );
```

        uint16_t irqRegs = SX126xGetIrqStatus( );
        SX126xClearIrqStatus( IRQ_RADIO_ALL );

        if( ( irqRegs & IRQ_TX_DONE ) == IRQ_TX_DONE )
        {
            TimerStop( &TxTimeoutTimer );
            if( ( RadioEvents != NULL ) && ( RadioEvents->TxDone != NULL ) )
            {
                RadioEvents->TxDone( );
            }
        }

        if( ( irqRegs & IRQ_RX_DONE ) == IRQ_RX_DONE )
        {
            uint8_t size;

            TimerStop( &RxTimeoutTimer );
            SX126xGetPayload( RadioRxPayload, &size , 255 );
            SX126xGetPacketStatus( &RadioPktStatus );
            if( ( RadioEvents != NULL ) && ( RadioEvents->RxDone != NULL ) )
            {
                RadioEvents->RxDone( RadioRxPayload, size, RadioPktStatus.Params.LoRa.RssiPkt, RadioPktStatus.Params.LoRa.SnrPkt );
            }
        }

        if( ( irqRegs & IRQ_CRC_ERROR ) == IRQ_CRC_ERROR )
        {
            if( ( RadioEvents != NULL ) && ( RadioEvents->RxError ) )
            {
                RadioEvents->RxError( );
            }
        }

        if( ( irqRegs & IRQ_CAD_DONE ) == IRQ_CAD_DONE )
        {
            if( ( RadioEvents != NULL ) && ( RadioEvents->CadDone != NULL ) )
            {
                RadioEvents->CadDone( ( ( irqRegs & IRQ_CAD_ACTIVITY_DETECTED ) == IRQ_CAD_ACTIVITY_DETECTED ) );
            }
        }

        if( ( irqRegs & IRQ_RX_TX_TIMEOUT ) == IRQ_RX_TX_TIMEOUT )
        {
            if( SX126xGetOperatingMode( ) == MODE_TX )
            {
                TimerStop( &TxTimeoutTimer );
                if( ( RadioEvents != NULL ) && ( RadioEvents->TxTimeout != NULL ) )
                {
                    RadioEvents->TxTimeout( );
                }
            }
            else if( SX126xGetOperatingMode( ) == MODE_RX )
            {
                TimerStop( &RxTimeoutTimer );
                if( ( RadioEvents != NULL ) && ( RadioEvents->RxTimeout != NULL ) )
                {
                    RadioEvents->RxTimeout( );
                }
            }
        }

        if( ( irqRegs & IRQ_PREAMBLE_DETECTED ) == IRQ_PREAMBLE_DETECTED )
        {
            //__NOP( );
        }

        if( ( irqRegs & IRQ_SYNCWORD_VALID ) == IRQ_SYNCWORD_VALID )
        {
            //__NOP( );
        }

        if( ( irqRegs & IRQ_HEADER_VALID ) == IRQ_HEADER_VALID )
        {
            //__NOP( );
        }

        if( ( irqRegs & IRQ_HEADER_ERROR ) == IRQ_HEADER_ERROR )
        {
            TimerStop( &RxTimeoutTimer );
            if( ( RadioEvents != NULL ) && ( RadioEvents->RxTimeout != NULL ) )
            {
                RadioEvents->RxTimeout( );
            }
        }
    }
}`

mluis1, 2018-10-10:
As you know in EU868 region the transmission duty-cycle is enforced. This means that when sending a frame the MAC layer may prevent you to send another frame up until the duty-cycle restriction allows it again. In the version you are using the MAC layer reports that it is busy (in develop branch the MAC layer now reports `LORAMAC_STATUS_DUTYCYCLE_RESTRICTED`).

Please note that it may take several minutes up until the MAC layer is allowed to transmit again. For **debug purposes only** you may disable the duty-cycle enforcement.
To do so please change `LORAWAN_DUTYCYCLE_ON` value to `false` in `main.c` file

```c
#if defined( REGION_EU868 )

#include "LoRaMacTest.h"

/*!
 * LoRaWAN ETSI duty cycle control enable/disable
 *
 * \remark Please note that ETSI mandates duty cycled transmissions. Use only for test purposes
 */
#define LORAWAN_DUTYCYCLE_ON                        true

#endif
```

For SX1272 and SX1276 shields the `Radio.IrqProcess()` is not necessary because everything is handled in the interrupt handlers.

For SX126x we must handle the radio interrupt outside the interrupt handler functions, that's why we introduced the `Radio.IrqProcess()`.

The `LoRaMacProcess` function replaces the `OnMacStateCheckTimerEvent` callback (which was called automatically every millisecond). The goal for this change was to avoid running a quite long process in an interrupt handler.

Please note that the provided examples should work without modifications and they have been tested against different network servers and have passed the LoRa-Alliance certification process. The only modifications that were made are the values from `Commissioning.h` file.

The master branch provides the latest certified version of the MAC layer implementation which is based on  [LoRaWAN Specification v1.0.2](https://lora-alliance.org/resource-hub/lorawantm-specification-v102) and [LoRaWAN Regional Parameters v1.0.2rB](https://www.lora-alliance.org/resource-hub/lorawantm-regional-parameters-v102rb) specifications.

The develop branch contains the current work preparing the next version release and it is based on [LoRaWAN Specification v1.0.3](https://lora-alliance.org/resource-hub/lorawantm-specification-v103) and [LoRaWAN Regional Parameters v1.0.3revA](https://lora-alliance.org/resource-hub/lorawan-regional-parameters-v103reva) specifications.

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1