Lora-net/LoRaMac-node #568: Overflow while counting enabled channels

HBevilacqua, 2018-10-09:
This issue arises when the loramac counts the number of channels which are registered in the channel mask (`channelsMask` static array).
I'm going to give an example for the `US915` region, but this issue applies for the other regions.

The maximum number of channels in US915 is 72:
`US915_MAX_NB_CHANNELS    72`

However, the `channelsMask` array can be set to 80 channels.
How?
e.g. By receiving a _LinkAdrRequest_ "0305FFFF61" that set `channelsMask` to:

>     channelsMask[0] = 0xFFFF
>     channelsMask[1] = 0xFFFF
>     channelsMask[2] = 0xFFFF
>     channelsMask[3] = 0xFFFF
>     channelsMask[4] = 0xFFFF

(because `linkAdrParams.ChMaskCtrl = 6`, see the `RegionUS915LinkAdrReq` function)
*5 * (channel mask to 0xFFFF) = 5 * 16 channels = 80 channels.*

Ok, for the moment no overflow but...
When I send a RF frame after processing the link adr, `CountNbOfEnabledChannels` function is called.
Two loops are nested to scan all bits of `channelsMask`.

For the first one, a condition checks the max number of channels:
`for( uint8_t i = 0, k = 0; i < US915_MAX_NB_CHANNELS; i += 16, k++ )`

But in the second loop which scans all bits into a block of 16-bit, no check is performed:
`for( uint8_t j = 0; j < 16; j++ )`

`channels[i + j]` is used in these loops. I remind yout that `channels` is an array of 72 structures. So an overflow occurs from `i = 64`, `j > 7` (from channels[72] to channels[79] because 80 channels are defined.

A check should be implemented such as:
`for( uint8_t j = 0; ( j < 16 ) && ( ( i + j ) < US915_MAX_NB_CHANNELS ); j++ )`

For better understanding, the `CountNbOfEnabledChannels `function is given below:

>     static uint8_t CountNbOfEnabledChannels( uint8_t datarate, uint16_t* channelsMask, ChannelParams_t* channels, Band_t* bands, uint8_t* enabledChannels, uint8_t* delayTx )
>     {
>         uint8_t nbEnabledChannels = 0;
>         uint8_t delayTransmission = 0;
> 
>         for( uint8_t i = 0, k = 0; i < US915_MAX_NB_CHANNELS; i += 16, k++ )
>         {
>             for( uint8_t j = 0; j < 16; j++ )
>             {
>                 if( ( channelsMask[k] & ( 1 << j ) ) != 0 )
>                 {
>                     if( channels[i + j].Frequency == 0 )
>                     { // Check if the channel is enabled
>                         continue;
>                     }
>                     if( RegionCommonValueInRange( datarate, channels[i + j].DrRange.Fields.Min,
>                                               channels[i + j].DrRange.Fields.Max ) == false )
>                     { // Check if the current channel selection supports the given datarate
>                         continue;
>                     }
>                     if( bands[channels[i + j].Band].TimeOff > 0 )
>                     { // Check if the band is available for transmission
>                         delayTransmission++;
>                         continue;
>                     }
>                     enabledChannels[nbEnabledChannels++] = i + j;
>                 }
>             }
>         }
> 
>         *delayTx = delayTransmission;
>         return nbEnabledChannels;
>     }

djaeckle, 2018-10-12:
Hi HBevilacqua,

thanks for the report. I will take a look at it and will come back to you soon.

HBevilacqua, 2018-10-12:
Hi djaeckle,

I'm using the v4.4.0, but it would seem that `CountNbOfEnabledChannels` function has not been updated since this version:
https://github.com/Lora-net/LoRaMac-node/blob/v4.4.2-rc.3/src/mac/region/RegionUS915.c

djaeckle, 2018-10-15:
I can confirm that this is an issue. I have set the issue to the next milestone.

BenAtPip, 2018-10-15:
How is this issue is affecting performance of the module? Is your device not correctly being set to a channel by the backend?

ConnyBusy, 2018-10-16:
I think the primary issue here is the processing of LinkADR command with ChMaskCntl 6 and 7. The node must ensure that the ChMask for these two cases is max of 0x00FF (only 8 channels available with500KHz bandwidth). Sending a channel mask greater than this should be rejected. I think this should affect only regions like US915, AU915 & CN470 with fixed Channel plans.

closed 2018-10-26 by djaeckle