Lora-net/LoRaMac-node #571: //Intentional fall-through for case statement

gpgib6, 2018-10-18:
This isn't really a problem but I'm getting compiler warnings "no break at the end of case".
If I add the following to those locations:
//no break
...then the warnings go away.
Perhaps //no break is the default as opposed to //Intentional fall-through, at least this is working for me in System Workbench for STM32.

mluis1, 2018-10-18:
Thanks for the report.

Is the warning being generated for each intentional fall-through case?

Could you please tell us the compiler version that you used. It maybe related to the version being used.

When we fixed the compiler warnings we have compiled all supported platforms and no warning has been generated.

The compiler version we used was:
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE
```

GCC accepts different kinds of comments to suppress this warning. We just picked one.
https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html

gpgib6, 2018-10-19:
Mine is a little older, maybe that's why.
I changed the comment to the following at the warning is also suppressed:
//Intentional fall-through, no break

C:\Ac6\SystemWorkbench\plugins\fr.ac6.mcu.externaltools.arm-none.win32_1.15.0.201708311556\tools\compiler\bin>arm-none-eabi-gcc.exe --version
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

mluis1, 2018-10-26:
I have just tried to add the ", no break" statement at the end of the comment and with GCC compiler version 7.2.1 the below warning is generated:
` warning: this statement may fall through [-Wimplicit-fallthrough=]`

Thus, for the moment we keep the current comment up until we find a better solution.

omogenot, 2018-10-26:
just to put my 1 cent in here.
The gcc arm compiler expects /* no break */ to remove the compile warning.

closed 2018-10-31 by mluis1