Lora-net/LoRaMac-node #573: FRMPayload field in Class B sometimes not encrypted

ConnyBusy, 2018-10-20:
Hi,
in class B the FRMPayload field is sometimes not encrypted specifically, when in test mode and implements the Downlink Counter. You may want to have a look why this is happening. I am not able to figure out where the error is coming from since it does not happen in Class A.

djaeckle, 2018-10-25:
Hi ConnyBusy,

we never saw this issue. Could you please provide more details about the issue.

ConnyBusy, 2018-10-25:
Hi djaeckle,
hier some logs.
UL :40:9F:24:82:00:90:36:00:E0:C6:05:FA:20:A0:F2 ==> decrypted FRMPayload(downlink counter) :00:03
DL :60:9F:24:82:00:00:04:00:00:BA:52:01:C2:68:67:56:CD:90:6B:CC:B4:30:B2 ==>decrypted  FRMPayload (LinkADRandRXParamSetup) :03:00:07:00:01:05:00:D2:AD:84
UL :40:9F:24:82:00:94:37:00:03:07:05:07:E0:**00:04**:3C:69:49:DC ==> decrypted FRMPayload(downlink counter) :71:DF (boldface was not encrypted so decryption resulted in the encrypted version) FOpts::03:07:05:07:
**Please note that the downlink was sent in a pingslot. It doesn't happen all the time and it happens only when the FRMPayload contains the downlink counter.**

ConnyBusy, 2018-11-05:
Hi djaeckle,
I have made an observation that  reproduces the issue here. When ever the Node acquires/receives a beacon then the issue is always seen for the next uplink.  I hope this helps you narrow your analysis.

djaeckle, 2019-02-07:
Hi ConnyBusy,

thanks for the feedback. Do you mean with 'test mode' the compliance test mode?

ConnyBusy, 2019-02-07:
Hi djaeckle,
yes, that is what I meant. These logs below might interest you: You can use the standard keys to decode the data field of the rxpk

{'rxpk': [{'chan': 1,
           'codr': '4/5',
           'data': 'QJ8kggCQBwDgN1TmKRo5',
           'datr': 'SF12BW125',
           'freq': 868.3,
           'lsnr': 9.0,
           'modu': 'LORA',
           'rfch': 1,
           'rssi': -30,
           'size': 15,
           'stat': 1,
           'time': '2019-01-26T19:16:40.352927Z',
           'tmms': 1232565419353,
           'tmst': 2481595940}]}

Received LORA Unconfirmed Data Up Frame from DevAddr: 00:82:24:9F
DR: SF12BW125 FCntUp: 00:00:00:07 on Freq CH: 868.3000 MHz FPort: E0
PHYPayload : 40:9F:24:82:00:90:07:00:E0:37:54:E6:29:1A:39
FRMPayload : 00:03
FOpts field: 
FCTRL field: 90 SNR:  9.0 RSSI: -30 CR: 4/5 GPSTimestamp: 2481595940

{'txpk': {'codr': '4/5',
          'data': 'YJ8kggAFCAAGBgYGBuA/W/xKzcM0r7izVw8LtOCS/iwPPFPmKr4hjkv/szmuEGMX2cKmDjK0O+7uYIe4nLGklA==',
          'datr': 'SF9BW125',
          'freq': 869.525,
          'imme': 'false',
          'ipol': 'true',
          'modu': 'LORA',
          'ncrc': 'true',
          'powe': 14,
          'prea': 8,
          'rfch': 0,
          'size': 64,
          'tmms': 1232565421980}}

Sending LORA Unconfirmed Data Down Frame in PingSlot 45 to DevAddr: 00:82:24:9F
DR: SF9BW125 FCntDwn: 00:00:00:08 on Freq CH: 869.5250 MHz FPort: E0
PHYPayload : 60:9F:24:82:00:05:08:00:06:06:06:06:06:E0:3F:5B:FC:4A:CD:C3:34:AF:B8:B3:57:0F:0B:B4:E0:92:FE:2C:0F:3C:53:E6:2A:BE:21:8E:4B:FF:B3:39:AE:10:63:17:D9:C2:A6:0E:32:B4:3B:EE:EE:60:87:B8:9C:B1:A4:94
FRMPayload : 03:C2:E6:A6:17:27:C5:8D:6F:D4:7F:2D:5E:CF:C0:34:F5:E7:0A:E2:A9:41:C9:09:21:11:FD:9E:1D:D8:19:F3:73:49:CA:61:46:80:5F:45:D6:B4:CE:1E:3A:BB
FOpts field: 06:06:06:06:06
FCTRL field: 05 CR: 4/5 GPSTimestamp: 1232565419353

{'txpk': {'codr': '4/5',
          'data': 'YJ8kggAFCQAGBgYGBuDVG7xenVMaUqw7JoRyWGrtMKqJ/znbzylIFcZZefN2idqm2pZetxq7yI96GqG62ZfILA==',
          'datr': 'SF9BW125',
          'freq': 869.525,
          'imme': 'false',
          'ipol': 'true',
          'modu': 'LORA',
          'ncrc': 'true',
          'powe': 14,
          'prea': 8,
          'rfch': 0,
          'size': 64,
          'tmms': 1232565422940}}
Sending LORA Unconfirmed Data Down Frame in PingSlot 46 to DevAddr: 00:82:24:9F
DR: SF9BW125 FCntDwn: 00:00:00:09 on Freq CH: 869.5250 MHz FPort: E0
PHYPayload : 60:9F:24:82:00:05:09:00:06:06:06:06:06:E0:D5:1B:BC:5E:9D:53:1A:52:AC:3B:26:84:72:58:6A:ED:30:AA:89:FF:39:DB:CF:29:48:15:C6:59:79:F3:76:89:DA:A6:DA:96:5E:B7:1A:BB:C8:8F:7A:1A:A1:BA:D9:97:C8:2C
FRMPayload : 03:C2:E6:A6:17:27:C5:8D:6F:D4:7F:2D:5E:CF:C0:34:F5:E7:0A:E2:A9:41:C9:09:21:11:FD:9E:1D:D8:19:F3:73:49:CA:61:46:80:5F:45:D6:B4:CE:1E:3A:BB
FOpts field: 06:06:06:06:06
FCTRL field: 05 CR: 4/5 GPSTimestamp: 1232565419353

{'txpk': {'codr': '4/5',
          'data': 'YJ8kggAFCgAGBgYGBuBdNrOi6eThsq1hvEnN/VNDUNKDeL6eX/pYzEd+Nd1r1rL991iReochMnUUAO1DnWF8CA==',
          'datr': 'SF9BW125',
          'freq': 869.525,
          'imme': 'false',
          'ipol': 'true',
          'modu': 'LORA',
          'ncrc': 'true',
          'powe': 14,
          'prea': 8,
          'rfch': 0,
          'size': 64,
          'tmms': 1232565423900}}

Sending LORA Unconfirmed Data Down Frame in PingSlot 47 to DevAddr: 00:82:24:9F
DR: SF9BW125 FCntDwn: 00:00:00:0A on Freq CH: 869.5250 MHz FPort: E0
PHYPayload : 60:9F:24:82:00:05:0A:00:06:06:06:06:06:E0:5D:36:B3:A2:E9:E4:E1:B2:AD:61:BC:49:CD:FD:53:43:50:D2:83:78:BE:9E:5F:FA:58:CC:47:7E:35:DD:6B:D6:B2:FD:F7:58:91:7A:87:21:32:75:14:00:ED:43:9D:61:7C:08
FRMPayload : 03:C2:E6:A6:17:27:C5:8D:6F:D4:7F:2D:5E:CF:C0:34:F5:E7:0A:E2:A9:41:C9:09:21:11:FD:9E:1D:D8:19:F3:73:49:CA:61:46:80:5F:45:D6:B4:CE:1E:3A:BB
FOpts field: 06:06:06:06:06
FCTRL field: 05 CR: 4/5 GPSTimestamp: 1232565419353
Gateway Command received: TXPK_ACK 
Gateway Command received: TXPK_ACK 
Gateway Command received: TXPK_ACK 
{'rxpk': [{'chan': 2,
           'codr': '4/5',
           'data': 'QJ8kggCQCAAABgAIBgAIBgAIBgAIBgAIBgAIBgAIBgAIBgAIBgAIBgAIBgAIBgAIBgAIBgAIQrhfZg==',
           'datr': 'SF12BW125',
           'freq': 868.5,
           'lsnr': 7.8,
           'modu': 'LORA',
           'rfch': 1,
           'rssi': -30,
           'size': 58,
           'stat': 1,
           'time': '2019-01-26T19:16:51.839959Z',
           'tmms': 1232565430840,
           'tmst': 2493082972}]}

Unknown CID: 0xD5 Stopped processing MAC buffer

Received LORA Unconfirmed Data Up Frame from DevAddr: 00:82:24:9F
DR: SF12BW125 FCntUp: 00:00:00:08 on Freq CH: 868.5000 MHz FPort: 00
PHYPayload : 40:9F:24:82:00:90:08:00:00:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:06:00:08:42:B8:5F:66
FRMPayload : D5:F7:F8:F7:EC:7B:1F:37:E7:D5:92:E5:6D:C5:A4:5F:4D:24:D8:68:6D:B8:62:3B:A1:C9:BF:3A:21:A5:9A:DB:92:C1:D5:61:9B:24:52:AA:BF:23:4F:46:6D
FOpts field: 
FCTRL field: 90 SNR:  7.8 RSSI: -30 CR: 4/5 GPSTimestamp: 2493082972

closed 2019-06-07 by mluis1