Lora-net/LoRaMac-node #576: No hex and bin files output

Lozeonjyu, 2018-10-24:
Hi,

I was trying to use my L073+SX1261 shield to load the code but I couldn't find the bin file. When I trace back the makefile under "apps" I can see

create_bin_output(${PROJECT_NAME}-${CLASS})
create_hex_output(${PROJECT_NAME}-${CLASS})

Should these two commands generate the hex and bin files? If not, how can I generate the bin file or load the code to L073?

Thanks

djaeckle, 2018-10-25:
Hi Lozeonjyu,

> Should these two commands generate the hex and bin files?

Yes. After a succssful build, you will find the binaries in directory (e.g. LoRaMac Class A) _build/src/apps/LoRaMac_.

Lozeonjyu, 2018-10-25:
> Hi Lozeonjyu,
> 
> > Should these two commands generate the hex and bin files?
> 
> Yes. After a succssful build, you will find the binaries in directory (e.g. LoRaMac Class A) _build/src/apps/LoRaMac_.

That's wired. I didn't see it. Could that be the setup problem? I was told that I should see a map file after generate but I couldn't find that either. I am sure I am looking at the right directory.

ConnyBusy, 2018-10-25:
Hi Lozeonjyu,
which commands did you run. Generally, you have to run CMake and after that run Make to generate/build the bin and hex files.
Could you also provide which tools you are using to build the executables?

Lozeonjyu, 2018-10-25:
> Hi Lozeonjyu,
> which commands did you run. Generally, you have to run CMake and after that run Make to generate/build the bin and hex files.
> Could you also provide which tools you are using to build the executables?

Actually, I used CMake-GUI. specified toolchain-arm-none-eabi.cmake as toolchain, configured, then generated.

Before that I make sure I have tools installed (windows):
Cmake
GNU Arm Embedded Toolchain
MinGW
OpenOCD

djaeckle, 2018-10-26:
Hi,

plase take a look at the Wiki: [Development-environment](https://github.com/Lora-net/LoRaMac-node/wiki/Development-environment). The easiest way to verify if your build system is setup correctly is to run cmake on the console. After you have cloned the repository, you could do:

$: cd LoRaMac-node
$: mkdir build
$: cd build
$: cmake -DCMAKE_TOOLCHAIN_FILE="cmake/toolchain-arm-none-eabi.cmake" -DTOOLCHAIN_PREFIX="/path/to/the/toolchain" -G "MinGW Makefiles" ..

mluis1, 2018-10-26:
In plus of the commands provided by @djaeckle you need to run the make command.
$: make

Please find below an example provided on issue #504 

```bash
MSYS /c/loramac-node$
MSYS /c/loramac-node$ mkdir build
MSYS /c/loramac-node$ cd build
MSYS /c/loramac-node/build$ cmake -DCMAKE_BUILD_TYPE=Debug \
           -DTOOLCHAIN_PREFIX="/c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major" \
           -DCMAKE_TOOLCHAIN_FILE="../cmake/toolchain-arm-none-eabi.cmake" \
           -DAPPLICATION="LoRaMac" \
           -DCLASS="classA" \
           -DCLASSB_ENABLED="OFF" \
           -DACTIVE_REGION="LORAMAC_REGION_EU868" \
           -DREGION_EU868="ON" \
           -DREGION_US915="OFF" \
           -DREGION_CN779="OFF" \
           -DREGION_EU433="OFF" \
           -DREGION_AU915="OFF" \
           -DREGION_AS923="OFF" \
           -DREGION_CN470="OFF" \
           -DREGION_KR920="OFF" \
           -DREGION_IN865="OFF" \
           -DREGION_RU864="OFF" \
           -DBOARD="NucleoL073" \
           -DMBED_RADIO_SHIELD="SX1262DVK1CAS"  ..

-- The C compiler identification is GNU 7.2.1
-- The CXX compiler identification is GNU 7.2.1
-- Check for working C compiler: /c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc
-- Check for working C compiler: /c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-g++
-- Check for working CXX compiler: /c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Linker script: /c/loramac-node/src/boards/NucleoL073/cmsis/arm-gcc/stm32l073xx_flash.ld
-- The ASM compiler identification is GNU
-- Found assembler: /c/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /c/loramac-node/build

MSYS /c/loramac-node/build$ make -j 4

Scanning dependencies of target peripherals
Scanning dependencies of target NucleoL073
Scanning dependencies of target radio
Scanning dependencies of target system
[  1%] Building C object src/radio/CMakeFiles/radio.dir/sx126x/radio.c.obj
[  2%] Building C object src/peripherals/CMakeFiles/peripherals.dir/gpio-ioe.c.obj
[  3%] Building C object src/system/CMakeFiles/system.dir/adc.c.obj
[  5%] Building C object src/radio/CMakeFiles/radio.dir/sx126x/sx126x.c.obj
[  6%] Building C object src/system/CMakeFiles/system.dir/delay.c.obj
[  7%] Building C object src/peripherals/CMakeFiles/peripherals.dir/mag3110.c.obj
[  8%] Building C object src/system/CMakeFiles/system.dir/eeprom.c.obj
[  8%] Built target radio
[ 10%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/board.c.obj
[ 11%] Building C object src/peripherals/CMakeFiles/peripherals.dir/mma8451.c.obj
Scanning dependencies of target mac
[ 12%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMac.c.obj
[ 13%] Building C object src/system/CMakeFiles/system.dir/fifo.c.obj
[ 15%] Building C object src/peripherals/CMakeFiles/peripherals.dir/mpl3115.c.obj
[... 55 lines cut ...]
[ 83%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_i2c_ex.c.obj
[ 83%] Built target mac
[ 84%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pwr.c.obj
[ 86%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pwr_ex.c.obj
[ 87%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rcc.c.obj
[ 88%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rcc_ex.c.obj
[ 89%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rtc.c.obj
[ 91%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rtc_ex.c.obj
[ 92%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_spi.c.obj
[ 93%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_uart.c.obj
[ 94%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_uart_ex.c.obj
[ 96%] Building C object src/boards/NucleoL073/CMakeFiles/NucleoL073.dir/sx1262dvk1cas-board.c.obj
[ 96%] Built target NucleoL073
Scanning dependencies of target LoRaMac-classA
[ 97%] Building C object src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/common/NvmCtxMgmt.c.obj
[ 98%] Building C object src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/classA/NucleoL073/main.c.obj
[100%] Linking C executable LoRaMac-classA
   text    data     bss     dec     hex filename
  74536     408    6928   81872   13fd0 LoRaMac-classA
[100%] Built target LoRaMac-classA
Scanning dependencies of target LoRaMac-classA.hex
Scanning dependencies of target LoRaMac-classA.bin
[100%] Built target LoRaMac-classA.hex
[100%] Built target LoRaMac-classA.bin
```

Lozeonjyu, 2018-10-26:
> Please, could you also provide a screenshot of your CMake-GUI settings and your paths. This will be useful to figure out if it's a configuration issue. Are you able to run generate without any errors?

Please check the pics below:
[image]
[image]

Config:
[image]

Generate:
[image]

It seems all right according to the message window. 

Environment variable on my laptop:
[image]

Thanks a lot. 😃

Lozeonjyu, 2018-10-26:
> Hi,
>
> plase take a look at the Wiki: [Development-environment](https://github.com/Lora-net/LoRaMac-node/wiki/Development-environment). The easiest way to verify if your build system is setup [...]

Thanks guys @djaeckle @mluis1 I will try on the console. I used GUI at the first place because it is easier for me to do the config.

mluis1, 2018-10-26:
You must execute the make command after using your GUI. The cmake-gui only generates the makefiles.
The make command is the one that performs the compilation tasks based on the cmake-gui generated makefiles.