Lora-net/LoRaMac-node #578: MibGetRequest(PHY_DEF_MAX_EIRP) returns incorrect EIRP value for US915

catsup, 2018-10-29:
When performing a Mib get request for the default max EIRP value, US915 returns 0dBm.

Given that _EIRP(dB) = ERP(dB) + 2.15_, I would have expected to see 32.

Other regions correctly return their respective EIRP values.

This appears to be due to a fall through in the switch/case statement:

```
RegionUS915.c:        case PHY_DEF_MAX_EIRP:
RegionUS915.c-        case PHY_DEF_ANTENNA_GAIN:
RegionUS915.c-        {
RegionUS915.c-            phyParam.fValue = 0;
RegionUS915.c-            break;
RegionUS915.c-        }
```

mluis1, 2018-10-29:
Duplicates #563

mluis1, 2018-10-30:
As per regional parameters document the US915 region is the only region that doesn't make use of EIRP and doesn't define MaxEIRP usage. It is only said that the max conducted output power is 30 dBm (1 Watt) and the antenna gain must be lower than 6 dBi.

That's why the regional implementation returns 0 for PHY_DEF_MAX_EIRP and PHY_DEF_ANTENNA_GAIN. These parameters aren't used on US915 region.
Although we use the same function to compute the TxPower `RegionCommonComputeTxPower`. We did it like this to avoid having a different function just for US915 region.

It means that when making a US915 device one must adapt the sx12xx-board.c file in order to get the right output power.

We would prefer to have the regional parameters specification to apply the same behavior for all regions but, unfortunately it is not the case.

We would recommend that you contact the LoRa-Allaince technical committee in order to have the specifications updated.

We propose to change the `RegionCommonComputeTxPower` function documentation as shown below.

```c
/*!
 * \brief Computes the txPower, based on the max EIRP and the antenna gain.
 * 
 * \remark US915 region uses a conducted power as input value for maxEirp.
 *         Thus, the antennaGain parameter must be set to 0.
 *
 * \param [IN] txPower TX power index.
 *
 * \param [IN] maxEirp Maximum EIRP.
 *
 * \param [IN] antennaGain Antenna gain. Referenced to the isotropic antenna.
 *                         Value is in dBi. ( antennaGain[dBi] = measuredAntennaGain[dBd] + 2.15 )
 *
 * \retval Returns the physical TX power.
 */
int8_t RegionCommonComputeTxPower( int8_t txPowerIndex, float maxEirp, float antennaGain );
```

catsup, 2018-10-30:
This does not address the problem raised here: MibGetRequest() is not returning a correct value nor an error condition.

My take would be to have MibGetRequest(PHY_DEF_MAX_EIRP) either return the max conducted power (ERP?) _converted to EIRP_, or an error status. Returning a success status with an EIRP of 0dB does not seem like the right thing to do.

mluis1, 2018-10-30:
Are you more happy if we change the code as follows:
```patch
diff --git a/src/mac/region/RegionUS915.c b/src/mac/region/RegionUS915.c
index 748da1d2..38d0c431 100644
--- a/src/mac/region/RegionUS915.c
+++ b/src/mac/region/RegionUS915.c
@@ -426,12 +426,17 @@ PhyParam_t RegionUS915GetPhyParam( GetPhyParams_t* getPhy )
             break;
         }
         case PHY_DEF_MAX_EIRP:
+        {
+            // WARNING: Please don't rely on this returned value for US915 region
+            //          only the conducted power matters.
+            phyParam.fValue = US915_DEFAULT_MAX_CONDUCTED_POWER + 2.15;
+            break;
+        }
         case PHY_DEF_ANTENNA_GAIN:
         {
             phyParam.fValue = 0;
             break;
         }
-
         case PHY_BEACON_CHANNEL_FREQ:
         {
             phyParam.Value = US915_BEACON_CHANNEL_FREQ;
@@ -718,7 +723,7 @@ bool RegionUS915TxConfig( TxConfigParams_t* txConfig, int8_t* txPower, TimerTime
     int8_t phyTxPower = 0;
 
     // Calculate physical TX power
-    phyTxPower = RegionCommonComputeTxPower( txPowerLimited, US915_DEFAULT_MAX_ERP, 0 );
+    phyTxPower = RegionCommonComputeTxPower( txPowerLimited, US915_DEFAULT_MAX_CONDUCTED_POWER, 0 );
 
     // Setup the radio frequency
     Radio.SetChannel( NvmCtx.Channels[txConfig->Channel].Frequency );
@@ -1098,7 +1103,7 @@ void RegionUS915SetContinuousWave( ContinuousWaveParams_t* continuousWave )
     uint32_t frequency = NvmCtx.Channels[continuousWave->Channel].Frequency;
 
     // Calculate physical TX power
-    phyTxPower = RegionCommonComputeTxPower( txPowerLimited, US915_DEFAULT_MAX_ERP, 0 );
+    phyTxPower = RegionCommonComputeTxPower( txPowerLimited, US915_DEFAULT_MAX_CONDUCTED_POWER, 0 );
 
     Radio.SetTxContinuousWave( frequency, phyTxPower, continuousWave->Timeout );
 }
diff --git a/src/mac/region/RegionUS915.h b/src/mac/region/RegionUS915.h
index 71faba29..f3c64d08 100644
--- a/src/mac/region/RegionUS915.h
+++ b/src/mac/region/RegionUS915.h
@@ -100,9 +100,9 @@
 #define US915_DEFAULT_TX_POWER                      TX_POWER_0
 
 /*!
- * Default Max ERP
+ * Default Max conducted power
  */
-#define US915_DEFAULT_MAX_ERP                      30.0f
+#define US915_DEFAULT_MAX_CONDUCTED_POWER           30.0f
 
 /*!
  * ADR Ack limit
```

catsup, 2018-10-30:
Yes, returning a valid EIRP value corresponding to the max ERP value seems more logical to me than returning 0. And it's much more consistent to use the same unit across the board IMV.

I don't see the point for the warning though, unless I misunderstand the purpose of PHY_DEF_MAX_EIRP - to me, it still represents the max allowed output power, just using a different unit, EIRP instead of ERP ?

Nor for changing the name of the max allowed output power from US915_DEFAULT_MAX_ERP to US915_DEFAULT_MAX_CONDUCTED_POWER, but I can live with that ;^)

closed 2018-10-31 by mluis1