Lora-net/LoRaMac-node #579: Build Error, Please help

1405224, 2018-11-05:
Hi guys,
I have been trying to build the project and i have followed the instructions, but I'm having the following error and I'm desperate for your help please.
please see below the output:

`[build] Starting build
[proc] Executing command: "C:/Program Files/CMake/bin/cmake.exe" --build "c:/Users/Noman/Desktop/Class B/LoRaMac-node-feature-5.0.0/build" --config Debug --target LoRaMac-classA -- /m /property:GenerateFullPaths=true
[build] Microsoft (R) Build Engine version 12.0.21005.1
[build] [Microsoft .NET Framework, version 4.0.30319.42000]
[build] Copyright (C) Microsoft Corporation. All rights reserved.
[build] 
[build] Build started 05/11/2018 23:02:28.
[build]      1>Project "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\apps\LoRaMac\LoRaMac-classA.vcxproj" on node 1 (default targets).
[build]      1>Project "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\apps\LoRaMac\LoRaMac-classA.vcxproj" (1) is building "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\ZERO_CHECK.vcxproj" (2) on node 1 (default targets).
[build]      2>InitializeBuildStatus:
[build]          Creating "Win32\Debug\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
[build]        CustomBuild:
[build]          All outputs are up-to-date.
[build]        FinalizeBuildStatus:
[build]          Deleting file "Win32\Debug\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild".
[build]          Touching "Win32\Debug\ZERO_CHECK\ZERO_CHECK.tlog\ZERO_CHECK.lastbuildstate".
[build]      2>Done Building Project "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\ZERO_CHECK.vcxproj" (default targets).
[build]      1>Project "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\apps\LoRaMac\LoRaMac-classA.vcxproj" (1) is building "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\peripherals\peripherals.vcxproj" (5) on node 4 (default targets).
[build]      5>InitializeBuildStatus:
[build]          Touching "peripherals.dir\Debug\peripherals.tlog\unsuccessfulbuild".
[build]      1>Project "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\apps\LoRaMac\LoRaMac-classA.vcxproj" (1) is building "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\system\system.vcxproj" (7) on node 3 (default targets).
[build]      7>InitializeBuildStatus:
[build]          Touching "system.dir\Debug\system.tlog\unsuccessfulbuild".
[build]      1>Project "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\apps\LoRaMac\LoRaMac-classA.vcxproj" (1) is building "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\boards\NucleoL073\NucleoL073.vcxproj" (3) on node 2 (default targets).
[build]      3>InitializeBuildStatus:
[build]          Touching "NucleoL073.dir\Debug\NucleoL073.tlog\unsuccessfulbuild".
[build]      7>CustomBuild:
[build]          All outputs are up-to-date.
[build]      3>CustomBuild:
[build]          All outputs are up-to-date.
[build]      5>CustomBuild:
[build]          All outputs are up-to-date.
[build]      7>ClCompile:
[build]          C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\CL.exe /c /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\crypto" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals\soft-se" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\mac" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\mac\region" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\radio" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\radio\sx126x" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\radio\sx1272" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\radio\sx1276" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards\NucleoL073" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards\NucleoL073\cmsis" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards\NucleoL073\..\mcu\stm32" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards\NucleoL073\..\mcu\stm32\cmsis" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards\NucleoL073\..\mcu\stm32\STM32L0xx_HAL_Driver\Inc" /nologo /Wall /WX- /O2 /Oy- /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Fo"system.dir\Debug\\" /Fd"system.dir\Debug\vc120.pdb" /Gd /TC /analyze- /errorReport:queue  -Og -g -mthumb -g2 -fno-builtin -mcpu=cortex-m0plus -Wextra -pedantic -Wno-unused-parameter -ffunction-sections -fdata-sections -fomit-frame-pointer -mabi=aapcs -fno-unroll-loops -ffast-math -ftree-vectorize -std=gnu99 -Og -g "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\adc.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\delay.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\eeprom.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\fifo.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\gpio.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\gps.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\i2c.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\nvmm.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\systime.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\timer.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\uart.c"
[build]      5>ClCompile:
[build]          C:\Program Files (x86)\Microsoft Visual Studio 12.0\VC\bin\CL.exe /c /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals\soft-se" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\system\crypto" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\mac" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\mac\region" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\radio" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\radio\sx126x" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\radio\sx1272" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\radio\sx1276" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards\NucleoL073" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards\NucleoL073\cmsis" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards\NucleoL073\..\mcu\stm32" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards\NucleoL073\..\mcu\stm32\cmsis" /I"C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\boards\NucleoL073\..\mcu\stm32\STM32L0xx_HAL_Driver\Inc" /nologo /Wall /WX- /O2 /Oy- /D "CMAKE_INTDIR=\"Debug\"" /D _MBCS /Gm- /MD /GS /fp:precise /Zc:wchar_t /Zc:forScope /Fo"peripherals.dir\Debug\\" /Fd"peripherals.dir\Debug\vc120.pdb" /Gd /TC /analyze- /errorReport:queue  -Og -g -mthumb -g2 -fno-builtin -mcpu=cortex-m0plus -Wextra -pedantic -Wno-unused-parameter -ffunction-sections -fdata-sections -fomit-frame-pointer -mabi=aapcs -fno-unroll-loops -ffast-math -ftree-vectorize -std=gnu99 -Og -g "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals\gpio-ioe.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals\mag3110.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals\mma8451.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals\mpl3115.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals\pam7q.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals\soft-se\aes.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals\soft-se\cmac.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\src\peripherals\soft-se\soft-se.c" "C:\Users\Noman\Desktop\Class B\LoRaMac-node-fea [... cut ...]
[build]        "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\apps\LoRaMac\LoRaMac-classA.vcxproj" (default target) (1) ->
[build]        "C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\radio\radio.vcxproj" (default target) (6) ->
[build]          cl : Command line error D8021: invalid numeric argument '/Wextra' [C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0\build\src\radio\radio.vcxproj]
[build] 
[build]     5 Warning(s)
[build]     5 Error(s)
[build] 
[build] Time Elapsed 00:00:00.71
[build] Build finished with exit code 1
`

mluis1, 2018-11-06:
Please refer to [development environment](https://github.com/Lora-net/LoRaMac-node/blob/develop/Doc/development-environment.md) document for instructions on how to setup a working environment. At the end of the document there are instructions on how to use [Visual Studio Code](https://code.visualstudio.com/Download).

From the provided log it looks like you are using Microsoft compiler which if I'm not mistaken isn't able by default to compile code for ARM processors.

The ARM GNU GCC compiler must be used in order to be able to compile the project.

If your goal is to use Visual Studio as an IDE you must setup the project in order to use the ARM GNU GCC compiler instead of the default compiler. I guess that there are several ways to do it.
* You may use directly the cmake + make tools or
* You may use a specific Makefile or
* You may change the compiler being called in the project options.

Please take a look at issue #555 where the users were able to build the project on VS2017.
You may also search on the web for such setup. I'm pretty sure someone did it already. (not necessarily for this project but for similar one). 

Another thing to do would be to remove spaces from the project path.
`C:\Users\Noman\Desktop\Class B\LoRaMac-node-feature-5.0.0` -> `C:\Users\Noman\Desktop\ClassB\LoRaMac-node-feature-5.0.0`
Usually GCC compiler doesn't like spaces in paths or file names.

1405224, 2018-11-06:
Thanks @mluis1 for your quick response, 
The issue is that after I click on the blue status bar of CMake Tools and choose a build type (Debug or Release) I can't see the ARM GNU GCC compiler in the kits list ( only the Microsoft compilers). 
I have over come this issue by selecting the [Unspecified] option. 
Now the build is successfuly completed, however, when I press Debug, a popup error messages shows up:

>  Unable to start debugging. Launch options string provided by the project system is invalid.
>  Unable to determine path to debugger. Please specify the "MIDebuggerPath" option.

and the output shows:

```
[build] Starting build
[proc] Executing command: "C:/Program Files/CMake/bin/cmake.exe" --build c:/Users/Noman/Desktop/ClassB/LoRaMac-node-feature-5.0.0/build --config Debug --target LoRaMac-classA -- -j 6
[build] [ 21%] Built target peripherals
[build] [ 24%] Built target radio
[build] [ 29%] Built target system
[build] [ 55%] Built target mac
[build] [ 96%] Built target NucleoL073
[build] [100%] Built target LoRaMac-classA
[build] Build finished with exit code 0
[proc] Executing command: "C:/Program Files (x86)/GNU Tools Arm Embedded/7 2018-q2-update/bin/arm-none-eabi-gdb.exe" --version
[proc] Executing command: gdb --version
```
Also at the end of the launch.json:
```
"MIMode": "gdb",
"miDebuggerPath": "C:/Program Files (x86)/GNU Tools Arm Embedded/7 2018-q2-update/bin/arm-none-eabi-gdb.exe",
"debugServerPath": "C:/OpenOCD/bin/openocd.exe"
```
 
Any ideas please?
Thanks

1405224, 2018-11-08:
Any thoughts on the above error guys?

Many thanks 
Noman

closed 2018-11-13 by 1405224