Lora-net/LoRaMac-node #582: Question on communication

gyha, 2018-11-07:
I have question on LoRa and LoRaWAN theoretically.
In the sample in pingpong, it is possible to commucation with each end devices.
But in LoRaWAN, when the end device is listening, it cannot receive the other end device message.
So what is want to know is that the message from the other device is blocked on phy layer or mac layer?
Could you explain it using such as PHDR or MHDR.
The reason why i ask this kinds of question is i am making multihop node by using sample codes.

Thank you in advance

reissjason, 2018-11-07:
It is blocked at the radio front-end. The I/Q signal is inverted so end-devices do not hear other devices and gateways do not hear other gateways.

See Radio.SetRxConfig and Radio.SetTxConfig iqInverted parameter in the src/mac/region/ files.

mluis1, 2019-02-12:
As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1