Lora-net/LoRaMac-node #583: Debug Error

1405224, 2018-11-12:
Hi Guys, 
I'll really appreciate any help regarding the issue below:

The build is successfuly completed, however, when I press Debug, a popup error messages shows up:

    Unable to start debugging. Launch options string provided by the project system is invalid.
    Unable to determine path to debugger. Please specify the "MIDebuggerPath" option.

and the output shows:

[build] Starting build
[proc] Executing command: "C:/Program Files/CMake/bin/cmake.exe" --build c:/Users/Noman/Desktop/ClassB/LoRaMac-node-feature-5.0.0/build --config Debug --target LoRaMac-classA -- -j 6
[build] [ 21%] Built target peripherals
[build] [ 24%] Built target radio
[build] [ 29%] Built target system
[build] [ 55%] Built target mac
[build] [ 96%] Built target NucleoL073
[build] [100%] Built target LoRaMac-classA
[build] Build finished with exit code 0
[proc] Executing command: "C:/Program Files (x86)/GNU Tools Arm Embedded/7 2018-q2-update/bin/arm-none-eabi-gdb.exe" --version
[proc] Executing command: gdb --version

Also at the end of the launch.json:

"MIMode": "gdb",
"miDebuggerPath": "C:/Program Files (x86)/GNU Tools Arm Embedded/7 2018-q2-update/bin/arm-none-eabi-gdb.exe",
"debugServerPath": "C:/OpenOCD/bin/openocd.exe"

Any ideas please?
Thanks

simonnilsson, 2018-11-12:
I believe that the problem is your **TOOLCHAIN_PREFIX** option. It does not allow spaces in the path, so you need to use the Windows short format.

I think this should be correct for your case:
```js
"TOOLCHAIN_PREFIX":"C:/PROGRA~2/GNUTOO~1/72018-~1",
```

1405224, 2018-11-13:
Thanks @simonnilsson that did solve the problem
Really appreciate your help.

simonnilsson, 2018-11-13:
@1405224 No problem. Don't forget to close the issue.

closed 2018-11-13 by 1405224