Lora-net/LoRaMac-node #586: NewChannelReq modifies default channels' datarates

jamesl-dm, 2018-11-14:
RegionEU868ChannelAdd allows a NewChannelReq to increase the maximum datarate of one of the default channels. However, the 1.1 spec says on line 1049 in section 5.7: 

> Depending on the region and frequency band used, in certain regions ([PHY]) the LoRaWAN
> specification imposes default channels which must be common to all devices and cannot be
>  modified by the NewChannelReq command .If the number of default channels is N, the
> default channels go from 0 to N-1, and the acceptable range for ChIndex is N to 15.

which seems to pretty clearly indicate that the NewChannelReq shouldn't be accepted if ChIndex corresponds to one of the default channels.

mluis1, 2018-11-14:
The current implementation doesn't let you change the default channels mandatory parameters. As such it complies with the specification. This behavior is checked by the LoRaWAN certification process and tests are successfully passed.

In order to be more flexible we implemented the function in such way that one could increase the datarate range in order to be able to use DR_6 and DR_7 datarates. This is the only thing that can be modified.

Please note that your citation is valid for all LoRaWAN versions.

```c
LoRaMacStatus_t RegionEU868ChannelAdd( ChannelAddParams_t* channelAdd )
...
    // Default channels don't accept all values
    if( id < EU868_NUMB_DEFAULT_CHANNELS )
    {
        // Validate the datarate range for min: must be DR_0
        if( channelAdd->NewChannel->DrRange.Fields.Min > DR_0 )
        {
            drInvalid = true;
        }
        // Validate the datarate range for max: must be DR_5 <= Max <= TX_MAX_DATARATE
        if( RegionCommonValueInRange( channelAdd->NewChannel->DrRange.Fields.Max, DR_5, EU868_TX_MAX_DATARATE ) == false )
        {
            drInvalid = true;
        }
        // We are not allowed to change the frequency
        if( channelAdd->NewChannel->Frequency != NvmCtx.Channels[id].Frequency )
        {
            freqInvalid = true;
        }
    }
...
}
```

jamesl-dm, 2018-11-14:
So are you saying the text of the standard is incorrect where it says 'the acceptable range for ChIndex is N to 15'? Or just that the certification process didn't manage to pick the issue up?

mluis1, 2018-11-14:
The specification is correct and certification process would catch any change to the expected and mandatory behavior. 
We just extend the functionality. This behavior doesn't go against the specification.

If you try to change a default channel frequency or changing the datarate range to something like DR_1 to DR_3 the end-device will refuse it.

The only default channel change that will be accepted is if you extend the datarate range.

Example:
EU868 first default channel is 868.1 MHz DR_0 to DR 5.
Changing the 1st default channel to
* 0 MHz, DR_0 to DR_5 - Refused because we are trying to disable the channel
* 868.7 MHz DR_0 to DR_5 - Refused due to invalid frequency change.
* 868.1 MHz DR_0 to DR_4 - Refused due to the invalid range change.
* 868.1 MHz DR_6 to DR_7 - Refused due to the invalid range change.
* 868.1 MHz DR_5 to DR_5 - Refused due to the invalid range change.
* 868.1 MHz DR_0 to DR7 - Accepted because we didn't change the default expected behavior.

mluis1, 2018-11-14:
I agree that the current implementation is not done exactly as specified. But, I think that the current implementation provides more flexibility for the users.

We can change the implementation to be exactly like specified in plus it will reduce a little bit the code size footprint in plus not all networks servers are aware of this behavior which means that it may never be used.

I re-open the issue and we will change the behavior to be exactly as specified.

jamesl-dm, 2018-11-14:
Thanks! Sorry for bugging you with small details like these. It's sometimes hard to know from outside of Semtech and the Alliance technical team exactly what the intention of the spec is - so your dev team ends up getting cross-examined for the finer details :)

mluis1, 2018-11-14:
No problem, this project has a lot of history and we implemented things that we tried and then have been abandoned in final specifications but not removed from the code base as they don't harm the functionality.

closed 2018-11-15 by mluis1