Lora-net/LoRaMac-node #587: Sticky Mac Commands resends forever

simonnilsson, 2018-11-14:
Hi, I have seen a possible bug in the stack and/or example applications. Using LoRaMac Class A application with LORAWAN_DUTYCYCLE_ON = 0. When server sends a downlink containing sticky commands (RXParamSetupReq, RXTimingSetupReq)  the device will respond immediately and keep sending them at 3~ second intervals. 

It looks like these sticky commands are only cleared when a downlink is received from server but since the uplink is unconfirmed they are never cleared. Is this intentional?

I'm on develop branch: 1af11bf4e3997351df16953100a10d665c176733
Server is a: https://www.loraserver.io/ @ v2.0.0

Downlink sent from server containing the sticky commands:
```
{
        "phyPayload": {
            "mhdr": {
                "mType": "UnconfirmedDataDown",
                "major": "LoRaWANR1"
            },
            "macPayload": {
                "fhdr": {
                    "devAddr": "0100020c",
                    "fCtrl": {
                        "adr": true,
                        "adrAckReq": false,
                        "ack": false,
                        "fPending": false,
                        "classB": false
                    },
                    "fCnt": 0,
                    "fOpts": [
                        {
                            "cid": "RXParamSetupReq",
                            "payload": {
                                "frequency": 869525000,
                                "dlSettings": "00"
                            }
                        },
                        {
                            "cid": "RXTimingSetupReq",
                            "payload": {
                                "delay": 1
                            }
                        }
                    ]
                },
                "fPort": null,
                "frmPayload": null
            },
            "mic": "f23a734a"
        }
    }
```

Response sent by device every 3 seconds:
```
    {
        "phyPayload": {
            "mhdr": {
                "mType": "UnconfirmedDataUp",
                "major": "LoRaWANR1"
            },
            "macPayload": {
                "fhdr": {
                    "devAddr": "0100020c",
                    "fCtrl": {
                        "adr": false,
                        "adrAckReq": false,
                        "ack": false,
                        "fPending": false,
                        "classB": false
                    },
                    "fCnt": 2,
                    "fOpts": [
                        {
                            "cid": "RXParamSetupReq",
                            "payload": {
                                "channelAck": true,
                                "rx2DataRateAck": true,
                                "rx1DROffsetAck": true
                            }
                        },
                        {
                            "cid": "RXTimingSetupReq",
                            "payload": null
                        }
                    ]
                },
                "fPort": 2,
                "frmPayload": [
                    {
                        "bytes": "FQ=="
                    }
                ]
            },
            "mic": "ba65f61c"
        }
    }
```

simonnilsson, 2018-11-15:
A Mlme Indication **MLME_SCHEDULE_UPLINK** is sent by the stack and that immediately schedules a new transmission by calling _OnTxNextPacketTimerEvent_.

I believe the **MLME_SCHEDULE_UPLINK** is working as intended because the sticky commands should be sent by end-device until a class A downlink is received by the end-device according to the specification.

The example application will send a message directly and not wait for the next scheduled uplink by calling _OnTxNextPacketTimerEvent_ and I suspect this is the problem.

The way I see it there are 2 possible solutions:

1. Make the uplink be confirmed for sticky commands. There will still be a problem if a response is not received by server however and it will continue to spam uplinks. So not a great solution.

2. Simply remove the call to _OnTxNextPacketTimerEvent_ and just let it be handled on the next regular uplink.

Am I on to something or is this completely wrong?

mluis1, 2018-11-15:
The sticky answers were introduced to the specification in order to ensure that critical MAC commands have been understood by the end-device. You can see the sticky answers like an acknowledgement.

When the server changes critical MAC parameters the end-device must indicate that it has changed the parameters and then the server must perform a downlink in order to confirm that everything is in sync.
You can see this behavior like a 3 way acknowledgment.

The MAC layer implementation notifies the application layer that an up-link should take place as soon as possible in order for the server to confirm that the parameters are synchronized.

On the current examples we decided to make an uplink as soon as possible. But, you are not obliged to do so.

Personally I think that when the server sends critical MAC command which generates sticky answers it should perform a downlink (even if it is an empty downlink) just to confirm the sent parameters.

This means that the issue is more on the server side as it doesn't confirm its request by performing a downlink when the end-device answers.

I don't know what @brocaar your server developer thinks but, maybe he can give his thoughts on this.

simonnilsson, 2018-11-15:
Okay, but even if the server decides to send a downlink in response (which as you say is not required by the spec) there is no guaranty the end-device receives this downlink (i.e. in bad radio conditions). In this case the current implementation only relies on the duty cycle limit to keep the device from constantly sending the uplinks.

I understand the applications in this repository are just example-implementations but I also think a lot of developers base their implementations on them (like me). Is it not a a good idea to handle this in a better way so the device will behave like a good LoRaWAN-citizen in this case?

mluis1, 2018-11-15:
We introduced this functionality because other network server providers requested to have an uplink sent as soon as possible in order to confirm the parameters.

As we cannot make every one happy. I would recommend that you change your code behavior in such way that it doesn't send so many uplinks. 
You could for instance execute the `MLME_SCHEDULE_UPLINK` notification handling once every N number of times in order to reduce the amount of uplinks. You could also just remove the call to `OnTxNextPacketTimerEvent` function and wait for your next scheduled uplink.

Please also note that the end-device will act as good LoRaWAN-citizen if you use the default settings. Those are in place to respect and follow the different regional regulators rules.
In your case as you are using EU868 region and have disabled the duty-cycle enforcement you are not anymore compliant with the regulations. Disabling the duty-cycle enforcement should only be made for debug purposes and for a short period of time, it must never be used in a final product.

simonnilsson, 2018-11-15:
I know the duty-cycle enforcement must be enabled for compliance, I only have it disabled for testing. The issue is still there even with the duty-cycle enabled, it's just hidden as the stack will block the transmissions.

I will go with option 2 above for my application. Remove the `OnTxNextPacketTimerEvent` call. And if server wants an immediate response to sticky commands it should send them as a confirmed-downlink.

mluis1, 2018-11-15:
Ok, nice. Could you please close the issue?

closed 2018-11-15 by simonnilsson