Lora-net/LoRaMac-node #589: Porting Stack to STM32L053

matomback, 2018-11-19:
I am trying to port the stack to STM32L053 Nucleo. I originally went with the L053 for support with I-CUBE-LRWAN example. Now I am trying to use the LoRaMac-Node stack with the L053 and SX1262MB2CAS radio.

I basically copied everything in src/boards/NucleoL073 to a new directory src/boards/NucleoL053, changing appropriate references from the L073 to L053. I did not make any significant changes to the files listed in the porting guide (board.c, gpio-board.c/.h, pinName-board.h, pinName-ioe.h, rtc-board.c/.h, spi-board.c/.h, sx1272-board.c/.h), as these seem to be the same for the STM32L0xx family.

After ironing out a few issues I almost have the stack compiling, however I get an error that RAM and FLASH have overflowed (see output below). This appears to be due to the smaller size of the memory on this board (64k/8k on STM32L053 as opposed to 192k/20k on STM32L073).

Is there any easy way around this? Anything I can do in the stack to knock off 10792 bytes flash, and 1488 bytes RAM? Or is the easiest solution to begin developing the application just to go with the STM32L073?

**Build Output**
	[build] Starting build
	[proc] Executing command: "C:/Program Files (x86)/CMake/bin/cmake.exe" --build c:/GIT/LoRa/LoRaMac-node_fork/LoRaMac-node/build --config Debug --target all -- -j 6
	[build] Scanning dependencies of target radio
	[build] Scanning dependencies of target system
	[build] Scanning dependencies of target peripherals
	[build] Scanning dependencies of target mac
	[build] [  3%] Building C object src/radio/CMakeFiles/radio.dir/sx126x/radio.c.obj
	[build] [  3%] Building C object src/peripherals/CMakeFiles/peripherals.dir/gpio-ioe.c.obj
	[build] [  6%] Building C object src/radio/CMakeFiles/radio.dir/sx126x/sx126x.c.obj
	[build] [  6%] Building C object src/system/CMakeFiles/system.dir/adc.c.obj
	[build] [  6%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMac.c.obj
	[build] [  7%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/board.c.obj
	[build] [  8%] Building C object src/system/CMakeFiles/system.dir/delay.c.obj
	[build] [ 10%] Building C object src/peripherals/CMakeFiles/peripherals.dir/mag3110.c.obj
	[build] [ 11%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacAdr.c.obj
	[build] [ 11%] Built target radio
	[build] [ 12%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/delay-board.c.obj
	[build] [ 13%] Building C object src/system/CMakeFiles/system.dir/eeprom.c.obj
	[build] [ 15%] Building C object src/peripherals/CMakeFiles/peripherals.dir/mma8451.c.obj
	[build] [ 16%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacClassB.c.obj
	[build] [ 17%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/eeprom-board.c.obj
	[build] [ 18%] Building C object src/system/CMakeFiles/system.dir/fifo.c.obj
	[build] [ 20%] Building C object src/peripherals/CMakeFiles/peripherals.dir/mpl3115.c.obj
	[build] [ 21%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacCommands.c.obj
	[build] [ 22%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/gpio-board.c.obj
	[build] [ 24%] Building C object src/system/CMakeFiles/system.dir/gpio.c.obj
	[build] [ 25%] Building C object src/peripherals/CMakeFiles/peripherals.dir/pam7q.c.obj
	[build] [ 26%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacConfirmQueue.c.obj
	[build] [ 27%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/lpm-board.c.obj
	[build] [ 29%] Building C object src/system/CMakeFiles/system.dir/gps.c.obj
	[build] [ 30%] Building C object src/peripherals/CMakeFiles/peripherals.dir/soft-se/aes.c.obj
	[build] [ 31%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacCrypto.c.obj
	[build] [ 32%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/rtc-board.c.obj
	[build] [ 34%] Building C object src/system/CMakeFiles/system.dir/i2c.c.obj
	[build] [ 36%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacFCntHandler.c.obj
	[build] [ 35%] Building C object src/peripherals/CMakeFiles/peripherals.dir/soft-se/cmac.c.obj
	[build] [ 37%] Building C object src/system/CMakeFiles/system.dir/nvmm.c.obj
	[build] [ 39%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/spi-board.c.obj
	[build] [ 40%] Building C object src/peripherals/CMakeFiles/peripherals.dir/soft-se/soft-se.c.obj
	[build] [ 41%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacParser.c.obj
	[build] [ 43%] Building C object src/system/CMakeFiles/system.dir/systime.c.obj
	[build] [ 44%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/uart-board.c.obj
	[build] [ 45%] Building C object src/peripherals/CMakeFiles/peripherals.dir/sx1509.c.obj
	[build] [ 46%] Building C object src/mac/CMakeFiles/mac.dir/LoRaMacSerializer.c.obj
	[build] [ 48%] Building C object src/system/CMakeFiles/system.dir/timer.c.obj
	[build] [ 49%] Building C object src/peripherals/CMakeFiles/peripherals.dir/sx9500.c.obj
	[build] [ 50%] Building C object src/mac/CMakeFiles/mac.dir/region/Region.c.obj
	[build] [ 51%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/sysIrqHandlers.c.obj
	[build] [ 53%] Building C object src/system/CMakeFiles/system.dir/uart.c.obj
	[build] [ 54%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionAS923.c.obj
	[build] [ 55%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/utilities.c.obj
	[build] [ 55%] Built target peripherals
	[build] [ 56%] Building ASM object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/cmsis/arm-gcc/startup_stm32l053xx.s.obj
	[build] [ 58%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionAU915.c.obj
	[build] [ 58%] Built target system
	[build] [ 59%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/cmsis/system_stm32l0xx.c.obj
	[build] [ 60%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionCN470.c.obj
	[build] [ 62%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal.c.obj
	[build] [ 63%] Building C object src/mac/CMakeFiles/mac.dir/region/RegionCN779.c.obj
	[build] [ 64%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_adc.c.obj
	[build] [ 65%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_adc_ex.c.obj
[... 14 lines cut ...]
	[build] [ 84%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pwr.c.obj
	[build] [ 86%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_pwr_ex.c.obj
	[build] [ 87%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rcc.c.obj
	[build] [ 88%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rcc_ex.c.obj
	[build] [ 89%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rtc.c.obj
	[build] [ 91%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_rtc_ex.c.obj
	[build] [ 91%] Built target mac
	[build] [ 92%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_spi.c.obj
	[build] [ 93%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_uart.c.obj
	[build] [ 94%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/__/mcu/stm32/STM32L0xx_HAL_Driver/Src/stm32l0xx_hal_uart_ex.c.obj
	[build] [ 96%] Building C object src/boards/NucleoL053/CMakeFiles/NucleoL053.dir/sx1262mbxcas-board.c.obj
	[build] [ 96%] Built target NucleoL053
	[build] Scanning dependencies of target LoRaMac-classA
	[build] [ 98%] Building C object src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/common/NvmCtxMgmt.c.obj
	[build] [ 98%] Building C object src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/classA/NucleoL053/main.c.obj
	[build] [100%] Linking C executable LoRaMac-classA
	[build] c:/progra~2/gnutoo~1/72018-~1/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: LoRaMac-classA section `.text' will not fit in region `FLASH'
[... cut ...]
	[build] c:/progra~2/gnutoo~1/72018-~1/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 10792 bytes
	[build] collect2.exe: error: ld returned 1 exit status
	[build] mingw32-make.exe[2]: *** [src\apps\LoRaMac\CMakeFiles\LoRaMac-classA.dir\build.make:251: src/apps/LoRaMac/LoRaMac-classA] Error 1
	[build] mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:402: src/apps/LoRaMac/CMakeFiles/LoRaMac-classA.dir/all] Error 2
	[build] mingw32-make.exe: *** [Makefile:83: all] Error 2
	[build] Build finished with exit code 2

SilasValera, 2018-11-19:
I am not sure if this suggestion will work:  
1. if you are going to use only LoRa modem, you can try removing code related to fsk, but you have to be very careful, this will reduce code size. Also try to remove any reference that are not used in your implementation of L053.
2. use optimization - for size and not for performance

matomback, 2019-01-03:
Thanks for the guidance. Tried optimizing but ended up switching to the L073 instead. Will close this issue.

closed 2019-01-03 by matomback