Lora-net/LoRaMac-node #594: CAD on the SX1261

photonthunder, 2018-11-28:
Curious if you have had any luck getting the CAD to work for the SX1261 (since it can detect the packet data as well as the preamble)?  I have played with the settings in Table 13-73 (cadDetPeak and cadDetMin) but I can't get any consistent responses.  I know the datasheet says "Choosing the right value is not easy..." but further insight would be very helpful.

mluis1, 2018-11-28:
We would recommend that you post your question on the [Semtech LoRa Community](https://semtech.force.com/lora) forum.

We can't really help you here as this project makes no use of the radio CAD features.

closed 2018-11-28 by mluis1

photonthunder, 2018-11-28:
Appreciate the response.  I posted on the Semtech Community but getting a response is a challenge.  In the responses to #335 and #175 the reason for not using CAD was that it only could detect the preamble.  With the SX1261 being able to detect packet data I thought you might be experimenting with CAD as a listen before talk feature.