Lora-net/LoRaMac-node #595: move function LoRaMacCheckForBeaconAcquisition from LoRaMac.c to LoRaMacClassB.c

nestorayuso, 2018-11-28:
I think we should move LoRaMacCheckForBeaconAcquisition from LoRaMac.c to LoRaMacClassB.c and use #ifdef LORAMAC_CLASSB_ENABLED statement to reduce code if class-B is not enabled at compile time.

mluis1, 2018-11-28:
Thanks @nestorayuso, we will do it as soon as possible.

nestorayuso, 2018-11-30:
The #ifdef LORAMAC_CLASSB_ENABLED statement can be used inside LoRaMac.c to reduce more code, and maybe a #ifdef LORAMAC_CLASSC_ENABLED statement to remove class-c code and multicast code.

Upgrading from v4.4.1 to v4.4.2 the code size increased a lot.

djaeckle, 2019-01-23:
Hi nestorayuso,

we will not move it for now, as it has some dependencies to the LoRaMac.c. However, we will consider it for the future.

closed 2019-02-07 by djaeckle