Lora-net/LoRaMac-node #597: Double increment of UpLinkCounter when receiving ACK for AdrAckReq (v4.4.0)

catsup, 2018-11-28:
Hello,

we are using a MAC based on release v4.4.0, and are seeing occasional missed uplinks on our servers. This was eventually traced down to a double increment of the UpLinkCounter variable in OnMacStateCheckTimerEvent(), when receiving an ACK for an AdrAckReq. 

This issue is very simple to reproduce by just sending uplinks from a device close to the gateway, with ADR enabled (it's also quite obvious from the code once you know what you're looking for).

Is this issue known ? Has it been addressed in later releases ?

I did try to doublecheck myself, but while the code is similar in v4.4.1 (and likely affected as well), it is radically different in v4.4.2, making direct comparisons impossible :-(.

Thanks,

mluis1, 2018-11-28:
Yes, this is a known issue that has been discovered after v4.4.1 version release.

The issue is solved in 4.4.2 develop branch (the code has been refactored).

The below patch should temporarily solve the issue for versions 4.4.1 and below. Please note that we didn't test this.
Look for `// <-- modified code` comments.

```c
            bool upLinkCounterFlag = false; // <-- modified code

            if( ( LoRaMacFlags.Bits.MlmeReq == 1 ) || ( ( LoRaMacFlags.Bits.McpsReq == 1 ) ) )
            {                
                if( ( ChannelsNbRepCounter >= ChannelsNbRep ) || ( LoRaMacFlags.Bits.McpsInd == 1 ) )
                {
                    ChannelsNbRepCounter = 0;

                    AdrAckCounter++;
                    if( IsUpLinkCounterFixed == false )
                    {
                        UpLinkCounter++;
                        upLinkCounterFlag = true; // <-- modified code
                    }

                    LoRaMacState &= ~MAC_TX_RUNNING;
                }
                else
                {
                    LoRaMacFlags.Bits.MacDone = 0;
                    // Sends the same frame again
                    ScheduleTx( );
                }
            }
        }

        if( LoRaMacFlags.Bits.McpsInd == 1 )
        {            
            if( ( McpsConfirm.AckReceived == true ) || ( AckTimeoutRetriesCounter > AckTimeoutRetries ) )
            {
                AckTimeoutRetry = false;
                NodeAckRequested = false;

                if( ( IsUpLinkCounterFixed == false ) && ( upLinkCounterFlag == false ) ) // <-- modified code
                {
                    UpLinkCounter++;
                }
                McpsConfirm.NbRetries = AckTimeoutRetriesCounter;

                LoRaMacState &= ~MAC_TX_RUNNING;
            }
        }
        upLinkCounterFlag = false; // <-- modified code
```

catsup, 2018-11-28:
Thanks for the confirmation Miguel.

We do already have a (tested) fix in place in our core, slightly different but with the same effect of only incrementing the counter once, I just wanted to make sure we are aware of the issue.

Closing the case.

closed 2018-11-28 by catsup