Lora-net/LoRaMac-node #599: LoRaMac 4.4.1 Ack timeout handle about confirmed message up

Dollar-Yao, 2018-12-03:
Hi, if the end-device send a confirmed message, and waiting for the downlink message from server. If the end-device actually received a downlink message but without set the ACK bit to 1 in FCtrl. So, at the end of OnRadioRxDone function, so the judgement of "if( AckTimeoutTimer.IsRunning == false )" can't be executed. After AckTimeoutTimer is timeout, we only set "AckTimeoutRetry = true;". So, there is no place to set LoRaMacFlags.Bits.MacDone to 1. Thus, in function of OnMacStateCheckTimerEvent, we can do nothing. Hope to correct me if I was wrong. Thanks.

mluis1, 2019-02-12:
We are sorry but, it is hard to understand what is the issue from your description.

As there is no activity for quite long time we close the issue.

closed 2019-02-12 by mluis1

singh667, 2021-01-08:
if i send the unconfirm data to the server .and when i got downlink data correct but ACKtimeout.Running==true  when its true and when it will false i am not getting this point .