Lora-net/LoRaMac-node #612: B-L072Z-LRWAN1 LoRaMac example sleep current is very high

jeremyherbert, 2018-12-10:
I checked out the latest master code, built it and flashed to to my board (B-L072Z-LRWAN1, only modification is the removal of the soldered on current shunts). I'm measuring ~1.1mA going into the VDD_RF pin (ie the SX1276 supply), but as far as I can tell it is supposed to be in sleep mode with <1uA of current.

I can modify the code to get the STM32 into it's proper low power state by disabling the TCXO (<2uA), but I haven't been able to work out how to actually put the SX1276 to sleep properly. Is this a known issue?

jeremyherbert, 2018-12-10:
I have found that there is a jumper on the board which sets the TCXO to always on (JP9). However, when I set the jumper to be controlled by PA12, the device can no longer join the network. Still investigating...

jeremyherbert, 2018-12-10:
It seems that the issue is that by default, the TCXO is permanently powered on by JP9, so all of the logic around the TCXO in the code doesn't appear to actually do anything. To control the power to the TCXO, JP9 needs to be put between pins 1 and 2.

It also seems that perhaps the logic is not quite working in the LoRaMac example for low power/STOP mode; when the micro performs a join, it starts the process with the SX1276 and then enters STOP mode. However, when doing this it drives PA12 to GND (I can't actually find the code/functionality that does it, but I have measured it occurring with an oscilloscope). This is the pin which powers the TCXO and as a result, the SX1276 is not able to successfully complete the LoRaWAN join. If STOP mode is disabled, everything works as expected and the join completes properly.

So it seems there really needs to be two separate sleep modes; one in which the MCU is waiting for the SX1276 to do something (TCXO must remain on), and one in which the SX1276 is completely idle (TCXO can be turned off).

jeremyherbert, 2018-12-10:
Ok it seems I was slightly wrong earlier. It doesn't seem to work correctly when the jumper is set to control the TCXO, regardless of whether the sleep modes are enabled.

I have found two issues so far, first is that when the MCU exits STOP mode, `SX1276IoInit` is called. It has this line in it (sx1276-board.c line 107) which turns off the TCXO:

`GpioInit( &TcxoPower, RADIO_TCXO_POWER, PIN_OUTPUT, PIN_PUSH_PULL, PIN_NO_PULL, 0 );`

So every time it wakes from sleep the TCXO is turned off. Instead, it should be

`GpioInit( &TcxoPower, RADIO_TCXO_POWER, PIN_OUTPUT, PIN_PUSH_PULL, PIN_NO_PULL, 1 );`

Next is that the TCXO is turned off too early by `SX1276SetAntSwLowPower`; if I add the fix as above and also comment out line 273 in `sx1276-board.c`:

`SX1276SetBoardTcxo( false );`

everything is working fine, regardless of whether sleep mode is enabled. However this is obviously not appropriate, because the TCXO is always enabled.

FWIW, I am testing sleep mode by changing `GetBoardPowerSource` in `board.c` to always return `BATTERY_POWER`

maklroland, 2018-12-16:
I have the same issue. I use B-L072Z-LRWAN1 for development and I am working on a product based on the Murata module. I designed the pin connections the same as the dev board (PA12 for TCXO controlling). However it doesn't seem to work when using LoRaMac. I used pure LoRa with the ST LoRa SDK and TCXO was working fine and achived very low currents (uAish). With Semtech stack TCXO is not handled properly so I am forced to set PA12 high which result is >1mA STOP current.
I was not able to resolve it just by comparing the ST codes and Semtech codes, but I will dive into it much deeper. I think there must be something trivial...

jeremyherbert, 2018-12-16:
@maklroland my solution so far is to just manually switch the TCXO on before a transmit, and off after. It seems to be working fine at the moment, but it is definitely a workaround.

maklroland, 2018-12-16:
@jeremyherbert unfortunately I cannot do that, as I use the LoRaMac which controls all the Radio.Xxx(); functions and these functions appear to be good by themselves. I checked all the Antenna switching and TCXO switching functions and although I found differences in the order of calling them (btw. ST SDK and Semtech SDK) I could not manage to resolve the issue by rearranging them.
It seems the LoRaMac itself is messing things up, but its a blind guess.

maklroland, 2018-12-16:
UPDATE: If I don't let the MCU go into STOP mode, which would otherwise invoke `BoardDeInitMcu`, everything is fine, stack is doing its job with the TCXO.
If I enable the STOP mode, and comment out `SpiDeInit( &SX1276.Spi );` and `SX1276IoDeInit( );` then move `SpiInit( &SX1276.Spi, SPI_1, RADIO_MOSI, RADIO_MISO, RADIO_SCLK, NC );` and `SX1276IoInit( );` into the `if( McuInitialized == false )` it works as well.
I guess the culprit is
`CRITICAL_SECTION_BEGIN( );
if (IsMacProcessPending == 1) {
	IsMacProcessPending = 0;
} else {
	BoardLowPowerHandler();
}
CRITICAL_SECTION_END( );`
as if LoRaMac is not always setting `IsMacProcessPending` in time and MCU can DeInit all the stuff before the stack is done working with the SX1276.

Regimbal, 2019-03-30:
Guys,
I'm facing the exact same problem.
However, we had some advance recently thanks to another posted issue #525 
@jeremyherbert in your post of the 10th december, you correct yourself saying, i quote:

> It doesn't seem to work correctly when the jumper is set to control the TCXO, regardless of whether the sleep modes are enabled.

This now seems to be solved by  @mluis1 in  [ #525 ](https://github.com/Lora-net/LoRaMac-node/issues/525#issuecomment-477650105) thanks to his proposed patch.

Now, things are still wrong when sleep modes are enable. I get a Tx Timeout.

Regimbal, 2019-04-01:
I've found a solution, but that's probably not the most elegant way to correct this behavior.
Basically, I observed that BoardInitMCU was called at mcu wakeup after PA_12 is set to one (or at least it prevents to work normally).
What I did is to initialize the PA_12 pin in a dedicated function, outside of SX1276IoInit.
Then I placed this dedicated function in the second McuInitialized test so that the init of PA_12 is done only at first MCU start.
In the end, the PA_12 pin is always initialized but still controlled on/off by the patch @mluis1 proposed.

This yields to
board.c >BoardInitMCU :
```
    SpiInit( &SX1276.Spi, SPI_1, RADIO_MOSI, RADIO_MISO, RADIO_SCLK, NC );
    SX1276IoInit( );
    if( McuInitialized == false )
    {
        SX1276InitBoardTcxo( );
        McuInitialized = true;
        if( GetBoardPowerSource( ) == BATTERY_POWER )
        {
            CalibrateSystemWakeupTime( );
        }
    }
```
In sx1276-board.c > SX1276IoInit
```
void SX1276IoInit( void )
{
    GpioInit( &SX1276.Spi.Nss, RADIO_NSS, PIN_OUTPUT, PIN_PUSH_PULL, PIN_PULL_UP, 1 );

    GpioInit( &SX1276.DIO0, RADIO_DIO_0, PIN_INPUT, PIN_PUSH_PULL, PIN_PULL_UP, 0 );
    GpioInit( &SX1276.DIO1, RADIO_DIO_1, PIN_INPUT, PIN_PUSH_PULL, PIN_PULL_UP, 0 );
    GpioInit( &SX1276.DIO2, RADIO_DIO_2, PIN_INPUT, PIN_PUSH_PULL, PIN_PULL_UP, 0 );
    GpioInit( &SX1276.DIO3, RADIO_DIO_3, PIN_INPUT, PIN_PUSH_PULL, PIN_PULL_UP, 0 );
    GpioInit( &SX1276.DIO4, RADIO_DIO_4, PIN_INPUT, PIN_PUSH_PULL, PIN_PULL_UP, 0 );
    GpioInit( &SX1276.DIO5, RADIO_DIO_5, PIN_INPUT, PIN_PUSH_PULL, PIN_PULL_UP, 0 );
    //GpioInit( &TcxoPower, RADIO_TCXO_POWER, PIN_OUTPUT, PIN_PUSH_PULL, PIN_NO_PULL, 0 );

#if defined( USE_RADIO_DEBUG )
    GpioInit( &DbgPinTx, RADIO_DBG_PIN_TX, PIN_OUTPUT, PIN_PUSH_PULL, PIN_NO_PULL, 0 );
    GpioInit( &DbgPinRx, RADIO_DBG_PIN_RX, PIN_OUTPUT, PIN_PUSH_PULL, PIN_NO_PULL, 0 );
#endif
}
```
In sx1276-board.c > new dedicated function to declare in sx1276-board.h two
```
void SX1276InitBoardTcxo( void )
{
    GpioInit( &TcxoPower, RADIO_TCXO_POWER, PIN_OUTPUT, PIN_PUSH_PULL, PIN_NO_PULL, 0 );
}
```

Sorry I did not prepared a proper git patch since i'm not sure that would be the best way to correct the issue. Still it works, and I checked PA_12 is switched on/off with a voltmeter on the jumper.

mluis1, 2019-04-04:
@Regimbal Thanks for the feedback and proposition.

I have prepared a patch that you will find below based on your proposal.

It would be nice to have feedback on this proposal from people using the different platforms options.
Once someone has validated he can edit this post and put a check (i.e: [x] ) in front of the tested option.
1. B-L072Z-LRWAN1
    - [ ] SX1276 - with TCXO
2. NAMote72
    - [ ] SX1272 - no TCXO
3. NucleoLxxx
    - [ ] SX1261MBxBAS - no TCXO
    - [ ] SX1262MBxCAS - no TCXO
    - [ ] SX1262MBxDAS - with TCXO
    - [ ] SX1272MB2DAS - no TCXO
    - [ ] SX1276MB1LAS - no TCXO
    - [ ] SX1276MB1MAS - no TCXO
4. SAML21
    - [ ] SX1276 - no TCXO
5. SKiM880B
    - [ ] SX1272 - no TCXO
6. SKiM881AXL
    - [ ] SX1272 - no TCXO
7. SKiM980A
    - [ ] SX1272 - no TCXO

[regimbal-proposal.patch.txt](https://github.com/Lora-net/LoRaMac-node/files/3043636/regimbal-proposal.patch.txt)

Regimbal, 2019-04-04:
you're welcome.
I've been testing on the B-L072-LRWAN1 and this is okay so far.
I've managed to get a nice 2uA consumption in stop mode on a board based on the murata module (providing additional optimizations when USE_DEBUGGER is switched off (ie disable debug pins and HAL debug).
We have other custom boards based on the uM880B, we will provide the results in a few days.

Surajmhaske, 2019-04-12:
Hi,
I am using B-L072Z-LRWAN1 with AT_Slave project from I-Cube-LORAWAN software. How can I measure current consumption? What changes are required to be done on board like jumper position and across which pins do I measure current

jeremyherbert, 2019-04-18:
Hi @Surajmhaske 

This is an issue about the LoRaMac-node stack which is not the same as the ST stack (although the ST stack appears to be derived from this, I have come across some weird bugs in it!), so you are probably asking in the wrong place. But the general approach I used to measure the current is:

1. Open SB14, SB19 and SB17
2. Close or measure across jumpers JP1, JP2, JP3

closed 2019-06-07 by mluis1