Lora-net/LoRaMac-node #613: Stuck in LORAMAC_STATUS_BUSY

gpgib6, 2018-12-10:
Using feature 5.00 lorawan 1.1, class C.
Communication starts out working great, then at some point function
LoRaMacMcpsRequest( McpsReq_t* mcpsRequest )
starts to only return LORAMAC_STATUS_BUSY;

It looks like the LoRaMAC gets stuck in this busy state, unless I'm doing something wrong...

Adamzat, 2019-01-30:
Hi gpgib6
I have the same problem like you using the Lora Stack version 4.4.0. Actually, when I stressed my module to send data periodically (each 5 seconds or less). Sometimes I fall in the state LORAMAC_STATUS_BUSY (note that at this state no downlink were received on RX2, so I don't understand while the LoraMac is busy !)
I did not have this problem in class A.
Any Idea ?

mluis1, 2019-02-12:
From your descriptions we don't know how to reproduce the issue. Thus, it is hard for us to figure out what could possibly cause the observed issues.

Could you please try to set some breakpoints and launch the debugger in order to try to find where the issue happens and potentially what causes it.

@Adamzat Could you please update your code at least to 4.4.1 release or even better to the code provided on develop branch. The observed issue may have already been solved.

closed 2019-02-12 by mluis1

Adamzat, 2019-02-12:
Thank you mluis1, 
I will check it.

inditn, 2019-03-26:
Hi there, any solution found regarding this issue please? thank you.

ChrisEAlfred, 2019-06-20:
We are using the Nucleo-L073 with SX1262MB2xAS on AS923 in Class C.
We on the latest develop git SHA: 6231a5d65806142a6f887e6183227b8da9e49238.

After about 2-4 hours and transmitting every 60 seconds, LoRaMacMcpsRequest is returning LORAMAC_STATUS_BUSY.

It appears that the previous and all other transmissions had a corresponding MCPS indication. However, the last successful transmit does not have a corresponding MCPS indication even after 60 seconds. From there on, all LoRaMacMcpsRequest calls return LORAMAC_STATUS_BUSY.

I placed a breakpoint in LoRaMacMcpsRequest and have confirmed that in LoRaMacIsBusy(), MacCtx.MacState = LORAMAC_TX_RUNNING
and MacCtx.AllowRequests = LORAMAC_REQUEST_HANDLING_ON, hence busy.

The MCPS request is:
mcpsRequest
	Type		unconfirmed
	fPort		3
	fBuffer		20001564
	fBufferSize	240
	Datarate	0

The MacCtx is:
-		MacCtx	{DevEui = 0x00000000, JoinEui = 0x00000000, ...}	LoRaMacCtx_t
		DevEui	0x00000000	uint8_t *
		JoinEui	0x00000000	uint8_t *
		PktBufferLen	253	uint16_t
+		PktBuffer		uint8_t [255]
+		TxMsg	{Type = LORAMAC_MSG_TYPE_DATA, Message = {...}}	LoRaMacMessage_t
+		AppData		uint8_t [255]
		AppDataSize	240 'ð'	uint8_t
+		RxPayload		uint8_t [255]
+		LastTxSysTime	{Seconds = 3245, SubSeconds = 386}	SysTime_t
		MacState	2	uint32_t
+		MacPrimitives	0x20001430 <_ZL4_app+88> {0x0801f091 <McpsConfirm(McpsConfirm_t*)>}	LoRaMacPrimitives_t *
+		MacCallbacks	0x20001440 <_ZL4_app+104> {0x0801f139 <BatteryLevel()>}	LoRaMacCallback_t *
+		RadioEvents	{TxDone = 0x0800a319 <OnRadioTxDone>, ...}	RadioEvents_t
+		TxDelayedTimer	{Timestamp = 0, ReloadValue = 0, IsStarted = false, ...}	TimerEvent_t
+		RxWindowTimer1	{Timestamp = 1475, ReloadValue = 1002, ...}	TimerEvent_t
+		RxWindowTimer2	{Timestamp = 999, ReloadValue = 2040, ...}	TimerEvent_t
		RxWindow1Delay	979	uint32_t
		RxWindow2Delay	1993	uint32_t
+		RxWindow1Config	{Channel = 2 '\002', Datarate = 5 '\005', ...}	RxConfigParams_t
+		RxWindow2Config	{Channel = 0 '\000', Datarate = 2 '\002', ...}	RxConfigParams_t
+		RxWindowCConfig	{Channel = 0 '\000', Datarate = 0 '\000', ...}	RxConfigParams_t
		AdrAckLimit	64	uint16_t
		AdrAckDelay	32	uint16_t
+		AckTimeoutTimer	{Timestamp = 3406, ReloadValue = 4408, ...}	TimerEvent_t
		ChannelsNbTransCounter	1 '\001'	uint8_t
		AckTimeoutRetries	1 '\001'	uint8_t
		AckTimeoutRetriesCounter	1 '\001'	uint8_t
		AckTimeoutRetry	false	_Bool
		NodeAckRequested	false	_Bool
		Channel	2 '\002'	uint8_t
		TxTimeOnAir	395	TimerTime_t
+		McpsIndication	{McpsIndication = MCPS_UNCONFIRMED, ...}	McpsIndication_t
+		McpsConfirm	{McpsRequest = MCPS_UNCONFIRMED, ...}	McpsConfirm_t
+		MlmeConfirm	{MlmeRequest = MLME_JOIN, ...}	MlmeConfirm_t
+		MlmeIndication	{MlmeIndication = MLME_JOIN, ...}	MlmeIndication_t
		RxSlot	RX_SLOT_NONE	LoRaMacRxSlot_t
+		MacFlags	{Value = 1 '\001', Bits = {...}}	LoRaMacFlags_t
		AllowRequests	LORAMAC_REQUEST_HANDLING_ON	LoRaMacRequestHandling_t
+		NvmCtx	0x20000820 <NvmMacCtx> {LORAMAC_REGION_AS923}	LoRaMacNvmCtx_t *

Chris Alfred