Lora-net/LoRaMac-node #614: AS923 Dwell Time Problem when TxParamSetupAns is Lost

psuggate, 2018-12-11:


jamesl-dm, 2019-01-21:
Any progress on resolving this? I've had this issue disconnect a few nodes during testing as well, and it looks like it'll need to be addressed by a standard change.

reissjason, 2019-01-21:
Why is DR2 used to in downlinks to the end-device following an uplink for DR0/DR1?
Rx1 should match the datarate used in the uplink regardless of dwelltime settings. Only if Rx1Offset is used would there ever be a non-matching datarate.
If DR0/DR1 is not allowed then Rx2 at default datarate of DR2 should be used to reach the device.

Would it not be better for the network server to send using DR0/DR1 to get the end-device to stop using DR0/DR1 continually due to incorrect settings? This should not be required since Rx2 is available using an approved datarate.

psuggate, 2019-01-22:
DR2 is being used for these downlinks because Actility's LRC is strictly enforcing:

> The default JoinReq Data Rate is DR2 (SF10/125KHz), this setting ensures that end-devices are compatible with the 400ms dwell time limitation until the actual dwell time limit is notified to the end-device by the network server via the MAC command “TxParamSetupReq”.

from pp.39, lines 1033-1035 , of v1.0.2 of the regional parameters.

I have talked to Actility about this, and suggested that using `RX2` would fix the problem, without violating the above rule, just as you also suggested. Their response:

>  "this is very unlikely to happen"

even though sending the downlink at a different DR vs the uplink, during RX1, seems out-of-spec to me.

Having `TxParamSetupAns` as a "sticky" MAC command does solve this problem, and is what Actility suggested.

jamesl-dm, 2019-01-22:
>Would it not be better for the network server to send using DR0/DR1 to get the end-device to stop using DR0/DR1 continually due to incorrect settings?

>even though sending the downlink at a different DR vs the uplink, during RX1, seems out-of-spec to me.

Ironically the DR0/DR1 use isn't due to incorrect settings - it's what the network server asked for. Its just that the network server doesn't realize its request has been actioned already. So it looks up the assumed current state of the node, and sees that the node ought to think the downlink dwell is still on. Then it checks the Rx1Offset table, and sees that an SF12 uplink with downlink dwell on should be replied to at SF10:
[image]

To detect and workaround the problem, the server would need to reason that the node shouldn't be sending at SF12 unless it got the TxParamSetupReq, which disables both the uplink and downlink dwell at the same time. If it could work this out, it wouldn't really even need to use Rx2 to reply - it could just update the assumed current state to the intended 'no dwell' settings straight away. But I guess that level of reasoning is just a step too far. It would be better if the standard had a more robust and simple way of resolving this corner case. Making the reply sticky sounds like it would do the job.

psuggate, 2019-01-22:
Yeah, the Network Server (NS) really should handle this in a smarter way -- the NS can infer from the `FCntUp` value that an uplink was lost as well...

And Acility's LRC (their name for the NS) violates line 1235, that you quote above:

> The RX1 receive window uses the same channel than the preceding uplink.

ymasmoudi, 2019-01-22:
1/ Regarding the line 1235, LRC (NS) is using the same channel to answer the preceding uplink. 
2/ And as answered previously, It uses correctly DR2 as it didn't receive any response from the sensor that the Dwell time was deactivated.

A missing FCntUp is a sign that an uplink was lost but it doesn't provide any indication whether the device did include a TxParamSetupAns to disable dwell time in that missing frame.

mluis1, 2019-01-22:
I have pushed this topic to be discussed at the next LoRa-Allaince technical committee meeting which will be held in San Diego the 4th to 6th February 2019.

Currently, we can't modify the code as the sticky `TxParamSetupReq` MAC command answer is not compliant with the current specification.

If the technical committee considers this proposition then the change will most certainly happen trough an ERRATA note for the 1.0.3 specification. Only then we can change this. Nowadays, no NS implementation knows how to handle a sticky `TxParamSetupReq`  MAC command.

mluis1, 2019-04-02:
Just to inform that the change request has been accepted by the LoRa-Alliance technical committee.

This will be fixed once we start working on LoRaWAN 1.0.4 specification implementation.

I think that we can close this issue.

closed 2019-04-02 by mluis1