Lora-net/LoRaMac-node #616: Question about the SX126xReset funtion

Lozeonjyu, 2018-12-13:
Hi,

I am using the Nucleo L073+Sx1261 shield to load the board. During the code reading, I find something that confuses me a little bit. 

In file sx1261mbxbas-board.c, it introduces the sx126x reset function:
```
void SX126xReset( void )
{
    DelayMs( 10 );
    GpioInit( &SX126x.Reset, RADIO_RESET, PIN_OUTPUT, PIN_PUSH_PULL, PIN_NO_PULL, 0 );
    DelayMs( 20 );
    GpioInit( &SX126x.Reset, RADIO_RESET, PIN_ANALOGIC, PIN_PUSH_PULL, PIN_NO_PULL, 0 ); // internal pull-up
    DelayMs( 10 );
}
```

It seems like the micro drives the reset pin low and let the reset pin go back, but why it initialize again as analog rather than just writegpio 1 to the reset pin? And why setting the pin analogic can let the reset pin go back high again?

Thanks

abrinlee, 2018-12-13:
Just clarifying... 

The questions is why the use of PIN_ANALOGIC (Which I believe makes it an analog input pin) for the GPIO pin type instead of PIN_OUTPUT and then set back to a "1".

mluis1, 2019-02-12:
The radio reset pin has an internal pull-up. 
Thus, in order to avoid the MCU driving a 1 we just set the pin as input.

closed 2019-02-12 by mluis1

abrinlee, 2019-02-12:
Thank you for responding. I appreciate the follow up. Close the issue.