Lora-net/LoRaMac-node #618: EU868 Duty Cycle and join network request

omogenot, 2018-12-15:
According to ETSI-300-220-1 Duty Cycle compliance, in paragraph 5.4.2:

> 5.4.2 Conformance
An assessment of the overall Duty Cycle shall be made for a representative period of Tobs over the observation bandwidth Fobs. Unless otherwise specified, Tobs is 1 hour and the observation bandwidth Fobs is the operational frequency band.
The representative period shall be the most active one in normal use of the device. As a guide "Normal use" is considered as representing the behaviour of the device during transmission of 99 % of transmissions generated during its operational lifetime.
**Procedures such as setup, commissioning and maintenance are not considered part of normal operation.**
Where an acknowledgement is used, the additional transmitter on-time from a message responder shall be declared only once whether included in the message initiator Duty Cycle or in the message responder Duty Cycle.
NOTE:
The intention of this rule is not to allow EUT to exceed the maximum duty cycle value.

It looks to me that the join procedure (being commissioning the device) shall not be limited by the duty cycle restrictions... Therefore duty cycle shall be disabled during a Join Request and re-enabled afterwards. Or do you have a different reading of this paragraph?

ConnyBusy, 2018-12-15:
@omogenot 
Hope the below clearifies you.
[image]

ConnyBusy, 2018-12-16:
For testing purpose you could disable the duty cycle. By the way, when a join accept is not received the device should be in sleep mode and only wake up for retransmission of JoinReq. I don't see why the device should be consuming more battery.

omogenot, 2018-12-16:
Obviously the system goes in sleep mode between two retries. However, there are 2 level of retries, no? First level is in the JoinRequest function in the LoRaWAN stack, and the second level at the application level. And at the application level, each time the JoinRequest returns a duty cycle error, the application has to wait some time (in low power) and retry. But if this retry returns a duty cycle error again, the energy spent in this useless call is just wasted energy => consuming more battery for nothing.
Anyway, forget it.

closed 2018-12-16 by omogenot