Lora-net/LoRaMac-node #619: Compile this project

AlessandroSpallina, 2018-12-15:
Hello,
i'm intrested about modify settings in the class A application in /src/apps/LoRaMac/classA/B-L072Z-LRWAN1/main.c

i have an smt nucleo with the lora board by mbed.

i configure all the compiling phase with: cmake -DCMAKE_TOOLCHAIN_FILE="cmake/toolchain-arm-none-eabi.cmake" -DAPPLICATION="LoRaMac" -DCLASS="classA" ..

then i type make and i flash the file in /build/src/apps/LoRaMac/LoRaMac-classA.bin

am i correct? because, if try to insert an obvious compiling error in the main.c file i will get no error!

thanks guys

ConnyBusy, 2018-12-16:
@AlessandroSpallina 
does the apparently error compile code run on the device?
My guess is that you have not setup your target shield correctly. So the compiled code doesn't use the code you modified

AlessandroSpallina, 2018-12-16:
how to check that? @ConnyBusy

ConnyBusy, 2018-12-18:
In your CMakelist.txt files

mluis1, 2019-02-12:
As there is no more activity we close the issue.

closed 2019-02-12 by mluis1