Lora-net/LoRaMac-node #620: SX1276IsChannelFree(MODEM_FSK, ...) doesn't set the band width for Rssi scanning.

ty4tw, 2018-12-16:
In case of the FSK mode, band width should be set in registers for RegionAS923 LBT in Japan.
````
bool SX1276IsChannelFree( RadioModems_t modem, uint32_t freq, int16_t rssiThresh, uint32_t maxCarrierSenseTime )
{
    bool status = true;
    int16_t rssi = 0;
    uint32_t carrierSenseTime = 0;

    SX1276SetModem( modem );
    SX1276SetChannel( freq );
    SX1276SetOpMode( RF_OPMODE_RECEIVER );

    if ( modem == MODEM_FSK )
    {
	SX1276Write( REG_RXBW, GetFskBandwidthRegValue( FSK_RSSI_BANDWITH ) );
	SX1276Write( REG_AFCBW, GetFskBandwidthRegValue( FSK_RSSI_BANDWITH ) );
    }
    DelayMs( 1 );
````

mluis1, 2019-02-12:
Thanks for the comment.

We agree that the the proposed change can be done.
But, as the MAC layer always use the MODEM_LORA to check if the channel is free we should be OK for the time being.

mluis1, 2019-11-12:
After further analysis this change requires some changes on the radio API as well on the stack operation sequences.

The current implementation always uses the LoRa modem to perform the check if the channel is free or not. It means that the bandwidth in use will always be bigger than the one required for the FSK modulation. Thus, we think that the code can stay as is for the time being.

We then close this issue but, will keep it in mind in case we go through a MAC implementation refactor.

closed 2019-11-12 by mluis1