Lora-net/LoRaMac-node #622: SX1276 unable to receive downlink payloads more than 68 bytes?

bryce-cahill, 2018-12-17:
I'm currently trying to implement the I-CUBE-LRWAN software expansion (ST's wrapped version of the Semtech Stack) with the SX1276 LoRa module. This is working fine when sending normal payloads.

However, sometimes we need to receive large downlink payloads and this is where we run into problems, We are using spreading factor 7 so the air time and the maximum payload length shouldn't be an issue. Any bytes in the payload after the 68th byte become corrupted. Using an logic analyser we can see that the packet is being sent incorrectly from the LoRa module itself. So it appears that the issue is likely due to the SX1276 not being configured correctly.

Any help solving this issue would be greatly appreciated, thanks.

asob, 2019-01-11:
I'm seeing the same issue I-CUBE-LRWAN. Anything below 70 bytes works as expected. Any luck on finding a fix for that?

mluis1, 2019-02-12:
Unfortunately we aren't aware of the modifications that ST made in his fork.
We would recommend that you ask your question directly to ST?

Have you tried to use the code provided on this project?

**Note:** The LoRa-Alliance certification process has tests that check the ability for the end-device to receive the maximum allowed payload size. The last time we ran the tests they were successful.

closed 2019-02-12 by mluis1